mainsail-crew/crowsnest #4: Installation error building v4l2rtspserver

JohnOCFII, 2022-02-05:
### What happened

Crowsnest installation failed during build of v4l2rtspserver.  This on a fresh, new installation of MainSailOS 0.6.1 on a Raspberry Pi 3B+

First error was:

```[ 54%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/MemoryBufferSink.cpp.o
In file included from /home/pi/v4l2rtspserver/src/HTTPServer.cpp:24:
/home/pi/v4l2rtspserver/inc/HTTPServer.h: In constructor ‘HTTPServer::HTTPClientConnection::HTTPClientConnection(RTSPServer&, int, sockaddr_in, Boolean)’:
/home/pi/v4l2rtspserver/inc/HTTPServer.h:128:162: error: no matching function for call to ‘RTSPServer::RTSPClientConnection::RTSPClientConnection(RTSPServer&, int&, sockaddr_in&, Boolean&)’
          : RTSPServer::RTSPClientConnection(ourServer, clientSocket, clientAddr, useTLS), m_TCPSink(NULL), m_StreamToken(NULL), m_Subsession(NULL), m_Source(NULL) {
                                                                                                    
```

Full installation / build log included below.

### What did you expect to happen

Expected installation to complete successfully.

### How to reproduce

Clean out installation directories and attempt installation again.

### Additional information

[crowsnest_build_failure.txt](https://github.com/mainsail-crew/crowsnest/files/8008357/crowsnest_build_failure.txt)

KwadFan, 2022-02-12:
Hi,
Please remove liblivemedia-dev

`sudo apt remove liblivemedia-dev`

and try again.

JohnOCFII, 2022-02-12:
That worked!  Thanks.

Question - would you prefer crowsnest issues/discussion here on Github or on Discord?

closed 2022-02-12 by JohnOCFII