mainsail-crew/crowsnest #5: Refactor device enumeration in logging.sh

WatskeBart, 2022-02-13:
### Is your feature request related to a problem? Please describe

When using a HDMI-to-CSI module (eg [B101](https://auvidea.eu/b101-hdmi-to-csi-2-bridge-15-pin-fpc/) or [C780](aliexpress.com/item/1005002907475714.html)), these devices or not found by logging.sh.

These kind of devices are not listed in /dev/v4l/by-id/ (only USB devices are)

### Describe the solution you'd like

Instead of using [/dev/v4l/by-id/](https://github.com/mainsail-crew/crowsnest/blob/master/libs/logging.sh#L88)
try `/dev/v4l/by-path/`

This way all possible video devices are enumerated.

### Describe alternatives you've considered

In my setup i've changed line 88 to `/dev/v4l/by-path/` and everything works as expected.

### Additional information

Example of /dev/v4l/by-path/:
`pi@klipper:~ $ ll /dev/v4l/by-path/
total 0
lrwxrwxrwx 1 root root 12 Feb 13 10:47 platform-3f801000.csi-video-index0 -> ../../video0
lrwxrwxrwx 1 root root 12 Feb 13 10:47 platform-3f801000.csi-video-index1 -> ../../video1
lrwxrwxrwx 1 root root 12 Feb 13 11:17 platform-3f980000.usb-usb-0:1.4:1.0-video-index0 -> ../../video2
lrwxrwxrwx 1 root root 13 Feb 13 10:47 platform-bcm2835-codec-video-index0 -> ../../video10
lrwxrwxrwx 1 root root 13 Feb 13 10:47 platform-bcm2835-isp-video-index0 -> ../../video13
lrwxrwxrwx 1 root root 13 Feb 13 10:47 platform-bcm2835-isp-video-index1 -> ../../video14
lrwxrwxrwx 1 root root 13 Feb 13 10:47 platform-bcm2835-isp-video-index2 -> ../../video15
lrwxrwxrwx 1 root root 13 Feb 13 10:47 platform-bcm2835-isp-video-index3 -> ../../video16`

Example of /dev/v4l/by-id/:
`pi@klipper:~ $ ll /dev/v4l/by-id
total 0
lrwxrwxrwx 1 root root 12 Feb 13 11:35 usb-OmniVision_Technologies__Inc._USB_Camera-B4.09.24.1-video-index0 -> ../../video2`

I really like the crowsnest addon because of ustreamer. Previously I used mjpg-streamer as part of KIAUH, but it slower and more limited then ustreamer.

meteyou, 2022-02-13:
I would not replace it. For example: you use a usb webcam and change the port, by-id would also fit after the change. By-path would not fit and have to be modified.

WatskeBart, 2022-02-13:
For me to use my HDMI-to-CSI `platform-3f801000.csi-video-index0 -> ../../video0`, /dev/v4l/by-path/ is the only option.

KwadFan, 2022-02-15:
So, I thinkered around that. Replacing by-id isnt an Option for me, because its mainly made to support webcams and not hdmi2csi hats, besides the fact I dont know why you want to use a hdmi source as feed for a webcam daemon. But I dont want to give it up, maybe as additional input for the logging function, but it is last on my todo for now. As a reminder, if you use moonrakers update Manager to update crowsnest it will fail if you modify files.
Regards

KwadFan, 2022-03-07:
@WatskeBart 

Please checkout the new branch https://github.com/mainsail-crew/crowsnest/tree/feat/csi-adaptors and test the changes.
If you report success, I will merge it ASAP.
Shouldnt you react in any form on this I will drop it.

Regards

KwadFan, 2022-03-26:
Hello @WatskeBart ,
sad that you didnt response, but I think I did it.
If this wont work for you feel free to open a new Issue.

Closing Issue due commit [62ee73b](https://github.com/mainsail-crew/crowsnest/commit/62ee73b6da1792d6d125a6acd2ddb0bbde217d3d)

Regards

closed 2022-03-26 by KwadFan