mainsail-crew/crowsnest #11: v4l2ctl does not properly apply after reboot

gentleseal, 2022-04-04:
### What happened

I defined `v4l2ctl: focus_auto=0,exposure_auto=1,exposure_absolute=300,focus_absolute=250`
However, after I reboot, everything starts up, I get a picture, but it is out of focus (seems focus is not set).
if I then restart webcamd manually, it restarts with all the values correctly set and in-focus picture

### What did you expect to happen

All `v4l2ctl` values to apply when rebooting

### How to reproduce

For my specific webcam, 
- put at least `v4l2ctl: focus_auto=0,focus_absolute=1000` in the config
- reboot the os
- *the camera starts with default focus value (not the super out of focus 1000 as set)*
- restart webcamd from mainsail
- *works as expected*

### Additional information

- Manually using `v4l2-ctl -c ...` after booting sets the value without any issues
- My previous setup was a script (with a bunch of `v4l2-ctl -c ...`) that ran when octoprint started (I appreciate that might not help much), which functioned fine
- Previously discussed in Discord https://discord.com/channels/758059413700345988/811863253637398558/960229045373644840

Config:
[webcam.conf.txt](https://github.com/mainsail-crew/crowsnest/files/8409154/webcam.conf.txt)
Rebooting log:
[webcamd.log](https://github.com/mainsail-crew/crowsnest/files/8409152/webcamd.log)
SSH after reboot:
[after_reboot.txt](https://github.com/mainsail-crew/crowsnest/files/8409155/after_reboot.txt)
Rebooting and then manually restarting webcamd log:
[webcamd_restarted.log](https://github.com/mainsail-crew/crowsnest/files/8409150/webcamd_restarted.log)

lixxbox, 2022-04-04:
Hi here,
I can't reproduce it unfortunately. My camera does not support focus options. However, I tested other parameters like brightness, contrast and saturation. After a reboot they were applied automatically.
I will test it with a few other webcams later.

Have you tried to read the parameters after a reboot? `v4l2-ctl -C ...`

gentleseal, 2022-04-04:
I forgot to add webcam model. It's an OEM one which might be the reason for its weirdness.
The amazon listing seems deleted.
lsusb gives `0c45:6366 Microdia`.

gentleseal, 2022-04-04:
I will try to read out the values after a reboot in a bit

gentleseal, 2022-04-04:
for `v4l2ctl: focus_auto=0,exposure_auto=1,exposure_absolute=300,focus_absolute=250`
here are the values after reboot:
```
pi@ender3:~ $ v4l2-ctl -C focus_auto
focus_auto: 0
pi@ender3:~ $ v4l2-ctl -C exposure_auto
exposure_auto: 1
pi@ender3:~ $ v4l2-ctl -C exposure_absolute
exposure_absolute: 300
pi@ender3:~ $ v4l2-ctl -C focus_absolute
focus_absolute: 180
```
All seems correct other than focus_absolute for some reason

gentleseal, 2022-04-04:
Looking into it more:

changing webcamd to read
```
v4l2_control
construct_streamer
v4l2_control
```
gets it to work for reasons I don't understand 

I also tried various delays and doing it after which did not seem to work.

From manually messing around with it, it seems like
1. `v4l2_control` occurs setting all values
2. Values **are set correctly** including `focus_absolute `
3. Doing `construct_streamer` aka starting to use the webcam for some reason resets `focus_absolute` to defaults
4. Setting it again after achieves the desired result 

🤯

KwadFan, 2022-04-04:
Based on that, would support my theory about the "flag=inactive" in focus_absolute.
But alexz has the same Problem without noticing it ( his machine runs 24/7 ), but no flag.

Now it is consideration of having a couple frames with bad quality or some Instance to regulary check that values and correct them if needed :thinking:

KwadFan, 2022-04-08:
Hello,

with the highly apriciated help of @gentleseal I had implemented a fix for that.
At this stage it's in the develop branch. Will be merged to master as soon I think its tested enough.

According commits are
2e3cf935686b95315bac260f28ba9b816dc2c830 and 3bfea5ad19d24d2d73cc63cf286c39aafadc2065

I will close this issue as soon its merged to master branch.

Regards

KwadFan, 2022-04-15:
So, today I pushed those changes to master branch.

This issue should be fixed now.

Please switch to master branch :)

Regards

closed 2022-04-15 by KwadFan