mainsail-crew/crowsnest #12: Update ustreamer link to actual version

alexandrkut, 2022-04-14:
### What happened

Crowsnest fail compile with old version ustreamer.

```
pi@klipper:~/crowsnest $ make build
make -C bin
make[1]: Entering directory '/home/pi/crowsnest/bin'
make ustreamer
make[2]: Entering directory '/home/pi/crowsnest/bin'
make[2]: warning: -j4 forced in makefile: resetting jobserver mode.
Compiling ustreamer with OMX Support.
WITH_OMX=1 make -C ustreamer
make[3]: Entering directory '/home/pi/crowsnest/bin/ustreamer'
make apps
make[4]: Entering directory '/home/pi/crowsnest/bin/ustreamer'
make -C src
make[5]: Entering directory '/home/pi/crowsnest/bin/ustreamer/src'
-- CC libs/base64.c
-- CC libs/frame.c
-- CC libs/logging.c
-- CC libs/memsink.c
-- CC libs/options.c
-- CC libs/unjpeg.c
-- CC ustreamer/blank.c
-- CC ustreamer/data/blank_jpeg.c
-- CC ustreamer/data/index_html.c
-- CC ustreamer/device.c
-- CC ustreamer/encoder.c
-- CC ustreamer/encoders/cpu/encoder.c
In file included from ustreamer/encoder.h:45,
                 from ustreamer/encoder.c:23:
ustreamer/encoders/omx/encoder.h:33:10: fatal error: IL/OMX_Core.h: No such file or directory
   33 | #include <IL/OMX_Core.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[5]: *** [Makefile:112: build/ustreamer/encoder.o] Error 1
make[5]: *** Waiting for unfinished jobs....

```

### What did you expect to happen

Wwith new version build ok.

### How to reproduce

git clone --recurse-submodules https://github.com/mainsail-crew/crowsnest.git
make build

KwadFan, 2022-04-15:
Hello,

To describe, what exactly happend and why there is only a manual fix, right now.

As I started to develop crowsnest there where no "bullseye" in sight. And "buster" relies on the OpenMAX IL Video encoder.
Something like M2M ( Machine-to-Machine ) was not an Option. But it is now, in detail with kernel <5.15.

The used ustreamer version 4.13 was the last actual version that supports OMX as Hardware accellerated encoder, which is used to encode MJPG Streams of cameras that only supports YUYV pixel format. As the latest version of MainsailOS, which is my development platform is based on "buster" I have to ensure that it works flawless.
Therefor I have to stay with the older Version of ustreamer because ustreamer in Version <5 dropped support of OMX and uses M2M instead. That is the technical description of your Issue.

As a solution update your ustreamer manually, for now. I will implement a fix, when I finished the transition of mainsailOS to "bullseye". So, please be patient. 

To manually update ustreamer do the following:
`cd ~
cd crowsnest/bin/ustreamer
git checkout master
git fetch
git pull
sudo systemctl stop webcamd.service
make clean
make -j4
sudo systemctl start webcamd.service`

This will update ustreamer to v5.3 ( I think its actual version right now ). This version is supported in "bullseye"

Regards

KwadFan, 2022-06-11:
Close due 6071c6f

closed 2022-06-11 by KwadFan