mainsail-crew/crowsnest #14: After installation and reboot, moonraker.conf is empty except the added lines for crowsnest-updating

top-gun, 2022-05-25:
### What happened

I installed crowsnest on a Mainsailos image from January. Not sure about the version, I don't know how to look that up atm.

Installation was via Putty - I copied the text from the documentation and let all lines run. I must admit I mistyped twice when I entered "YES" as I thought "y" would be sufficient and had to restart the last line :)

The installation worked its way and instructed me to reboot. After reboot, I got a message from Mainsail that my moonraker.conf is missing the [server] section and I was running on a backup copy.

The moonraker.conf had lost all standard content, and only the fresh lines about updating the crowsnest code were present. I was able to copy-paste the old content from the moonraker-logs and got a working setup.

### What did you expect to happen

The installation should not delete the content of moonraker.conf, instead append the update section for crowsnest to the end.

### How to reproduce

Not sure if this would work again, as I have not tried to reproduce it.

Basically, follow the installation instructions. Don't type "yes", repeat the installation command. Do that twice ;)
At the third go of install, precisely type "YES" (I know this looks stupid, but it was late)

### Additional information

I can only imagine there was some uncommon artifact in my moonraker.conf, but atm I don't know how to recover an old version of the file.

top-gun, 2022-05-25:
[moonraker (1).log](https://github.com/mainsail-crew/crowsnest/files/8768460/moonraker.1.log)

This log goes back to 22/05/20, maybe it helps.

KwadFan, 2022-05-25:
I will test that today, but in generall Crowsnest Installer shouldnt add any moonraker entrys as long you use `make unattended`.
So, I'll have to review the according part.

I let you know if I find something. Could you test the installation another time?

Regards

commit fab3772e (2022-05-25): fix(install): fixes error that leads to unusable moonraker.conf

This should fix #14
closed 2022-05-25 by KwadFan