mainsail-crew/crowsnest #16: Package 'libjpeg8-dev' has no installation candidate

4noxx, 2022-05-26:
### What happened

cant install crownest

### What did you expect to happen

klipper@raspberrypi:~/crowsnest $ make install
crowsnest - A Webcam Daemon for Raspberry Pi OS

        Some Parts of the Installer requires 'root' privileges.
        You will be prompted for your 'sudo' password, if needed.

Running apt update first ...
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.

Install webcamd Service ...
Installing 'crowsnest' Dependencies ...

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Installing 'crowsnest' Dependencies ... [OK]
Linking webcamd ... [OK]
Copying webcamd.service file ... [OK]
Linking logrotate file ... [OK]
Enable webcamd.service on boot ... [OK]
Do you want to add [update_manager] entry?(y/n) n
Adding Crowsnest Update Manager entry to moonraker.conf ... [SKIPPED]
Add User klipper to group 'video' ... [SKIPPED]
==> User klipper is already in group 'video'
Build dependend Stream Apps ...
Installing 'ustreamer' Dependencies ...

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

E: Package 'libjpeg8-dev' has no installation candidate
ERROR: Error 100 occured on line 267
ERROR: Stopping install.sh.
Goodbye...
make: *** [Makefile:36: install] Error 1

### How to reproduce

see (2)

lbibass, 2022-05-28:
This is because you're probably using Bullseye. They removed that dependency.  

You can get around this by updating ustreamer to the newest git release, and changing the dependency to libjpeg instead.

KwadFan, 2022-05-29:
Will fix that during next week. But as @lbibass already said, known issue.

commit 6071c6fc (2022-06-11): refactor(installl): refactor install chain

This commit should cover the following:

Missing insatllation candidate of libjpeg8-dev in bullseye images,
fixes #16 and #15

Refactores installation of ustreamer.
This should cover #12

Dropped ustreamer as submodule and clone it via install script.
For buster based images it reset the repo to last commit of v4.13 (omx support)
All other will get master branch (latest commit at insatllation)

This enables posibility to write an app updater.

Fixes custompios module according to changes.
closed 2022-06-11 by KwadFan