mainsail-crew/crowsnest #25: Use pkcon (PackageKit's command line client) to manupulate packages

compihu, 2022-07-11:
### Is your feature request related to a problem? Please describe

Install script currently fails on non-debian-based distros as it uses apt to manipulate packages.

### Describe the solution you'd like

Using PackageKit (https://www.freedesktop.org/software/PackageKit) will make it possible to install on any distributions supporting PackageKit despite the package manager they're using internally (most of the active distros)

### Describe alternatives you've considered

Supporting multiple package managers by hand would be the pain. This is what PackageKit offers.

KwadFan, 2022-07-23:
I really like that Idea, and I am up to implement something like this.
But it will take time, first of all I need need to get deeper into PackageKit to understand it and how to use properly.

The only concern I have, is how to get PackageKit during the Installation for various Distros without any interaction of the user? Ideas?

I will keep this open, I am really interessted in that.

Regards

KwadFan, 2022-11-15:
@compihu Do you have a special Distribution in Mind? I want to start playing around with pkcon ...