mainsail-crew/crowsnest #27: Crowsnest doesn't seem to write "focus_absolute" configuration value as described in ReadMe.md

chri-kai-in, 2022-08-06:
### What happened

Crowsnest was installed successfully and various values for v4l2-ctl could be entered in configuration file:

#### crowsnest.conf
#### This is mainsail / MainsailOS default config.
#### See:
#### https://github.com/mainsail-crew/crowsnest/blob/master/README.md
#### for details to configure to your needs.

#####################################################################
####                                                            #####
####      Information about ports and according URL's           #####
####                                                            #####
#####################################################################
####                                                            #####
####    Port 8080 equals /webcam/?action=[stream/snapshot]      #####
####    Port 8081 equals /webcam2/?action=[stream/snapshot]     #####
####    Port 8082 equals /webcam3/?action=[stream/snapshot]     #####
####    Port 8083 equals /webcam4/?action=[stream/snapshot]     #####
####                                                            #####
#####################################################################

[crowsnest]
log_path: ~/klipper_logs/crowsnest.log  # Default logfile in ~/klipper_logs/crowsnest.log
log_level: verbose                      # Valid Options are quiet/verbose/debug
delete_log: false                       # Deletes log on every restart, if set to true

[cam 1]
mode: mjpg                              # mjpg/rtsp
port: 8080                              # Port
device: /dev/video0                     # See Log for available ...
resolution: 1920x1080                   # widthxheight format
max_fps: 15                             # If Hardware Supports this it will be forced, ohterwise ignored/coerced.
#custom_flags:                          # You can run the Stream Services with custom flags.
v4l2ctl: focus_auto=0,focus_absolute=5,sharpness=110,brightness=30,power_line_frequency=1  # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.

All parameters for v4l2-ctl are applied, except the value "focus_absolute=5" This is always set to value "2", whenever crowsnest is restarted. Only with commandline "v4l2-ctl -d /dev/video0 -c focus_absolute=5" the current configuration value in the camera module can be successfully altered. 

### What did you expect to happen

All v4l2-ctl parameters added in the list should be applied to the camera module, including values for focus_absolute.

### How to reproduce

1) Apply above configuration to your crowsnest test environment and apply it with hitting the save & restart button in Mainsail
2) read out current configuration values in your camera module. (v4l2-ctl -d /dev/video0 --list-ctrls-menus). All values should have been altered to the ones in Crowsnest Konfiguration except "focus_absolute"
3) issue "v4l2-ctl -d /dev/video0 -c focus_absolute=5" on command line and read out altest configuration values. Value for  "focus_absolute" shall be altered to "5"

chri-kai-in, 2022-08-06:
Added logfile with debug information and applied crowsnest.conf file:
[Crowsnest_focus_absolute_issue.zip](https://github.com/mainsail-crew/crowsnest/files/9275316/Crowsnest_focus_absolute_issue.zip)
.

KwadFan, 2022-08-08:
Hey,
could you share a screenshot to verify if your cam is able to set to that value please?

Also a screenshot after altering via commandline.

KwadFan, 2022-08-08:
So, I read your log.
There is something strange going on. You configured value 4, but it reads out value 7 after applying your set value.
I have to investigate that further. Please be patient I will fix that asap

chri-kai-in, 2022-08-08:
I played with the configuration during the analysis of the issue and changed the value back and forth between 4&5. Could be the configuration values in the log files are changing between the several restarts. Do you still need the screenshots ? Then i will take some tomorrow and provide them here.

KwadFan, 2022-08-09:
> I played with the configuration during the analysis of the issue and changed the value back and forth between 4&5. Could be the configuration values in the log files are changing between the several restarts. Do you still need the screenshots ? Then i will take some tomorrow and provide them here.

Would be great, because I am hunting for it but can not explain why v4l2-ctl gives back 7 if 6 is the max of your cam^^

KwadFan, 2022-08-09:
So, I found the bug, needed some help from @lixxbox but we discovered it.
Thanks for testing @zellneralex and Itzner :)

closed 2022-08-09 by KwadFan

chri-kai-in, 2022-08-09:
Confirm it is solved with me. Thank You for your quick reaction, analysis and solution of the issue !