mainsail-crew/crowsnest #28: Failed to install ustreamer dependencies

SHKinsem, 2022-08-09:
### What happened

I am not really sure what happened.
The following lines were copied from the ssh terminal.

```
> Installing 'ustreamer' Dependencies ...
> 
> WARNING: apt does not have a stable CLI interface. Use with caution in scripts.
> 
> E: Package 'libjpeg8-dev' has no installation candidate
> E: Unable to locate package libraspberrypi-dev
> ERROR: Error 100 occured on line 325
> ERROR: Stopping install.sh.
> Goodbye...
> make: *** [Makefile:35: install] Error 1
```

### What did you expect to happen

It should be successfully Installed

### How to reproduce

Using Orangpi Zero
system: Armbian 22.05.3 Buster with Linux 5.15.48-sunxi

Following the instructions, I pasted the commands to the ssh terminal, after a while the problem occurs.

bryanboettcher, 2022-08-11:
Seconded, same setup.

KwadFan, 2022-08-11:
I couldnt test that specific Hardware, because I only use Rpi's, but...

I want to provide a solution, so, one of the team members owns a orange pi. I'll try to fix it.

Maybe one of you could switch the branch for testing before I merge the fix to master?

Let me know if you are capable of and willing to do that :)

Regards Kwad

KwadFan, 2022-08-11:
So, getting things done feels good :)

Please use fix/orangeinstall branch and test it.
We were able to get it running on OrangePI Zero with buster and bullseye.

Happy testing :)

If you report back success it will get merged asap

SHKinsem, 2022-08-12:
OK. I tried bullseye and it works, thx

closed 2022-08-12 by KwadFan

alejmrm, 2022-09-06:
One question, How do I "reinstall" or allow to re-run the configuration of this feature? In my case, it seems like between Version: v3.0.0-13-g4b2532d and Version: v3.0.1-2-g946fad1 something when wrong and "ustreamer" libraries are not present in my /usr/bin/ directory. Also the log for crowsnest has the following:

crowsnest - A webcam Service for multiple Cams and Stream Services.
[08/20/22 15:25:51] crowsnest: Version: v3.0.1-2-g946fad1
[08/20/22 15:25:51] crowsnest: Prepare Startup ...
[08/20/22 15:25:51] crowsnest: INFO: Host information:
[08/20/22 15:25:51] crowsnest: Host Info: Distribution: Raspbian GNU/Linux 10 (buster)
[08/20/22 15:25:51] crowsnest: Host Info: Release: MainsailOS release 0.7.0 (buster)
[08/20/22 15:25:51] crowsnest: Host Info: Kernel: Linux 5.10.103-v7l+ armv7l
[08/20/22 15:25:52] crowsnest: Host Info: Model: Raspberry Pi 4 Model B Rev 1.5
[08/20/22 15:25:52] crowsnest: Host Info: Available CPU Cores: 4
[08/20/22 15:25:52] crowsnest: Host Info: Available Memory: 1734408 kB
[08/20/22 15:25:52] crowsnest: Host Info: Diskspace (avail. / total): 25G / 30G
[08/20/22 15:25:52] crowsnest: INFO: Checking Dependencys
[08/20/22 15:25:52] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[08/20/22 15:25:52] crowsnest: Dependency: 'find' found in /usr/bin/find.
[08/20/22 15:25:52] crowsnest: Dependency: 'logger' found in /usr/bin/logger.
[08/20/22 15:25:52] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[08/20/22 15:25:52] crowsnest: Dependency: 'ffmpeg' found in /usr/bin/ffmpeg.
[08/20/22 15:25:52] crowsnest: Dependency: 'ustreamer' not found. Exiting!
[08/20/22 15:25:57] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.

And since then, I am not able to see any webcam in mainsail. I guess I could use ffmpeg but I can't find information here in the documentation how to set such feature.

I think if I have a step-by-step instructions in how to re-install crown nest that will make it.

commit a3fba610 (2022-08-12): fix: fix error using installer on armbian

This should fix #28