mainsail-crew/crowsnest #29: Raspi cam V2 not detected with fresh new install..

yannvautrin, 2022-08-13:
### What happened

Hi All,
I just did a fresh install of Mainsail OS with PI Imager and  kiauh.sh, all is ok, but no way to have the Raspi cam working....

When trying the `ls  /dev/v4l/by-id/` command, no file or directory found...

in raspi-config, there is no "Legacy Camera" option...

Distrib version is Raspbian GNU/LINUX 10 (Buster)

### What did you expect to happen

Webcam up and running

### How to reproduce

trying to get camera ID with:

`ls  /dev/v4l/by-id/`

ls: cannot access '/dev/v4l/by-id/': No such file or directory

### Additional information

[08/13/22 08:53:38] crowsnest: Version: v3.0.1
[08/13/22 08:53:38] crowsnest: Prepare Startup ...
[08/13/22 08:53:38] crowsnest: INFO: Checking Dependencys
[08/13/22 08:53:38] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[08/13/22 08:53:38] crowsnest: Dependency: 'find' found in /usr/bin/find.
[08/13/22 08:53:38] crowsnest: Dependency: 'logger' found in /usr/bin/logger.
[08/13/22 08:53:38] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[08/13/22 08:53:38] crowsnest: Dependency: 'ffmpeg' found in /usr/bin/ffmpeg.
[08/13/22 08:53:38] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[08/13/22 08:53:38] crowsnest: Dependency: 'rtsp-simple-server' found in bin/rtsp-simple-server/rtsp-simple-server.
[08/13/22 08:53:38] crowsnest: INFO: Detect available Devices
[08/13/22 08:53:38] crowsnest: ERROR: No usable Devices Found. Stopping crowsnest.

KwadFan, 2022-08-13:
Hey there,
So, first off all in Debian Buster ( Raspberry Pi OS ) is no "legacy camera" this option is available only Bullseye.
It was implemented because they changed the whole camera stack in bullseye to "Libcamera".

Cameras that need support of the new "libcamera" stack wont work either, because they are not supported by ustreamer ( crowsnest backend ).

Could you please give some more details about the used PiCamera? What I can say already that ArduCams (16MP/64MP) will never get supported because they relie on the old mjpg-streamer with "house made" patches, which I will not implement.

Regards

yannvautrin, 2022-08-13:
Hiya,

Thanks for your reply!

Camera is an original Raspberry Pi camera 2.1 8 megapixel connected to a Raspi Pi4B with original Raspi 50cm flat cable.

KwadFan, 2022-08-13:
Could you please share a link?

yannvautrin, 2022-08-13:
[https://www.kubii.fr/home/1653-nouveau-module-camera-v2-640522710881.html?src=raspberrypi](url)

yannvautrin, 2022-08-13:
Just to add more information, camera was working on previous config (Raspi Desktop Debian OS Lite, Legacy camera in raspi-config, Klipper, mainsail, crowsnest, etc...)

KwadFan, 2022-08-13:
Ok, that leaves a trace, did you try to set legacy vie raspi-config? This probably would disable it because its enabled by default.

Could you attach your config and /boot/config.txt please?
Also output of ǜcgencmd get_camera` would help.

This type of cam should work right out of the box

yannvautrin, 2022-08-13:
`vcgencmd get_camera` returns `supported=1 detected=1

and here is my boot.txt`:

`# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

####################################################
####     MainsailOS specific configurations     ####
####################################################
####      DO NOT CHANGE SECTION BELOW !!!       ####
####   UNLESS YOU KNOW WHAT YOU ARE DOING !!!   ####
####################################################

## For more options and information see
## https://www.raspberrypi.com/documentation/computers/config_txt.html
## Some settings may impact device functionality. See link above for details

## For additional information about device filters see
## https://www.raspberrypi.com/documentation/computers/config_txt.html#model-filters

[pi0]
## This affects Pi Zero(W) and Pi Zero2
## Due lag of RAM, limit GPU RAM
gpu_mem=128

[pi2]
gpu_mem=256

[pi3]
## Use 256 if 1Gb Ram Model!
gpu_mem=128
# gpu_mem=256

[pi4]
## Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2
## Do not use more than 256Mb on Pi Model 4, it uses its own Management.
gpu_mem=256

[all]

## SPI Interface is enabled by default for Input Shaper
## To revert that comment out line #48
## This colides with Hyperpixel Display!
## Hyperpixel Screen uses the same Pin for Backlight.

## Enable Hardware UART for Serial Communication
enable_uart=1
dtoverlay=disable-bt

## Enable VideoCore at boot, needed for Raspicams and DSI devices.
start_x=1

### EXPERIMENTAL - Enable 64bit Kernel
### The 64-bit kernel will only work on:
### Raspberry Pi 3, 3+, 4, 400, Zero 2 W and 2B rev 1.2
### and Raspberry Pi Compute Modules 3, 3+ and 4.
# arm_64bit=1

####################################################
`

KwadFan, 2022-08-16:
So, I see no reason why it shouldnt work. Everything is in place where it needs to be and its detected.

Could you please upload crowsnest.log once again?

And to save time please join our discord and feel free to ping me.

KwadFan, 2022-08-19:
I will recieve the same cam as described in the Issue and will do some test.

But, I faced the same Issue on discord and it worked right out of the box.
The Issue was the User tried to enable the cam via raspi-config, which disabled it because the installer enables it by default ;)

So, stay tuned.

KwadFan, 2022-08-20:
After I received a raspicam v2, I tested it with a MainsailOS 0.7.0 Image and without any modification it worked right of the box.
Therefor I close that Issue. It seems you probably misconfigured something or your cam is damaged.

Regards Kwad

closed 2022-08-20 by KwadFan

yannvautrin, 2022-08-21:
I replaced the webcam as it died...the new one works perfectly (same model, same settings!).
Thanks for following up and helping!