mainsail-crew/crowsnest #31: ustreamer won't build on RPI 4 Bullseye 64bit

nofuturekid, 2022-08-16:
### What happened

The "delivered" version of ustreamer is very outdated and will not build on RaspianOS Bullseye.

### What did you expect to happen

make install will run through

### How to reproduce

* freshly installed system
* install klipper/moonraker/mainsail via KIAUH
* clone crowsnest and follow instructions

during the build of ustreamer it fails.

### Additional information

### Workaround (with unclean repository afterwards)

If one manually changes the OMX commit number in "bin/makefile" to a more recent one it builds fine.
ie:

```
#USTREAMER_OMX_BRANCH = 61ab2a8
USTREAMER_OMX_BRANCH = 5502758a7ebdfbc6c8960ea50b26bf808aec75cc
```

KwadFan, 2022-08-16:
I am sorry but the mentioned Issue is not a case, because it says OMX_BRANCH, bullseye doesnt have OMX.
It uses a technic called M2M and thats default, for buster images it pulls the latest branch, only on buster images it resets to the last known version of ustreamer working with buster. And thats 4.13, all versions after have no OMX support.

nofuturekid, 2022-08-16:
Still it tries tries to build the OMX version. So the detection is not working.

KwadFan, 2022-08-16:
This would fix it for bullseye 64 maybe, but what is about buster? Did you try buster with that commit ? Is omx truly working?
I think you should read again how this is used and how OMX dependency is detected.

But for the sake of correctness, i flashed bullseye 64bit sd and test it. But till now it wasnt planned to support 64bit bullseye because the benefits are to small.

nofuturekid, 2022-08-16:
Btw, had the same issues with 32bit Bullseye, can't say if OMX is working, but the build is going throug and afterwards the camera (USB) works.

I checked how the it is detecting that is has to checkout the OMX or the 'normal' version, and it's checking out the version specified in the OMX commit variable.

When it comes to the differences between 32bit or 64bit, if you have Pi4 with 8Gb... 

I highly appreciate your try to reproduce and hopefully fix. If you experience problems in reproducing the issue, I probably can do that tomorrow as it always failed for me.

nofuturekid, 2022-08-16:
I guess it's not using OMX with me overriding the commit variable but just checking out the latest version in general.
If that's the case, the detection which version to checkout needs to be improved, to not falsely using the OMX commit when it's not supported by the OS.

KwadFan, 2022-08-16:
> Btw, had the same issues with 32bit Bullseye, can't say if OMX is working, but the build is going throug and afterwards the camera (USB) works.
> 
> I checked how the it is detecting that is has to checkout the OMX or the 'normal' version, and it's checking out the version specified in the OMX commit variable.
> 
> It's not that complicated: ./bin/Makefile
> 
> ```
> # OpenMAX IL Support
> OMX_SUPPORT = $(shell [ -d /opt/vc/include ] && echo 1 || echo 0)
> ```
> 
> And this is returning "1" and therefore the "OMX" Version is checked out.
> 
> When it comes to the differences between 32bit or 64bit, if you have Pi4 with 8Gb...
> 
> I highly appreciate your try to reproduce and hopefully fix. If you experience problems in reproducing the issue, I probably can do that tomorrow as it always failed for me.

Ok, to clarify that, yes, you found it but its logic is reversed.

If the test fails the or (|| echo 0) comes into play. So it only echos 1 if the test not fails :)

KwadFan, 2022-08-16:
I am wondering about that because then you have that folder taht shouldnt exist ^^

KwadFan, 2022-08-16:
[image]

KwadFan, 2022-08-16:
Thats from the Image I downloaded a couple Minutes ago via RPI Imager,

Does it have any content on your machine?

nofuturekid, 2022-08-16:
[image]

KwadFan, 2022-08-16:
Ok thats not funny. so we have to know where this is comming from.

nofuturekid, 2022-08-16:
`for file in $(find /opt/vc/include); do dpkg-query -S ${file}; done`
Nothing. Always `dpkg-query: no path found matching pattern...`

Can't explain it myself.

KwadFan, 2022-08-16:
I would try to `mv` and delete it, reboot and try to install ^^

nofuturekid, 2022-08-16:
That would for sure fix the build process for Crowsnest - but I'm a little afraid of the other dragons then.

KwadFan, 2022-08-16:
> That would for sure fix the build process for Crowsnest - but I'm a little afraid of the other dragons then.

I understand your concerns, therefor mv to save location. That will give as roo to investigate the root cause if something fails on your side. But, tbh, if a fresh Image doesnt have the folder and your packages are uptodate, what should happen *shrug*

nofuturekid, 2022-08-16:
I will do a new installation, tomorrow and after every step I will check the existance of this folder and it's content then.

KwadFan, 2022-08-16:
> I will do a new installation, tomorrow and after every step I will check the existance of this folder and it's content then.

That would be great :+1:

nofuturekid, 2022-08-16:
> > I will do a new installation, tomorrow and after every step I will check the existance of this folder and it's content then.
> 
> That would be great 👍

Sure. I will help - as it's "my" problem ;) - I will come back to you tomorrow.

nofuturekid, 2022-08-17:
I don't know what to say, but now it worked without a flaw, `/opt` stayed clean.
So I still don't know how I ended up with `/opt/vc/include` populated in all my tries before.
I'm so sorry for wasting your time. Thanks for the help anyway.

BTW the procedure:

- Install RaspbianOS Bullseye Lite 64bit (04.04.2022), with Raspi-Imager
- sudo apt update
- sudo apt upgrade (got new firmware & kernel) - no: raspi-config
- sudo apt install screen mc vim aptitude git -y
- sudo apt autoremove --purge -y
- git clone https://github.com/th33xitus/kiauh.git
- ./kiauh/kiauh.sh (Klipper / Moonraker / Mainsail / Fluidd - NO MJPG-STREAMER)
  - Install Klipper (Python 3)
  - Install Moonraker
  - Install Mainsail
  - Install Fluidd
- Install Crowsnest
- Install Timelapse

KwadFan, 2022-08-19:
You're welcome!

But, in the end you saved me much time to debug. So, win win situation I would say :)

Regards Kwad

closed 2022-08-19 by KwadFan