mainsail-crew/crowsnest #33: Webcam Stream has low quality after clean start.

Dreadnouhgt, 2022-08-28:
### What happened

When i start my pi the camera stream is horrible, a restart of crowsnest fixes the issue

Newest Version of Crowsnest. Camera is connected via the raspberry camera port. 

### What did you expect to happen

Camera working properly right from the start. 

### How to reproduce

Seems like im the only one with that error

KwadFan, 2022-09-04:
Hi,

So, I am now able to reproduce it every time on reboot.
And thanks to lixxbox I might have found a solution.

Please try using
`v4l2ctl: video_bitrate_mode=1,video_bitrate=15000000`
for your raspicam section.

I am already working on a solution to fix that.

closed 2022-09-04 by KwadFan

mhr93, 2022-11-19:
hello,
I'm having the same issue running current version:
[image]

After raspberry reboot, image looks like this:
[image]
After rebooting crowsnest service:
[image]

I have check video_bitrate and video_bitrate_mode are set correctly after executing "blockyfix" function, I have tried to add an "sleep" after this step to make sure there is no some kind of time delay when changing configuration, tried to execute the blockyfix some steps before... and I have no clue why it has this behaivour.

I attach here a debug level log with the full trace from start to end of the service starting after rasberry pi reboot and after crowsnest service reboot (I added comments to point each case). Both cases are almost identical
[pi_and_crowsnest_restart.log](https://github.com/mainsail-crew/crowsnest/files/10045830/pi_and_crowsnest_restart.log)

Thanks

commit dabb66c9 (2022-09-04): fix: fixes blocky view after reboot

This should fix #33

By default a variable bitrate is set, this leads
to a "blocky" view.

blockyfix function sets constant bitrate and a moderate
bitrate if not set by user.