mainsail-crew/crowsnest #34: Ability to set ustreamer host on config file

leandrotoledo, 2022-08-31:
### Is your feature request related to a problem? Please describe

In addition to use the cameras on Mainsail, I also have them set up behind a reverse proxy that is in different server, therefore the ustream instance should accept connections from 0.0.0.0 but currently it's hardcoded to 127.0.0.1 at launch. 

It would be convenient if that's an option on `crowsnest.conf`.

Thoughts?

### Describe the solution you'd like

```
--- a/libs/ustreamer.sh
+++ b/libs/ustreamer.sh
-    start_param=( --host 127.0.0.1 -p "${pt}" )
+    start_param=( --host "${host}" -p "${pt}" )
```

### Describe alternatives you've considered

I have tried to overload ```custom_flags: --host 0.0.0.0``` but it executes with ```ustream ... --host 127.0.0.1 --host 0.0.0.0```, so it's not a very elegant solution.

KwadFan, 2022-09-04:
Hey there,

I know its not an elegant solution. But here is the catch, ustreamer by itself uses 127.0.0.1 if --host is not set.
But if I launch ustreamer without --host set it sometimes ( under which circumstances ever) it refuses to start.

Because we are using nginx as reverse proxy already it is "hardcoded" to use the proxy instead of all "networks".

I am aware of setups they dont use a proxy or running crowsnest in stand alone mode, I think I will implement a solution like "standalone: true" to avoid that duplicate of host address.

Please be patient :)

KwadFan, 2022-11-15:
@leandrotoledo It took a long time but I did it :rofl: 

So, please be kind and test that on your side, use develop branch for that.
If you report success I will merge that asap.

Regards Kwad

closed 2022-11-18 by KwadFan

leandrotoledo, 2023-11-20:
@KwadFan welp, it took me even longer to confirm 🤣, but I had to perform a fresh Klipper installation, and I encountered this issue again. I can now confirm that setting `no_proxy: true` indeed solves the problem as intended. Thank you!

fluxor99232, 2023-12-04:
Is "no_proxy: true" only affecting ustreamer? Would love to be able to use this with camera-streamer! Or am I getting something wrong there? Been pulling my hair out and spent ours trying to figure out why I couldn't access my camera stream for my server on the same network hosting Klipper, Fluidd and Moonraker. (I like to use separate devices for these jobs because the camera and Klipperscreen only need to be running while the printer is on, so my server switches both the printer and the Pi running Crowsnest on/off as needed).

mryel00, 2023-12-04:
@fluxor99232 `no_proxy` is currently only for ustreamer. camera-streamer didn't have such an option and allowed it by default on the master branch. But on the main branch, which is only used on bookworm images, it now restricts it. You should add `--http-listen=0.0.0.0` to your `custom_flags` inside the crowsnest.conf to achieve the same effect as with `no_proxy`.
In future we will ofc add that parameter for camera-streamer, but currently we have to support two different versions that will make this unnecessary "difficult".

fluxor99232, 2023-12-05:
> @fluxor99232 `no_proxy` is currently only for ustreamer. camera-streamer didn't have such an option and allowed it by default on the master branch. But on the main branch, which is only used on [...]

Works flawlessly, thanks!

Slightly unrelated, but since my motivation for asking the question was to give webrtc a try:
Is webrtc not available when using the cameras-streamer mode in combination with the legacy camera workaround for OV5647 and/or klipperscreen setups (both are in use in my setup)?

mryel00, 2023-12-05:
> Is webrtc not available when using the cameras-streamer mode in combination with the legacy camera workaround for OV5647 and/or klipperscreen setups (both are in use in my setup)?

That's right. This is a problem with camera-streamer itself. RTSP won't work too.

The OV5647 workaround is only for fake v1 picams, as they cannot be used with libcamera. I will update the docs in the future to make clear that this is only a problem with fake v1 picams.

Another workaround for klipperscreen is to use some specific device tree overlays, those differ from setup to setup and might not work on all setups. Here a link for an example: https://github.com/ayufan/camera-streamer/issues/21#issuecomment-1294750888
On bullseye you might also need to update the kernel to a newer version.
As you can see this whole topic with touchscreen and libcamera is getting complex pretty fast, that's why I won't add something like that to the docs, as we cannot be 100% sure that it will work for most users and won't break something else.

fluxor99232, 2023-12-05:
@mryel00 I see! Great infos, thank you very much!

commit 8fe11005 (2022-11-15): feat: add no_proxy parameter

    This allows you to run crowsnest without any
    proxy server needed.

    This resolves #34

    Signed-off-by: Stephan Wendel <me@stephanwe.de>