mainsail-crew/crowsnest #35: Pi Cam v2, can't start stream

gregspruce, 2022-09-02:
### What happened

Set up new install of crowsnest, have a Pi cam v2 attached. All I get is the NO SIGNAL error.

Crowsnest config:
[crowsnest]
log_path: ~/klipper_logs/crowsnest.log  # Default logfile in ~/klipper_logs/crowsnest.log
log_level: debug                    # Valid Options are quiet/verbose/debug
delete_log: true                     # Deletes log on every restart, if set to true

[picamv2]
mode: mjpg                           # mjpg/rtsp
port: 8080                              # Port
device: /dev/v4l/by-path/platform-fe801000.csi-video-index0
#device: /dev/video0                     # See Log for available ...
resolution: 1640x1232                     # widthxheight format
max_fps: 15                             # If Hardware Supports this it will be forced, ohterwise ignored/coerced.
#custom_flags:                          # You can run the Stream Services with custom flags.
#v4l2ctl:                               # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.

Error:

[09/02/22 13:52:40] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[09/02/22 13:52:40] crowsnest: Version: v3.0.1-2-g946fad1
[09/02/22 13:52:40] crowsnest: Prepare Startup ...
[09/02/22 13:52:40] crowsnest: INFO: Host information:
[09/02/22 13:52:40] crowsnest: Host Info: Distribution: Raspbian GNU/Linux 11 (bullseye)
[09/02/22 13:52:40] crowsnest: Host Info: Kernel: Linux 5.15.32-v7l+ armv7l
[09/02/22 13:52:40] crowsnest: Host Info: Model: Raspberry Pi 4 Model B Rev 1.4
[09/02/22 13:52:40] crowsnest: Host Info: Available CPU Cores: 4
[09/02/22 13:52:40] crowsnest: Host Info: Available Memory: 8088488 kB
[09/02/22 13:52:40] crowsnest: Host Info: Diskspace (avail. / total): 52G / 59G
[09/02/22 13:52:40] crowsnest: INFO: Checking Dependencys
[09/02/22 13:52:41] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[09/02/22 13:52:41] crowsnest: Dependency: 'find' found in /usr/bin/find.
[09/02/22 13:52:41] crowsnest: Dependency: 'logger' found in /usr/bin/logger.
[09/02/22 13:52:41] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[09/02/22 13:52:41] crowsnest: Dependency: 'ffmpeg' found in /usr/bin/ffmpeg.
[09/02/22 13:52:41] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[09/02/22 13:52:41] crowsnest: Dependency: 'rtsp-simple-server' found in bin/rtsp-simple-server/rtsp-simple-server.
[09/02/22 13:52:41] crowsnest: Version Control: ustreamer is up to date. (v5.21)
[09/02/22 13:52:41] crowsnest: Version Control: rtsp-simple-server is up to date. (v0.19.1)
[09/02/22 13:52:41] crowsnest: Version Control: ffmpeg is up to date. (4.3.4-0+deb11u1+rpt3)
[09/02/22 13:52:42] crowsnest: INFO: Print Configfile: '/home/pi/klipper_config/crowsnest.conf'
[09/02/22 13:52:42] crowsnest: 		[crowsnest]
[09/02/22 13:52:42] crowsnest: 		log_path: ~/klipper_logs/crowsnest.log
[09/02/22 13:52:42] crowsnest: 		log_level: debug
[09/02/22 13:52:42] crowsnest: 		delete_log: true
[09/02/22 13:52:42] crowsnest: 		
[09/02/22 13:52:42] crowsnest: 		[cam 1]
[09/02/22 13:52:42] crowsnest: 		mode: mjpg
[09/02/22 13:52:42] crowsnest: 		port: 8080
[09/02/22 13:52:42] crowsnest: 		device: /dev/v4l/by-path/platform-fe801000.csi-video-index0
[09/02/22 13:52:42] crowsnest: 		resolution: 1640x1232
[09/02/22 13:52:42] crowsnest: 		max_fps: 15
[09/02/22 13:52:42] crowsnest: INFO: Detect available Devices
[09/02/22 13:52:42] crowsnest: INFO: Found 1 total available Device(s)
[09/02/22 13:52:42] crowsnest: INFO: Found 1 available csi device(s)
[09/02/22 13:52:42] crowsnest: /dev/v4l/by-path/platform-fe801000.csi-video-index0 -> /dev/video0
[09/02/22 13:52:42] crowsnest: V4L2 Control:
[09/02/22 13:52:42] crowsnest: No parameters set for [cam 1]. Skipped.
[09/02/22 13:52:42] crowsnest: Try to start configured Cams / Services...
[09/02/22 13:52:43] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue...
[09/02/22 13:52:44] crowsnest: Starting ustreamer with Device /dev/v4l/by-path/platform-fe801000.csi-video-index0 ...
[09/02/22 13:52:44] crowsnest: DEBUG: ustreamer [cam 1]: Parameters: --host 127.0.0.1 -p 8080 -d /dev/v4l/by-path/platform-fe801000.csi-video-index0 --device-timeout=2 -r 1640x1232 -f 15 --allow-origin=* --static /home/pi/crowsnest/ustreamer-www
[09/02/22 13:52:44] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1315.903 main] -- Using internal blank placeholder
[09/02/22 13:52:44] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1315.903 main] -- Enabling HTTP file server: /home/pi/crowsnest/ustreamer-www
[09/02/22 13:52:45] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1315.903 main] -- Listening HTTP on [127.0.0.1]:8080
[09/02/22 13:52:45] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1315.903 stream] -- Using V4L2 device: /dev/v4l/by-path/platform-fe801000.csi-video-index0
[09/02/22 13:52:45] crowsnest: ... Done!
[09/02/22 13:52:45] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1315.904 stream] -- Using desired FPS: 15
[09/02/22 13:52:45] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1315.904 http] -- Starting HTTP eventloop ...
[09/02/22 13:52:45] crowsnest: DEBUG: ustreamer [cam 1]: ================================================================================
[09/02/22 13:52:46] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1315.904 stream] -- Device fd=8 opened
[09/02/22 13:52:46] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1315.904 stream] -- Using input channel: 0
[09/02/22 13:52:46] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1315.904 stream] -- Using resolution: 1640x1232
[09/02/22 13:52:46] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1315.904 stream] -- Using format: YUYV
[09/02/22 13:52:46] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1315.904 stream] -- Querying HW FPS changing is not supported
[09/02/22 13:52:47] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1315.904 stream] -- Using IO method: MMAP
[09/02/22 13:52:47] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1315.915 stream] -- Requested 5 device buffers, got 5
[09/02/22 13:52:47] crowsnest: DEBUG: ustreamer [cam 1]: -- ERROR [1315.915 stream] -- Can't start capturing: Invalid argument
[09/02/22 13:52:47] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1315.924 stream] -- Device fd=8 closed
[09/02/22 13:52:47] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [1315.924 stream] -- Sleeping 1 seconds before new stream init ...
[09/02/22 13:52:47] crowsnest: DEBUG: ustreamer [cam 1]: ================================================================================

### What did you expect to happen

Wnated to start with 1 pi cam, then add a nozzle cam, but pi cam wont work

### How to reproduce

Config and error posted

KwadFan, 2022-09-04:
Hi,

[09/02/22 14:36:33] crowsnest: /dev/v4l/by-path/platform-fe801000.csi-video-index0 -> /dev/video0

indicates that this camera is either not an real v2, might be an Arducam maybe, or you are simply not using the legacy camera stack on bullseye, which is needed by ustreamer(crowsnests backend).

Also as far as I know there is no legacy cam support for bullseye 64bit.

I will keep an eye on it but at this point there is not much I can do.

As a little spoiler, the backend might change very quickly, matter of time :)

gregspruce, 2022-09-05:
Technically, its a pi cam NoIR v2. And at 1 point i was accidentally using an arducam, possible that I pasted the log from the arducam run. But was getting the same errors. Have temporarily stopped trying to get crowsnest working. Especially with the little tease of the backend change :D

closed 2022-09-05 by gregspruce