mainsail-crew/crowsnest #36: Webcam stopped working

Stooovie, 2022-09-07:
### What happened

Webcam stopped working. No changes in config. Webcam works on other machines. 

### What did you expect to happen

Webcam to work. 

### How to reproduce

Plug in a Webcam, no image. 

### Additional information

crowsnest - A webcam Service for multiple Cams and Stream Services.
[09/06/22 08:57:58] crowsnest: Version: v3.0.3
[09/06/22 08:57:58] crowsnest: Prepare Startup ...
[09/06/22 08:57:59] crowsnest: INFO: Host information:
[09/06/22 08:57:59] crowsnest: Host Info: Distribution: Raspbian GNU/Linux 10 (buster)
[09/06/22 08:57:59] crowsnest: Host Info: Release: MainsailOS release 0.7.0 (buster)
[09/06/22 08:57:59] crowsnest: Host Info: Kernel: Linux 5.10.103-v7+ armv7l
[09/06/22 08:57:59] crowsnest: Host Info: Model: Raspberry Pi 3 Model B Rev 1.2
[09/06/22 08:57:59] crowsnest: Host Info: Available CPU Cores: 4
[09/06/22 08:57:59] crowsnest: Host Info: Available Memory: 892584 kB
[09/06/22 08:57:59] crowsnest: Host Info: Diskspace (avail. / total): 2.0G / 7.1G
[09/06/22 08:57:59] crowsnest: INFO: Checking Dependencys
[09/06/22 08:57:59] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[09/06/22 08:57:59] crowsnest: Dependency: 'find' found in /usr/bin/find.
[09/06/22 08:57:59] crowsnest: Dependency: 'logger' found in /usr/bin/logger.
[09/06/22 08:57:59] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[09/06/22 08:57:59] crowsnest: Dependency: 'ffmpeg' found in /usr/bin/ffmpeg.
[09/06/22 08:57:59] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[09/06/22 08:57:59] crowsnest: Dependency: 'rtsp-simple-server' found in bin/rtsp-simple-server/rtsp-simple-server.
[09/06/22 08:57:59] crowsnest: ERROR: Error 127 occured on line 31
[09/06/22 08:57:59] crowsnest: ERROR: Stopping crowsnest.
[09/06/22 08:57:59] crowsnest: Goodbye...
[09/06/22 08:57:59] crowsnest: ERROR: Error 1 occured on line 31
[09/06/22 08:58:00] crowsnest: ERROR: Stopping crowsnest.
[09/06/22 08:58:00] crowsnest: Goodbye...

atomikhawk, 2022-09-07:
Have this exact issue currently. Came home from work and the camera is no longer working.

Stooovie, 2022-09-08:
I fixed mine by reinstalling crowsnest. Ssh to crowsnest folder and run "make install". Boom, fixed.

KwadFan, 2022-09-08:
Hey there,

I have a trace but couldnt confirm this fully right now.

The error occures in the versioncontrol library and at line no 31, it usually tries to determine ustreamers version.
This let me think there is a update that seems to break ustreamer, maybe a dependencie that is crutial for ustreamer.

As the Issue author describes, that a `make install` fixes it, I suggest instead of `make install` use
`make buildclean
make build`

This recompiles ustreamer. That should fix it for now till I could provide a better solution.
Regards Kwad

Stooovie, 2022-09-14:
Happened again, without me touching the OS at all. 
```
make buildclean
make build
```
doesn't fix the issue this time, nor does replugging the cam, shutting down the Rpi host or anything else. [Here's](https://hastebin.com/kehahilodu
) the log. Please help.

KwadFan, 2022-09-30:
I think we fixed it in discord.
Closing for now :)

Regards Kwad

closed 2022-09-30 by KwadFan