mainsail-crew/crowsnest #37: Add support for H264 encoding

thbl, 2022-09-20:
### Is your feature request related to a problem? Please describe

H264 would make stream more smooth and save bandwidth 

### Describe the solution you'd like

Add support for H264 encoding in mainsail / crownsnest

thbl, 2022-09-20:
There is even an integration example in the ustream repo :)
https://github.com/pikvm/ustreamer#janus

KwadFan, 2022-09-30:
Well,

I dont know when it will happen, but I plan to use a new backend in the future.
That will have more then a janus webrtc server ;)

Closing that for now, stay tuned.

ph0tic, 2022-11-09:
just wanted to show a little love for this request. just realized the used c920 avoided the h.264 purge (maybe?) and really would love a framerate higher than a constant... 2-4fps... even if its just 8, or even 6. or 5.

NonaSuomy, 2023-02-18:
Does this work now with the new backend?

meteyou, 2023-02-19:
not really... we tested https://github.com/ayufan/camera-streamer but it doesn't work really stable (mainsail PR for the client: https://github.com/mainsail-crew/mainsail/pull/1275). we also implemented a HLS client for the RTSP mode in crowsnest. this works fine, but it has a really long delay (too long for my taste)

NonaSuomy, 2023-02-20:
Maybe try using the stuff that PiKVM built. I believe all his stuff is customized for low latency feeds.

meteyou, 2023-02-20:
@NonaSuomy i tested this one from PiKVM: https://github.com/pikvm/ustreamer/blob/master/docs/h264.md

But it looks like this kills the snapshot URL. We need a solution to have a low-latency stream + snapshots for timelapse.

NonaSuomy, 2023-02-20:
Can you explain how you tested it? If so then I will set it up the same then ask him for assistance on how to restore a snapshot url as well?

NonaSuomy, 2023-02-20:
[image]

meteyou, 2023-02-20:
I followed the guide from pikvm and then started the webcam with it...

closed 2023-05-22 by KwadFan