mainsail-crew/crowsnest #38: make install fail

shortbus, 2022-09-20:
### What happened

I followed the instructions on the github.

uname -ra
Linux Hostname 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux

cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

I know it says Debian, but it was a raspi install. I used Raspberry Pi Imager and chose Raspberry Pi OS Lite (64 bit).

### What did you expect to happen

I expected the compile to work and "make install" to finish.

### How to reproduce

cd ~
git clone https://github.com/mainsail-crew/crowsnest.git
cd ~/crowsnest
make install

### Additional information

my output:
cd ~
git clone https://github.com/mainsail-crew/crowsnest.git
cd ~/crowsnest
make install

crowsnest - A Webcam Daemon for Raspberry Pi OS

	Some Parts of the Installer requires 'root' privileges.
	You will be prompted for your 'sudo' password, if needed.

Running apt update first ...
Hit:1 http://security.debian.org/debian-security bullseye-security InRelease
Hit:2 http://deb.debian.org/debian bullseye InRelease                            
Hit:3 http://deb.debian.org/debian bullseye-updates InRelease                    
Hit:4 http://archive.raspberrypi.org/debian bullseye InRelease                   
Reading package lists... Done                              
Building dependency tree... Done
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.

Install crowsnest Service ...
Installing 'crowsnest' Dependencies ...

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Installing 'crowsnest' Dependencies ... [OK]
Linking crowsnest ... [OK]
Copying crowsnest.service file ... [OK]
Linking logrotate file ... [OK]
Enable crowsnest.service on boot ... [OK]
Do you want to add [update_manager] entry? (y/N) N
Adding Crowsnest Update Manager entry to moonraker.conf ... [SKIPPED]
Add User pi to group 'video' ... [SKIPPED]
==> User pi is already in group 'video'
Build dependend Stream Apps ...
Cloning ustreamer repository ...
Cloning into 'bin/ustreamer'...
remote: Enumerating objects: 8383, done.
remote: Counting objects: 100% (1263/1263), done.
remote: Compressing objects: 100% (533/533), done.
remote: Total 8383 (delta 824), reused 1081 (delta 697), pack-reused 7120
Receiving objects: 100% (8383/8383), 5.29 MiB | 4.12 MiB/s, done.
Resolving deltas: 100% (5684/5684), done.
Installing 'ustreamer' Dependencies ...

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Installing 'ustreamer' Dependencies ... [OK]
make[1]: Entering directory '/home/pi/crowsnest/bin'
INFO: ustreamer found.
make ustreamer-bin
make[2]: Entering directory '/home/pi/crowsnest/bin'
make[2]: warning: -j4 forced in makefile: resetting jobserver mode.
Compiling ustreamer with OMX Support.
Changening to commit '61ab2a8' 
WITH_OMX=1 make -C ustreamer
make[3]: Entering directory '/home/pi/crowsnest/bin/ustreamer'
make apps
make[4]: Entering directory '/home/pi/crowsnest/bin/ustreamer'
make -C src
make[5]: Entering directory '/home/pi/crowsnest/bin/ustreamer/src'
-- CC libs/base64.c
-- CC libs/frame.c
-- CC libs/logging.c
-- CC libs/memsink.c
-- CC libs/options.c
-- CC libs/unjpeg.c
-- CC ustreamer/blank.c
-- CC ustreamer/data/blank_jpeg.c
-- CC ustreamer/data/index_html.c
-- CC ustreamer/device.c
-- CC ustreamer/encoder.c
-- CC ustreamer/encoders/cpu/encoder.c
-- CC ustreamer/encoders/hw/encoder.c
-- CC ustreamer/http/bev.c
-- CC ustreamer/http/mime.c
-- CC ustreamer/http/path.c
-- CC ustreamer/http/server.c
-- CC ustreamer/http/static.c
-- CC ustreamer/http/unix.c
-- CC ustreamer/http/uri.c
-- CC ustreamer/main.c
-- CC ustreamer/options.c
-- CC ustreamer/stream.c
-- CC ustreamer/workers.c
-- CC ustreamer/encoders/omx/component.c
-- CC ustreamer/encoders/omx/encoder.c
-- CC ustreamer/encoders/omx/formatters.c
-- CC ustreamer/encoders/omx/vcos.c
-- CC ustreamer/h264/encoder.c
-- CC ustreamer/h264/stream.c
ustreamer/h264/encoder.c: In function ‘_h264_encoder_compress_raw’:
ustreamer/h264/encoder.c:326:16: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
  326 |     in->data = (uint8_t *)vcsm_vc_hdl_from_hdl(src_vcsm_handle);
      |                ^
-- CC dump/file.c
-- CC dump/main.c
== LD ustreamer-dump.bin
== LD ustreamer.bin
/usr/bin/ld: skipping incompatible /opt/vc/lib/libbcm_host.so when searching for -lbcm_host
/usr/bin/ld: skipping incompatible /opt/vc/lib/libvcos.so when searching for -lvcos
/usr/bin/ld: skipping incompatible /opt/vc/lib/libvcsm.so when searching for -lvcsm
/usr/bin/ld: cannot find -lvcsm
/usr/bin/ld: skipping incompatible /opt/vc/lib/libvcsm.so when searching for -lvcsm
/usr/bin/ld: skipping incompatible /opt/vc/lib/libopenmaxil.so when searching for -lopenmaxil
/usr/bin/ld: cannot find -lopenmaxil
/usr/bin/ld: skipping incompatible /opt/vc/lib/libopenmaxil.so when searching for -lopenmaxil
/usr/bin/ld: skipping incompatible /opt/vc/lib/libmmal.so when searching for -lmmal
/usr/bin/ld: cannot find -lmmal
/usr/bin/ld: skipping incompatible /opt/vc/lib/libmmal.so when searching for -lmmal
/usr/bin/ld: skipping incompatible /opt/vc/lib/libmmal_core.so when searching for -lmmal_core
/usr/bin/ld: cannot find -lmmal_core
/usr/bin/ld: skipping incompatible /opt/vc/lib/libmmal_core.so when searching for -lmmal_core
/usr/bin/ld: skipping incompatible /opt/vc/lib/libmmal_util.so when searching for -lmmal_util
/usr/bin/ld: cannot find -lmmal_util
/usr/bin/ld: skipping incompatible /opt/vc/lib/libmmal_util.so when searching for -lmmal_util
/usr/bin/ld: skipping incompatible /opt/vc/lib/libmmal_vc_client.so when searching for -lmmal_vc_client
/usr/bin/ld: cannot find -lmmal_vc_client
/usr/bin/ld: skipping incompatible /opt/vc/lib/libmmal_vc_client.so when searching for -lmmal_vc_client
/usr/bin/ld: skipping incompatible /opt/vc/lib/libmmal_components.so when searching for -lmmal_components
/usr/bin/ld: cannot find -lmmal_components
/usr/bin/ld: skipping incompatible /opt/vc/lib/libmmal_components.so when searching for -lmmal_components
collect2: error: ld returned 1 exit status
make[5]: *** [Makefile:101: ustreamer.bin] Error 1
make[5]: Leaving directory '/home/pi/crowsnest/bin/ustreamer/src'
make[4]: *** [Makefile:38: apps] Error 2
make[4]: Leaving directory '/home/pi/crowsnest/bin/ustreamer'
make[3]: *** [Makefile:28: all] Error 2
make[3]: Leaving directory '/home/pi/crowsnest/bin/ustreamer'
make[2]: *** [Makefile:52: ustreamer-bin] Error 2
make[2]: Leaving directory '/home/pi/crowsnest/bin'
make[1]: *** [Makefile:42: all] Error 2
make[1]: Leaving directory '/home/pi/crowsnest/bin'
ERROR: Error 2 occured on line 393
ERROR: Stopping install.sh.
Goodbye...
make: *** [Makefile:35: install] Error 1

shortbus, 2022-09-20:
after changing github filters I saw issue #31
https://github.com/mainsail-crew/crowsnest/issues/31

I did tath and I compiled.

KwadFan, 2022-09-30:
Well, I will check if there is an error in my scripts with the latest Version of rpi 64 bit OS, but we started to transition to bullseye and it runs flawless till now, at least at image creation.