mainsail-crew/crowsnest #39: Ustreamer fails to start automatically

sshantt, 2022-09-23:
### What happened

Ustreamer fails to start automatically

```
[09/23/22 10:04:46] crowsnest: INFO: No usable CSI Devices found.
[09/23/22 10:04:46] crowsnest: V4L2 Control:
[09/23/22 10:04:46] crowsnest: No parameters set for [cam 1]. Skipped.
[09/23/22 10:04:47] crowsnest: INFO: Blockyfix: Setting video_bitrate_mode to constant.
[09/23/22 10:04:47] crowsnest: Try to start configured Cams / Services...
[09/23/22 10:04:48] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue...
[09/23/22 10:04:48] crowsnest: Starting ustreamer with Device /dev/video0 ...
[09/23/22 10:04:49] crowsnest: ERROR: Start of ustreamer [cam 1] failed!
[09/23/22 10:04:50] crowsnest: ... Done!
```

### What did you expect to happen

Stream to start automatically

### How to reproduce

Install crowsnest, reboot

### Additional information

Running ustreamer from the bin directory manually works just fine 

```
ustreamer git:(master) ./ustreamer --device=/dev/video0 -r 1920x1080 --device-timeout=8 --quality 100 --format=MJPEG --host=0.0.0.0 --port=8080 -f 30
-- INFO  [327.915      main] -- Using internal blank placeholder
-- INFO  [327.916      main] -- Listening HTTP on [0.0.0.0]:8080
-- INFO  [327.916    stream] -- Using V4L2 device: /dev/video0
-- INFO  [327.916    stream] -- Using desired FPS: 30
-- INFO  [327.917      http] -- Starting HTTP eventloop ...
================================================================================
-- INFO  [327.918    stream] -- Device fd=8 opened
-- INFO  [327.918    stream] -- Using input channel: 0
-- INFO  [327.919    stream] -- Using resolution: 1920x1080
-- INFO  [327.920    stream] -- Using pixelformat: MJPEG
-- INFO  [327.920    stream] -- Using HW FPS: 30
-- ERROR [327.920    stream] -- Device does not support setting of HW encoding quality parameters
-- INFO  [327.920    stream] -- Using IO method: MMAP
-- INFO  [327.933    stream] -- Requested 5 device buffers, got 5
-- INFO  [329.086    stream] -- Capturing started
-- INFO  [329.087    stream] -- Switching to HW encoder: the input is (M)JPEG ...
-- INFO  [329.087    stream] -- Using JPEG quality: encoder default
-- INFO  [329.087    stream] -- Creating pool JPEG with 1 workers ...
-- INFO  [329.087    stream] -- Capturing ...
```

KwadFan, 2022-09-30:
Could you please attach the full log, or better set to `log_level: debug` and try to reproduce?

Regards Kwad

sshantt, 2022-10-05:
```
[10/04/22 08:01:41] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[10/04/22 08:01:42] crowsnest: Version: v3.0.3-1-gee59a12
[10/04/22 08:01:42] crowsnest: Prepare Startup ...
[10/04/22 08:01:42] crowsnest: INFO: Host information:
[10/04/22 08:01:42] crowsnest: Host Info: Distribution: Raspbian GNU/Linux 10 (buster)
[10/04/22 08:01:42] crowsnest: Host Info: Release: MainsailOS release 0.6.1 (buster)
[10/04/22 08:01:42] crowsnest: Host Info: Kernel: Linux 5.15.56-v7+ armv7l
[10/04/22 08:01:42] crowsnest: Host Info: Model: Raspberry Pi 3 Model B Rev 1.2
[10/04/22 08:01:42] crowsnest: Host Info: Available CPU Cores: 4
[10/04/22 08:01:42] crowsnest: Host Info: Available Memory: 761956 kB
[10/04/22 08:01:42] crowsnest: Host Info: Diskspace (avail. / total): 106G / 118G
[10/04/22 08:01:42] crowsnest: INFO: Checking Dependencys
[10/04/22 08:01:43] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[10/04/22 08:01:43] crowsnest: Dependency: 'find' found in /usr/bin/find.
[10/04/22 08:01:43] crowsnest: Dependency: 'logger' found in /usr/bin/logger.
[10/04/22 08:01:43] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[10/04/22 08:01:43] crowsnest: Dependency: 'ffmpeg' found in /usr/bin/ffmpeg.
[10/04/22 08:01:43] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[10/04/22 08:01:43] crowsnest: Dependency: 'rtsp-simple-server' found in bin/rtsp-simple-server/rtsp-simple-server.
[10/04/22 08:01:44] crowsnest: Version Control: ustreamer is up to date. (v4.13)
[10/04/22 08:01:45] crowsnest: Version Control: rtsp-simple-server is up to date. (v0.19.1)
[10/04/22 08:01:48] crowsnest: Version Control: ffmpeg is up to date. (4.1.9-0+deb10u1+rpt1)
[10/04/22 08:01:48] crowsnest: INFO: Print Configfile: '/home/pi/klipper_config/crowsnest.conf'
[10/04/22 08:01:48] crowsnest: 		[crowsnest]
[10/04/22 08:01:48] crowsnest: 		log_path: ~/klipper_logs/crowsnest.log
[10/04/22 08:01:48] crowsnest: 		log_level: verbose
[10/04/22 08:01:48] crowsnest: 		delete_log: true
[10/04/22 08:01:48] crowsnest: 		
[10/04/22 08:01:48] crowsnest: 		[cam 1]
[10/04/22 08:01:48] crowsnest: 		mode: mjpg
[10/04/22 08:01:48] crowsnest: 		port: 8080
[10/04/22 08:01:49] crowsnest: 		device: /dev/video0
[10/04/22 08:01:49] crowsnest: 		resolution: 1920 x 1080
[10/04/22 08:01:49] crowsnest: 		max_fps: 30
[10/04/22 08:01:49] crowsnest: INFO: Detect available Devices
[10/04/22 08:01:49] crowsnest: INFO: Found 1 total available Device(s)
[10/04/22 08:01:49] crowsnest: Detected 'Raspicam' Device -> /dev/video0
[10/04/22 08:01:49] crowsnest: Supported Formats:
[10/04/22 08:01:49] crowsnest: 		[0]: 'YU12' (Planar YUV 4:2:0)
[10/04/22 08:01:49] crowsnest: 		Size: Stepwise 32x32 - 4056x3040 with step 2/2
[10/04/22 08:01:49] crowsnest: 		[1]: 'YUYV' (YUYV 4:2:2)
[10/04/22 08:01:49] crowsnest: 		Size: Stepwise 32x32 - 4056x3040 with step 2/2
[10/04/22 08:01:49] crowsnest: 		[2]: 'RGB3' (24-bit RGB 8-8-8)
[10/04/22 08:01:49] crowsnest: 		Size: Stepwise 32x32 - 4056x3040 with step 2/2
[10/04/22 08:01:49] crowsnest: 		[3]: 'JPEG' (JFIF JPEG, compressed)
[10/04/22 08:01:49] crowsnest: 		Size: Stepwise 32x32 - 4056x3040 with step 2/2
[10/04/22 08:01:49] crowsnest: 		[4]: 'H264' (H.264, compressed)
[10/04/22 08:01:49] crowsnest: 		Size: Stepwise 32x32 - 4056x3040 with step 2/2
[10/04/22 08:01:49] crowsnest: 		[5]: 'MJPG' (Motion-JPEG, compressed)
[10/04/22 08:01:49] crowsnest: 		Size: Stepwise 32x32 - 4056x3040 with step 2/2
[10/04/22 08:01:50] crowsnest: 		[6]: 'YVYU' (YVYU 4:2:2)
[10/04/22 08:01:50] crowsnest: 		Size: Stepwise 32x32 - 4056x3040 with step 2/2
[10/04/22 08:01:50] crowsnest: 		[7]: 'VYUY' (VYUY 4:2:2)
[10/04/22 08:01:50] crowsnest: 		Size: Stepwise 32x32 - 4056x3040 with step 2/2
[10/04/22 08:01:50] crowsnest: 		[8]: 'UYVY' (UYVY 4:2:2)
[10/04/22 08:01:50] crowsnest: 		Size: Stepwise 32x32 - 4056x3040 with step 2/2
[10/04/22 08:01:50] crowsnest: 		[9]: 'NV12' (Y/CbCr 4:2:0)
[10/04/22 08:01:50] crowsnest: 		Size: Stepwise 32x32 - 4056x3040 with step 2/2
[10/04/22 08:01:50] crowsnest: 		[10]: 'BGR3' (24-bit BGR 8-8-8)
[10/04/22 08:01:50] crowsnest: 		Size: Stepwise 32x32 - 4056x3040 with step 2/2
[10/04/22 08:01:50] crowsnest: 		[11]: 'YV12' (Planar YVU 4:2:0)
[10/04/22 08:01:50] crowsnest: 		Size: Stepwise 32x32 - 4056x3040 with step 2/2
[10/04/22 08:01:50] crowsnest: 		[12]: 'NV21' (Y/CrCb 4:2:0)
[10/04/22 08:01:50] crowsnest: 		Size: Stepwise 32x32 - 4056x3040 with step 2/2
[10/04/22 08:01:50] crowsnest: 		[13]: 'RX24' (32-bit XBGR 8-8-8-8)
[10/04/22 08:01:50] crowsnest: 		Size: Stepwise 32x32 - 4056x3040 with step 2/2
[10/04/22 08:01:50] crowsnest: Supported Controls:
[10/04/22 08:01:50] crowsnest: 		
[10/04/22 08:01:50] crowsnest: 		User Controls
[... 120 lines cut ...]
[10/04/22 08:01:55] crowsnest: 		1: Auto
[10/04/22 08:01:55] crowsnest: 		exposure_metering_mode 0x009a0919 (menu) : min=0 max=3 default=0 value=0
[10/04/22 08:01:55] crowsnest: 		0: Average
[10/04/22 08:01:55] crowsnest: 		1: Center Weighted
[10/04/22 08:01:55] crowsnest: 		2: Spot
[10/04/22 08:01:55] crowsnest: 		3: Matrix
[10/04/22 08:01:55] crowsnest: 		scene_mode 0x009a091a (menu) : min=0 max=13 default=0 value=0
[10/04/22 08:01:55] crowsnest: 		0: None
[10/04/22 08:01:55] crowsnest: 		8: Night
[10/04/22 08:01:55] crowsnest: 		11: Sports
[10/04/22 08:01:55] crowsnest: 		
[10/04/22 08:01:55] crowsnest: 		JPEG Compression Controls
[10/04/22 08:01:55] crowsnest: 		
[... cut ...]
[10/04/22 08:01:56] crowsnest: No parameters set for [cam 1]. Skipped.
[10/04/22 08:01:57] crowsnest: INFO: Blockyfix: Setting video_bitrate_mode to constant.
[10/04/22 08:01:57] crowsnest: Try to start configured Cams / Services...
[10/04/22 08:01:57] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue...
[10/04/22 08:01:58] crowsnest: Starting ustreamer with Device /dev/video0 ...
[10/04/22 08:01:59] crowsnest: ERROR: Start of ustreamer [cam 1] failed!
[10/04/22 08:01:59] crowsnest: ... Done!

```

KwadFan, 2022-10-07:
Ok, I dont know why, it is doing that strange thing, tbh.

But, I guess it might be the device Timeout. By default I use 5 as value.
See https://github.com/mainsail-crew/crowsnest/blob/ee59a12d59cb2db173a93b266c4766b853bd0364/libs/ustreamer.sh#L40

Maybe that kind of cam needs a bit more. Try
`custom_flags: --device-timeout=8` and see what it does...
If it then starts correctly keep it :) I'll do my tests on a pi 4 4gb. maybe therefor the initialize phase is a bit shorter.
If that fixes it I will bump it up.

sshantt, 2022-10-09:
Resolution input has whitespaces before and after the "x". Thats what was causing the issue

closed 2022-10-09 by sshantt