mainsail-crew/crowsnest #40: Add MainsailOS 0.7.0 Support

emersonsc, 2022-09-26:
### Is your feature request related to a problem? Please describe

No

### Describe the solution you'd like

I have tested crowsnest on MainsailOS 0.7.0 for several weeks now with no issues or bugs.

meteyou, 2022-09-26:
is this a feature request or just a feedback?

sry... I do not understand what is meant by this issue.

emersonsc, 2022-09-26:
It's labeled as a feature request. I was trying to do a pull request but couldn't figure out how to do it. It says to do a pull request if you've installed this on other OS and found it to work so that's what I'm trying to do.

emersonsc, 2022-09-26:
From the crowsnest readme file:

"This is not tested on other Distributions. If you test that on other Distributions,
feel free to open a Pull Request to enhance Documentation."

KwadFan, 2022-09-30:
Will add tested Distros in the future to Readme,

So far, tested with
- Debian 10 VM (x86)
- Debian 11 VM (x86)
- Debian 10/11 Bare Metal (amd64)
- FluiddPi
- Ubuntu 20.04 armv8
- Ubuntu 22.04 (amd64)
- MainsailOS (0.5.0 - latest)