mainsail-crew/crowsnest #41: WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

theschles, 2022-10-20:
### What happened

Fresh install of crowsnest on nearly-fresh install of Mainsail (which was freshly installed on a freshly-built RaspbianOS Lite 64-bit Bullseye on a Raspberry Pi 4B)

```
➜ make install
crowsnest - A Webcam Daemon for Raspberry Pi OS

	Some Parts of the Installer requires 'root' privileges.
	You will be prompted for your 'sudo' password, if needed.

Running apt update first ...
Hit:1 http://deb.debian.org/debian bullseye InRelease
Get:2 http://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
Hit:3 http://security.debian.org/debian-security bullseye-security InRelease
Hit:4 http://archive.raspberrypi.org/debian bullseye InRelease
Fetched 44.1 kB in 1s (39.3 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.

Install crowsnest Service ...
Installing 'crowsnest' Dependencies ...

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Scanning processes...
Scanning processor microcode...
Scanning linux images...
```

Note the `WARNING: apt does not have a stable CLI interface. Use with caution in scripts.`

That happens later on in the same script as well

### What did you expect to happen

No warning because whatever is trying to be grep'd / awk'd / sed'd / etc in the script out of `apt` should be retrieved using an appropriate method

### How to reproduce

1. clone `crowsnest` afresh
2. cd to the directory
3. `make install`
4. Watch same warning appear over and over again

Apexpredation, 2022-10-21:
Not sure if this is the proper way to do this, but I wanted to also report I am getting this warning. Running clean install on Debian11 bullseye minimal kernel5.16. I have read it could be that just, "apt" was used in a script instead of, "apt-git" and its just a warning because using only apt will make progress bars, and that could cause errors in some scripts.

theschles, 2022-10-21:
> Not sure if this is the proper way to do this, but I wanted to also report I am getting this warning. Running clean install on Debian11 bullseye minimal kernel5.16. I have read it could be that [...]

So a few options come to mind:
1. As suggested by @Apexpredation , switch to `apt-get` with `-qq` option as that's going to reduce the output considerably.  According to the manual for `apt`, `apt` is intended to be interactive.  The goal in the script is non-interactivity.  If you're not already doing it, the script will need to ask for the user's password through `sudo` earlier in the script...
2. use `apt ... > /dev/null 2>&1` to force STDERR to STDOUT...which then is dumped to /dev/null

KwadFan, 2022-10-22:
How about simply ignoring a Warning as long it works as intended?

LordGaav, 2022-10-22:
How about just using apt-get? There is zero reason to use apt in a script. It is literally a wrapper around apt-get / apt-cache / dpkg with a slightly nicer interface, which you don't need in a script.

theschles, 2022-10-22:
@KwadFan / @lixxbox  / @meteyou  / @Robnex  would one of you please review and approve https://github.com/mainsail-crew/crowsnest/pull/43 ?

Thanks.

Robnex, 2022-10-23:
Hello,
check tomorrow with following steps:

- new install of RaspbianOS Lite 64-bit
- Kiauh: Install klipper + moonraker + mainsail
- git clone https://github.com/mainsail-crew/crowsnest.git
- cd ~/crowsnest
- make install

Also get the message "WARNING: apt does not have a stable CLI interface. Use with caution in scripts"

`
Install crowsnest Service ...
Installing 'crowsnest' Dependencies ...

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Extracting templates from packages: 100%
Installing 'crowsnest' Dependencies ... [OK]

`

KwadFan, 2022-10-23:
> @KwadFan / @lixxbox / @meteyou / @Robnex would one of you please review and approve #43 so this issue can be resolved?
> 
> Thanks.

This wouldnt make it any faster! And please do not involve any of the listed contributors.
For short, I am the Main maintainer and "Founder" and I take action if I decide to.

In general help from Contributors is highly appriciated, but this PR comes to a time where to many parts are moving these days.

I am open to your suggestion and I might sound rude with my first question but I dont like ranting here.
So, I am not dumb I know the downsides of using apt in scripts. But I would welcome it a bit more if there where an ask if it is mayyyyybbeeee intended, instead of bashing while I shouldnt use this or that...

To be fair, yes, I planned already to switch to apt-get, as done today in sonar ( my other project here ).
The reason why I am not merge your PR is simple, due the changes that are done in 
https://github.com/Arksine/moonraker/issues/516
I have to rethink and rebuild the whole installation procedure. Why?
I have to ensure that "old" path and "new" file structure will work.

Also I planned a long time ago to switch the "backend" to a, imho, better alternative called camera-streamer.
But, as life goes, I was lacking of time. But I already started on a unpublished branch.

Therefor I am sorry to refuse your PR, to any other time PR's by enthusiasts like you are highly welcome and appriciated.
Thank you for your effort.

As last but not least, I dont know exactly if I had it documented well but please open PRs against the develop branch!
This branch has to pass a couple test from Team Members before they get into the "wild" (master branch).

I let this issue open that you can track if I had made changes, I will quote it in my commits.

Regards Kwad

theschles, 2022-10-24:
Thank you @KwadFan  for the explanation.

KwadFan, 2022-11-01:
@theschles 

I would aprriciate it if you test the branch mentioned in #41

Let me know your opinion please.

Regards Kwad

theschles, 2022-11-06:
Hi @KwadFan just saw this.  In the middle of a print job at the moment.  Will try sometime this week.  Thank you for continuing to work on this issue!

KwadFan, 2022-11-06:
You can use the develop branch. I merged it already there. Will change the behavior slightly. New users are overwhelmed by make config, so will make that optional.

commit d4a2c081 (2022-11-04): refactor: implements new install mechanism

This installer requires to be configured due
make config

Changes:

Add tools/configure.sh script and make config.

Refactor of uninstall.sh according to new installer

Add dev-helper.sh to provide various informations for developing
and also could check existing installation as kinda "unit-test"

This should also fix #46
and fix #41
closed 2022-11-15 by KwadFan