mainsail-crew/crowsnest #42: 64-bit RaspbianOS Bullseye - make install - ustreamer/device.h:44:11: fatal error: interface/vcsm/user-vcsm.h: No such file or directory

theschles, 2022-10-20:
### What happened

This is related to #31 -- however my error is different.

Fresh install of RaspbianOS Bullseye 64-bit lite
Fresh install of KIAUH, Moonraker, Mainsail

Had installed MJPG-Streamer via KIAUH, but it wouldn't compile `input_raspicam.so` as much as I tried.  One of the things I had tried was a tip to symlink library directories:
```
sudo ln -s /usr/include /opt/vc/
sudo ln -s /usr/lib/aarch64-linux-gnu /opt/vc/lib
```

So I gave up on MJPG-Streamer and uninstalled it via KIAUH, then cloned this morning `crowsnest`

Build fails with:
```
-- CC ustreamer/encoder.c
In file included from ustreamer/device.c:23:
ustreamer/device.h:44:11: fatal error: interface/vcsm/user-vcsm.h: No such file or directory
   44 | # include <interface/vcsm/user-vcsm.h>
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
-- CC ustreamer/encoders/cpu/encoder.c
In file included from ustreamer/encoder.h:38,
                 from ustreamer/encoder.c:23:
ustreamer/device.h:44:11: fatal error: interface/vcsm/user-vcsm.h: No such file or directory
   44 | # include <interface/vcsm/user-vcsm.h>
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

I will note that #31  stated that -- for him/her -- the build succeeded if one changes the USTREAMER_OMX_BRANCH to a newer branch.  I can confirm that is true in my case as well.

> If one changes the ustreamer OMX branch commit number in "bin/makefile" to a more recent one it builds fine.
> ie:
```
#USTREAMER_OMX_BRANCH = 61ab2a8
USTREAMER_OMX_BRANCH = 5502758a7ebdfbc6c8960ea50b26bf808aec75cc
```
>

HELP?

My full `make install`:
```
➜ make install
crowsnest - A Webcam Daemon for Raspberry Pi OS

	Some Parts of the Installer requires 'root' privileges.
	You will be prompted for your 'sudo' password, if needed.

Running apt update first ...
Hit:1 http://deb.debian.org/debian bullseye InRelease
Hit:2 http://security.debian.org/debian-security bullseye-security InRelease
Hit:3 http://deb.debian.org/debian bullseye-updates InRelease
Hit:4 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.

Install crowsnest Service ...
Installing 'crowsnest' Dependencies ...

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Installing 'crowsnest' Dependencies ... [OK]
Linking crowsnest ... [OK]
Copying crowsnest.service file ... [OK]
Linking logrotate file ... [OK]
Enable crowsnest.service on boot ... [OK]
Do you want to add [update_manager] entry? (y/N) y
Update Manager entry already exists moonraker.conf ... [SKIPPED]
Add User phil to group 'video' ... [SKIPPED]
==> User phil is already in group 'video'
Build dependend Stream Apps ...
Cloning ustreamer repository ...
Cloning into 'bin/ustreamer'...
remote: Enumerating objects: 8478, done.
remote: Total 8478 (delta 0), reused 0 (delta 0), pack-reused 8478
Receiving objects: 100% (8478/8478), 5.28 MiB | 6.10 MiB/s, done.
Resolving deltas: 100% (5737/5737), done.
Installing 'ustreamer' Dependencies ...

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Installing 'ustreamer' Dependencies ... [OK]
make[1]: Entering directory '/home/phil/crowsnest/bin'
INFO: ustreamer found.
make ustreamer-bin
make[2]: Entering directory '/home/phil/crowsnest/bin'
make[2]: warning: -j4 forced in makefile: resetting jobserver mode.
Compiling ustreamer with OMX Support.
Changening to commit '61ab2a8'
WITH_OMX=1 make -C ustreamer
make[3]: Entering directory '/home/phil/crowsnest/bin/ustreamer'
make apps
make[4]: Entering directory '/home/phil/crowsnest/bin/ustreamer'
make -C src
make[5]: Entering directory '/home/phil/crowsnest/bin/ustreamer/src'
-- CC libs/base64.c
-- CC libs/frame.c
-- CC libs/logging.c
-- CC libs/memsink.c
-- CC libs/options.c
-- CC libs/unjpeg.c
-- CC ustreamer/blank.c
-- CC ustreamer/data/blank_jpeg.c
-- CC ustreamer/data/index_html.c
-- CC ustreamer/device.c
-- CC ustreamer/encoder.c
In file included from ustreamer/device.c:23:
ustreamer/device.h:44:11: fatal error: interface/vcsm/user-vcsm.h: No such file or directory
   44 | # include <interface/vcsm/user-vcsm.h>
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
-- CC ustreamer/encoders/cpu/encoder.c
In file included from ustreamer/encoder.h:38,
                 from ustreamer/encoder.c:23:
ustreamer/device.h:44:11: fatal error: interface/vcsm/user-vcsm.h: No such file or directory
   44 | # include <interface/vcsm/user-vcsm.h>
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
-- CC ustreamer/encoders/hw/encoder.c
make[5]: *** [Makefile:112: build/ustreamer/device.o] Error 1
make[5]: *** Waiting for unfinished jobs....
make[5]: *** [Makefile:112: build/ustreamer/encoder.o] Error 1
make[5]: Leaving directory '/home/phil/crowsnest/bin/ustreamer/src'
make[4]: *** [Makefile:38: apps] Error 2
make[4]: Leaving directory '/home/phil/crowsnest/bin/ustreamer'
make[3]: *** [Makefile:28: all] Error 2
make[3]: Leaving directory '/home/phil/crowsnest/bin/ustreamer'
make[2]: *** [Makefile:52: ustreamer-bin] Error 2
make[2]: Leaving directory '/home/phil/crowsnest/bin'
make[1]: *** [Makefile:42: all] Error 2
make[1]: Leaving directory '/home/phil/crowsnest/bin'
ERROR: Error 2 occured on line 393
ERROR: Stopping install.sh.
Goodbye...
make: *** [Makefile:35: install] Error 1
```

### What did you expect to happen

For it to build and install without me editing installation files

### How to reproduce

1. Make new RaspbianOS 64-bit Bullseye Lite instance
2. Install via KIAUH Klipper, Moonraker, and Mainsail
3. `sudo ln -s /usr/include /opt/vc/`
4. `sudo ln -s /usr/lib/aarch64-linux-gnu /opt/vc/lib`
5. Clone crowsnest afresh
6. Don't modify it
7. `make install`
8. See errors
9. Modify install files: change line to `USTREAMER_OMX_BRANCH = 5502758a7ebdfbc6c8960ea50b26bf808aec75cc`
10. `make buildclean`
11. `make install`
12. Works just fine...

KwadFan, 2022-10-22:
Hey @theschles,

I am sorry to sound rude, but you messed it up by your self.
How? You simply symlinked files that not exist anymore in Bullseye.
MJPG Streamer relies, at least for Raspi Cams, on a thing calle OpenMAX IL, which is an proprietary "firmware" to enable some decode/encode function of the inbuild GPU of the used SoC (the CPU).
Raspberry Foundation ditched it because they are using a technic called M2M, which is part of the linux kernel in bullseye.

Here ist the catch:
crowsnest installer looks for the /opt/vc directory to decide which version of ustreamer has to be used, which also dropped support for OpenMax in recent versions.
So, if you symlink those, it generates a false positive, the directory exist but not the header files which are used by ustreamer.
Revert that asap and try again.

Therefor I will close this. It isnt an Issue produced by crowsnest, I am sorry but the error was produced on your side.
To approve that it will work use a naked Raspbian bullseye :)
I am already working on bullseye version of mainsailOS and it runs flawless there...

Regards Kwad

closed 2022-10-22 by KwadFan

theschles, 2022-10-22:
> I am already working on bullseye version of mainsailOS and it runs flawless there...

@KwadFan  thank you for the info.  Very enlightening!  I'm still a n00b / Google-coder on some of this stuff.

Help me, though: is mainsailOS built on the 32-bit or 64-bit flavor of Raspberry Pi OS Lite?  I looked at the mainsailOS repo and while I see references to arm64 architecture, I don't see whether it's actually starting with 32-bit Raspberry Pi OS Lite or 64-bit Raspberry Pi OS Lite.

The Raspberry Pi Imager tool doesn't say either -- at least for the current version:
[image]

It would be great if I could choose between mainsailOS 32-bit and 64-bit...