mainsail-crew/crowsnest #46: Wrong path

jafc76, 2022-10-30:
### What happened

After a clean Install on my Raspberry Pi of Klipper, Moonraker, Mainsail, KlipperScreen and Crowsnest I can see that the path of crowsnest.conf and crowsnest.log is wrong.

It should be:
/home/pi/printer_data/config for crowsnest.conf
/home/pi/printer_data/logs for crownsnest.log

It's installed here:
/home/pi/klipper_config for crowsnest.conf
/home/pi/klipper_logs for crownsnest.log

So I was changing the configuration of the camera but had no effect and nothing changed until I changed the same file on the old path.

### What did you expect to happen

The path should be changed.

### How to reproduce

After to update, everybody will have the same problem

### Additional information

Thank you so much!

KwadFan, 2022-11-01:
Hello,
Thank you for reporting! Yes, this is a known issue.

I would be happy if you could try this branch:
https://github.com/mainsail-crew/crowsnest/tree/ref/installer
I refactored the whole mechanism. Any thoughts and improvements on this are welcome.

This should also get rid of #41 

Regards Kwad

jafc76, 2022-11-01:
Thanks mate!!

Guilouz, 2022-11-03:
And in case of multi instances path must be /home/pi/printer_1_data/config

KwadFan, 2022-11-03:
I have no multiinstance in mind here. But as you have to configure the installer for your desired paths its up to you.

Default paths are set for single instance.
Despite the fact it make not a real sense to me, till yet. because all cams should be connected to the same pi.
To be honest, I never had that in mind, but I dont know if it will run with different cfg and env files :man_shrugging:

tinymachines3d, 2022-11-03:
@KwadFan what about update manager entry for this other branch? When I change to primary branch: ref/installer, it breaks.

KwadFan, 2022-11-03:
> @KwadFan what about update manager entry for this other branch? When I change to primary branch: ref/installer, it breaks.

That branch is only thought for testing. As soon I merged that you can switch back to master branch

tinymachines3d, 2022-11-03:
@KwadFan cloning the ref/installer branch works and is nicely setup. How much longer does it need tested?

Guilouz, 2022-11-03:
Confirmed tested too and it's working for multi instances too by typing the correct path.

KwadFan, 2022-11-03:
> @KwadFan cloning the ref/installer branch works and is nicely setup. How much longer does it need tested?

Well, I am not fully finished. Wrote a "unit test" today.
That will be merged to develop this night, at least tommorow morning. If that test gets a go by "internal gates" ( named mete and alexz ;) ) it will land in master.

But I aprriciate your tests! Thats a huge help for me, Thank you guys!

Guilouz, 2022-11-03:
> @KwadFan what about update manager entry for this other branch? When I change to primary branch: ref/installer, it breaks.

Working fine for me

jsdokos, 2022-11-05:
This also appears to be working fine for me on the following:

Fluidd v1.21.1 
klipper v0.10.0-620-g11dd273b
moonraker v0.7.1-747-g779997c

jsdokos, 2022-11-06:
If you change your update manager to the following it should remove the error in the short term:

[update_manager crowsnest]
type: git_repo
path: ~/crowsnest
**primary_branch: ref/installer #master**
origin: https://github.com/mainsail-crew/crowsnest.git
managed_services: crowsnest
install_script: tools/install.sh

KwadFan, 2022-11-07:
Mentioning #52

commit d4a2c081 (2022-11-04): refactor: implements new install mechanism

This installer requires to be configured due
make config

Changes:

Add tools/configure.sh script and make config.

Refactor of uninstall.sh according to new installer

Add dev-helper.sh to provide various informations for developing
and also could check existing installation as kinda "unit-test"

This should also fix #46
and fix #41
closed 2022-11-15 by KwadFan