mainsail-crew/crowsnest #49: Unable to start RTSP stream - Configuration issue?

mk-mrshll, 2022-11-04:
### What happened

When trying to start up a camera with rtsp, I get a failure in the logs. Following the error description, it seems like rtsp-simple-server is having a problem getting ffmpeg to feed the rtsp stream.

Here's the lines in the `crowsnest.log` :
```
[11/04/22 09:22:29] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[11/04/22 09:22:29] crowsnest: Version: v3.0.3-1-gee59a12
[11/04/22 09:22:29] crowsnest: Prepare Startup ...
[11/04/22 09:22:30] crowsnest: INFO: Host information:
[11/04/22 09:22:30] crowsnest: Host Info: Distribution: Ubuntu 20.04.5 LTS
[11/04/22 09:22:30] crowsnest: Host Info: Kernel: Linux 4.9.191 aarch64
[11/04/22 09:22:30] crowsnest: Host Info: Available CPU Cores: 4
[11/04/22 09:22:30] crowsnest: Host Info: Available Memory: 999676 kB
[11/04/22 09:22:30] crowsnest: Host Info: Diskspace (avail. / total): 8.5G / 15G
[11/04/22 09:22:30] crowsnest: INFO: Checking Dependencys
[11/04/22 09:22:30] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[11/04/22 09:22:30] crowsnest: Dependency: 'find' found in /usr/bin/find.
[11/04/22 09:22:30] crowsnest: Dependency: 'logger' found in /usr/bin/logger.
[11/04/22 09:22:30] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[11/04/22 09:22:30] crowsnest: Dependency: 'ffmpeg' found in /usr/bin/ffmpeg.
[11/04/22 09:22:30] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[11/04/22 09:22:30] crowsnest: Dependency: 'rtsp-simple-server' found in bin/rtsp-simple-server/rtsp-simple-server.
[11/04/22 09:22:31] crowsnest: Version Control: ustreamer is up to date. (v5.28)
[11/04/22 09:22:34] crowsnest: Version Control: rtsp-simple-server is up to date. (v0.19.1)
[11/04/22 09:22:35] crowsnest: Version Control: ffmpeg is up to date. (4.2.7-0ubuntu0.1)
[11/04/22 09:22:35] crowsnest: INFO: Print Configfile: '/home/pi/printer_data/config/crowsnest.conf'
[11/04/22 09:22:35] crowsnest: 		[crowsnest]
[11/04/22 09:22:35] crowsnest: 		log_path: ~/printer_data/logs/crowsnest.log
[11/04/22 09:22:35] crowsnest: 		log_level: verbose
[11/04/22 09:22:35] crowsnest: 		delete_log: true
[11/04/22 09:22:35] crowsnest: 		
[11/04/22 09:22:35] crowsnest: 		[cam 1]
[11/04/22 09:22:35] crowsnest: 		mode: rtsp
[11/04/22 09:22:35] crowsnest: 		port: 8080
[11/04/22 09:22:35] crowsnest: 		device: /dev/video0
[11/04/22 09:22:35] crowsnest: 		resolution: 1920x1080
[11/04/22 09:22:35] crowsnest: 		max_fps: 30
[11/04/22 09:22:35] crowsnest: INFO: Detect available Devices
[11/04/22 09:22:35] crowsnest: INFO: Found 1 total available Device(s)
[11/04/22 09:22:36] crowsnest: INFO: Found 1 available camera(s)
[11/04/22 09:22:36] crowsnest: /dev/v4l/by-id/usb-046d_Logi_Webcam_C920e_9D537C9F-video-index0 -> /dev/video0
... skipping some lines here ...
[11/04/22 09:22:43] crowsnest: INFO: No usable CSI Devices found.
[11/04/22 09:22:43] crowsnest: V4L2 Control:
[11/04/22 09:22:44] crowsnest: No parameters set for [cam 1]. Skipped.
[11/04/22 09:22:45] crowsnest: Try to start configured Cams / Services...
[11/04/22 09:22:46] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue...
[11/04/22 09:22:46] crowsnest: Starting rtsp-simple-server with config /home/pi/crowsnest/file_templates/crowsnest-rtsp.yml ...
[11/04/22 09:22:47] crowsnest: Starting ffmpeg (rtsp stream source) with Device /dev/video0 ...
[11/04/22 09:22:48] crowsnest: ... Done!
[11/04/22 09:22:51] crowsnest: ERROR: Start of ffmpeg (rtsp stream source) [cam 1] failed!
```

Here's what the rtsp-simple-server.log looks like:
```
2022/11/04 09:22:46 INF rtsp-simple-server v0.19.1
2022/11/04 09:22:46 INF [RTSP] listener opened on :8554 (TCP), :8000 (UDP/RTP), :8001 (UDP/RTCP)
2022/11/04 09:22:46 INF [RTMP] listener opened on :1935
2022/11/04 09:22:46 INF [HLS] listener opened on :8888
2022/11/04 09:22:49 INF [RTSP] [conn [::1]:42546] opened
2022/11/04 09:22:49 INF [RTSP] [session 535526372] created by [::1]:42546
2022/11/04 09:22:49 INF [RTSP] [conn [::1]:42546] closed (invalid SDP: unable to parse track 1: unable to get clock rate: attribute 'rtpmap' not found)
```

Camera is a Logitech C920e hooked up via usb to an FLSUN speed-pad.

### What did you expect to happen

The camera should have started streaming via rtsp and be viewable via a rtsp client like VLC.

### How to reproduce

Configure your webcam as so:
```
[crowsnest]
log_path: ~/printer_data/logs/crowsnest.log
log_level: verbose
delete_log: true

[cam 1]
mode: rtsp 
port: 8080
device: /dev/video0
resolution: 1920x1080
max_fps: 30 
#custom_flags:
#v4l2ctl:
```

Reboot, and notice that the webcam is not streaming.

### Additional information

I found this issue on rtsp-simple-server's repo, but I'm not sure how much it helps:
https://github.com/aler9/rtsp-simple-server/issues/884

KwadFan, 2022-11-04:
What kind of machine you are using? I guess an x86_64 ?

Also please attach output of
`v4l2-ctl -d /dev/v4l/by-id/usb-046d_Logi_Webcam_C920e_9D537C9F-video-index0 --list-formats-ext`

Regards Kwad

mk-mrshll, 2022-11-04:
> What kind of machine you are using? I guess an x86_64 ?
> 
> Also please attach output of `v4l2-ctl -d /dev/v4l/by-id/usb-046d_Logi_Webcam_C920e_9D537C9F-video-index0 --list-formats-ext`
> 
> Regards Kwad

This is ARM64 (Ubuntu 20.04.5 LTS (GNU/Linux 4.9.191 aarch64)). The device is an FLSUN SpeederPad that uses a raspberry pi-like system.

Also, my apologies, that output was the stuff a skipped in the log output. Of course what I decided to omit is what you needed!
Here it is:
```
pi@speeder-pad:~$ v4l2-ctl -d /dev/v4l/by-id/usb-046d_Logi_Webcam_C920e_9D537C9F-video-index0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.133s (7.500 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 160x90
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.133s (7.500 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 160x120
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.133s (7.500 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 176x144
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.133s (7.500 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 320x180
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.133s (7.500 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.133s (7.500 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
[... 176 lines cut ...]
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.133s (7.500 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1600x896
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.133s (7.500 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.133s (7.500 fps)
                        Interval: Discrete 0.200s (5.000 fps)
```

KwadFan, 2022-11-05:
Well, it took a bit to figure out what is going on here, but I think I got an answer to my questions.

https://github.com/mainsail-crew/crowsnest/blob/ee59a12d59cb2db173a93b266c4766b853bd0364/libs/rtspsimple.sh#L31

Thats where the "magic" happends. As you said you are running Ubuntu chances are literally near zero it has "OMX".
Also OMX support is dropped in Raspberry Pi OS Bullseye, I never kept that in mind to be honest.

So, what does that all mean?
ffmpeg is setup to use a lib called h264_omx to "re-encode" if the cam cant nativly deliver H264 Videoframes.
If it has the ability it simply copies the frames and send them to the used simple-rtsp-server.

Your device have either. So, unlucky you, I cant do something against it right now.
I want to replace crowsnests beckend that is able to deliver also rtsp streams but we are far from any guessing when that will be finished and how good it will run on your device.
You could try to "patch" the mentioned line to use lib_x264 but be aware that is CPU enconding and might outperform your device

Regards Kwad

mk-mrshll, 2022-11-05:
@KwadFan Thanks for looking into this, and for the detailed explanation. Since my device doesn't have OMX (is there a way to easily check?), if I were to get a webcam that could natively deliver H264, it would allow to start streaming RTSP?

KwadFan, 2022-11-05:
Fortunatly, I own a cam with inbuild h264. Will try that tommorow with an x86 pc on ubuntu, but I guess yes.

KwadFan, 2022-11-07:
Well, I used an extra small x86 (Dual Core, 2G Ram) and yeah... Sad Story but even that hits its Limits very quickly.
I tested 800x600@30FPS(which is the minimum framerate used by rtsp) and that consumes the whole CPU power.
Using a cam with h264 encoder, does work yes, but with sometimes various results. So, I would say it may work.
Is it reliable? Ehrm.. more less than more, I would say.

Regards Kwad

mk-mrshll, 2022-11-07:
Thanks again Kwad, I'll close this issue now.

closed 2022-11-07 by mk-mrshll