mainsail-crew/crowsnest #50: how to setup raspcam

Toogo, 2022-11-06:
### Is your feature request related to a problem? Please describe

Setup / config raspcam

### Describe the solution you'd like

Nowhere is told/steps how to activate the raspcam

KwadFan, 2022-11-06:
Hey @Toogo ,

I have to apologize, my skills in documentation are lacking, yes.

But in terms of raspicam setup some good news for you!

How to setup? Simply do nothing... Use /dev/video0 in your crowsnest.conf ... done!

It is a bit strange to setup everything that is needed to work properly there for I do everything in the installer for you.
It sets bcm2835-v4l2 in /etc/modules to ensure the module will be laoded, also it copies the bcm2835-v4l2.conf to modprobe.d to ensure it has at least till 2k Video, video capabilitys instead of its fallback using snapshots ( should improve video latency and performance).
Also in /boot/config.txt start_x=1 and gpu_mem should be set, to enable the legacy camera stack that is needed for ustreamer to support raspicams ( bullseye and buster )

So, you see, how to setup? it is...
If it isnt working for you please ensure that the steps are made and provide a log.
(Unless you hit "N" when it asks for Raspicamfix in make config)

Regards Kwad

HinataKato, 2022-11-23:
> Hey @Toogo ,
>
> I have to apologize, my skills in documentation are lacking, yes.
>
> But in terms of raspicam setup some good news for you!
>
> How to setup? Simply do nothing... Use /dev/video0 in your [...]

The script doesn't work in Ubuntu (because the config.txt in Ubuntu is at /boot/firmware/config.txt).

KwadFan, 2022-11-23:
> > Hey @Toogo ,
> > I have to apologize, my skills in documentation are lacking, yes.
> > But in terms of raspicam setup some good news for you!
> > How to setup? Simply do nothing... Use /dev/video0 in your crowsnest.conf ... done!
> > It is a bit strange to setup everything that is needed to work properly there for I do everything in the installer for you. It sets bcm2835-v4l2 in /etc/modules to ensure the module will be laoded, also it copies the bcm2835-v4l2.conf to modprobe.d to ensure it has at least till 2k Video, video capabilitys instead of its fallback using snapshots ( should improve video latency and performance). Also in /boot/config.txt start_x=1 and gpu_mem should be set, to enable the legacy camera stack that is needed for ustreamer to support raspicams ( bullseye and buster )
> > So, you see, how to setup? it is... If it isnt working for you please ensure that the steps are made and provide a log. (Unless you hit "N" when it asks for Raspicamfix in make config)
> > Regards Kwad
> 
> The script doesn't work in Ubuntu (because the config.txt in Ubuntu is at /boot/firmware/config.txt).
> 
> Edit : Looks like modules fix didn't work either, I didn't see bcm2835-v4l2 in modules or modprobe

Well, if I have some spare time I will do some tests on that.
But in general it doesnt make sense to me using Ubuntu on a Raspberry Pi.
As far as I know, Ubuntu doesnt provide any of the userland scripts or binarys that are shipped with Raspberry Pi OS.
So, its unlikely that things not working as expected, especcialy with things like firmware blobs or kernel modules.

The way easier route for you should be using MainsailOS or a Raspberry Pi OS ( legacy or latest, does'nt matter).
Regards Kwad

KwadFan, 2022-11-23:
I want a broad range of supported Os'es but takes time. Will see what I can do based on your input :)

HinataKato, 2022-11-23:
> I want a broad range of supported Os'es but takes time. Will see what I can do based on your input :)

Glad to hear that :D

HinataKato, 2022-11-24:
> @HinataKato
> 
> Could you tell me which version you'd used? I flashed an ubuntu server 22.04 LTS today and its config.txt is located in "system-boot"

im using Ubuntu server from the official site
[image]
That's weird, mines are in the /boot/firmware
[image]

KwadFan, 2022-11-24:
Well, I have a trace.

I flashed the Image using rpi-imager ( which should use official repos ).
But, the Image has kernel 5.10 and after running `apt upgrade` they installed kernel 5.15 ...

And thats the issue here. They changed the locatation to `/boot/firmware` :facepalm: 

Ubuntu is doing Ubuntu things... Again.. :rofl: 

So, yeah, I'll try to fix that.

KwadFan, 2022-11-24:
@HinataKato 

Just to confirm.
Used Ubuntu 22.04 LTS 64bit with an Raspicam V2 (screenshot) and V1

[image]

HinataKato, 2022-11-24:
> @HinataKato
> 
> Just to confirm. Used Ubuntu 22.04 LTS 64bit with an Raspicam V2 (screenshot) and V1
> 
> [image]

Yep, the install script’s edit config.txt is working, but the Raspicam Fix isn't; it doesn't echo "Device is Raspberry Pi," and the modules file is empty
[image]

KwadFan, 2022-11-24:
@HinataKato 
> Edit: The space between = and the value will not work.

Doesnt matter they get ignored. Otherwise you wont have a picture with a raspi cam.

So, yes, because the fix will only work with Raspberry PiOS since Ubuntu has no bcm2835-v4l2 module. If it isnt there I cant force it to use.

Use v4l2-ctl --list-device to see cam "path"

HinataKato, 2022-11-24:
> @HinataKato
>
>
> Doesnt matter they get ignored. Otherwise you wont have a picture with a raspi cam.
>
> So, yes, because the fix will only work with Raspberry PiOS since Ubuntu has no bcm2835-v4l2 [...]

I guess you're talking about modules? in Ubuntu it had those modules, and I tested the fix, and it fixed my trash frame rate.
[image]

KwadFan, 2022-11-24:
@HinataKato 

Only to be sure I tested your theory about the blanks... Nope, as said doesnt matter.
[image]

Only to be sure what kind of Raspicam do you own? Is that an Arducam?

HinataKato, 2022-11-24:
>This drives me bonkers ....

Same here lol
> Only to be sure what kind of Raspicam do you own? Is that an Arducam?

An rpi official v1. 2 cam
> Only to be sure I tested your theory about the blanks... Nope, as said doesnt matter. !>

Here's my test
[image]
[image]

>But why the heck do you have that module while I have not ^^

Maybe it's device limited issue? Mine's are 3B,I will test it someday with blank system to see if I can reproduce this situation 😫 (or maybe just wait the upstream to release a version that use Linux's camera stack lol)

KwadFan, 2022-11-24:
I think our best bet will be v4l2m2m ... But time will tell.

I think about a solution to get rid of the blanks meanwhile. Also thinking about for "live" installs grepping the module name as Indicator for the Raspicam fix

We'll how well this turns out.

HinataKato, 2022-12-06:
>I will test it someday with blank system to see if I can reproduce this situation

I just thought about it these days, and then today I burned the power supply port; have to wait for the repair, lol

KwadFan, 2022-12-12:
@HinataKato 

I'll hope that will fix your issue too. If there is a need for further fixes, please open a new issue

Regards Kwad

commit 607bdda9 (2022-11-24): fix: fixes missing legacy cam stack in ubuntu

Ubuntu with kernel 5.15 moves its config.txt to
`/boot/firmware/config.txt`

This should fix #50 (fixes#50)
commit f0ed428e (2022-11-24): fix: fixes missing legacy cam stack in ubuntu

Ubuntu with kernel 5.15 moves its config.txt to
`/boot/firmware/config.txt`

This should fix #50 (fixes#50)
commit 722c7ca3 (2022-12-12): fix: fix error introduced by crudini

Used version of crudini puts spaces between values and parameters
This leads to an unusable config.txt

To fix that behavior simply remove blanks with sed.

This may fix #50
closed 2022-12-12 by KwadFan