mainsail-crew/crowsnest #54: Error 127, ustreamer won't start

INeDiA, 2022-11-08:
### What happened

I don't know if this is related to crowsnest, ustreamer or bullseye, anyway I did a fresh install and immediately find this at the end of my log:

```
[11/08/22 23:51:31] crowsnest: ERROR: Error 127 occured on line 31
[11/08/22 23:51:31] crowsnest: ERROR: Stopping crowsnest.
[11/08/22 23:51:31] crowsnest: Goodbye...
[11/08/22 23:51:31] crowsnest: ERROR: Error 1 occured on line 31
[11/08/22 23:51:31] crowsnest: ERROR: Stopping crowsnest.
[11/08/22 23:51:31] crowsnest: Goodbye...
```

Then tried to start ustreamer manually:
```
pi@raspberrypi:~ $ crowsnest/bin/ustreamer/ustreamer
crowsnest/bin/ustreamer/ustreamer: error while loading shared libraries: libopenmaxil.so: cannot open shared object file: No such file or directory
```

I noticed this error message while the crowsnest installer was compiling ustreamer:
```
== LD ustreamer.bin
/usr/bin/ld: warning: libbrcmGLESv2.so, needed by /opt/vc/lib/libopenmaxil.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libbrcmEGL.so, needed by /opt/vc/lib/libopenmaxil.so, not found (try using -rpath or -rpath-link)
```

Tried to fix it by doing so:
```
sudo ln -s /usr/lib/arm-linux-gnueabihf/libGLESv2.so /usr/lib/libbrcmGLESv2.so
sudo ln -s /usr/lib/arm-linux-gnueabihf/libEGL.so /usr/lib/libbrcmEGL.so
```

Now it compiles without any error or warning, but when launched it still gives the missing libopenmaxil.so error.

### What did you expect to happen

ustreamer to start without issues

### How to reproduce

just did a fresh install

### Additional information

```
[11/08/22 23:51:30] crowsnest: Version: v3.0.3-32-gf7f6e74
[11/08/22 23:51:30] crowsnest: Prepare Startup ...
[11/08/22 23:51:30] crowsnest: INFO: Host information:
[11/08/22 23:51:31] crowsnest: Host Info: Distribution: Raspbian GNU/Linux 11 (bullseye)
[11/08/22 23:51:31] crowsnest: Host Info: Kernel: Linux 5.15.76-v7l+ armv7l
[11/08/22 23:51:31] crowsnest: Host Info: Model: Raspberry Pi 4 Model B Rev 1.5
```

arraylabs, 2022-11-09:
I've also run into the error 127. Fresh install of Mainsail OS yesterday, worked for a couple prints and now the 127 error.

KwadFan, 2022-11-09:
I really should work on easier understand able error messages -.- 

But you already pointed out whats going on.

For buster Images I have to revert some commits of ustreamer because it uses openmaxIL. To check if it has Sources installed, it looks vor a dir in /opt/vc.

I cant say why it is in your Image bit if it is bullseye this folder shouldnt exist.

Delete it and recompile

INeDiA, 2022-11-09:
> I really should work on easier understand able error messages -.-
>
> But you already pointed out whats going on.
>
> For buster Images I have to revert some commits of ustreamer because it uses [...]

Sorry for opening a new issue, I swear I searched for the error but nothing showed up...
Well this isn't a fresh bullseye install, it was upgraded from buster some time ago, maybe it left some spoils.

I deleted the whole /opt/vc and now crowsnest service starts without any issue, thanks!

I still get errors from ustreamer in my logs but I don't know if this is on me

```
crowsnest: DEBUG: ustreamer [cam]: Parameters: --host 127.0.0.1 -p -d --device-timeout=2 -r -f --allow-origin=* --static /home/pi/crowsnest/ustreamer-www
crowsnest: DEBUG: ustreamer [cam]: Invalid value for '--port=-d': min=1, max=65535
```

Looks like it completely ignores my config
```
crowsnest: ERROR: Parameter device
max_fps
mode
port
resolution not found in Section [cam]. Start skipped!
[11/09/22 09:40:52] crowsnest: WARN: Unknown Mode configured!
[11/09/22 09:40:52] crowsnest: WARN: Using 'mode: mjpg' as fallback!
[11/09/22 09:40:52] crowsnest: ERROR: Parameter device
```

crowsnest config:
```
[cam]
mode: mjpg                              # mjpg/rtsp
port: 8080                              # Port
device: /dev/video0                     # See Log for available ...
resolution: 640x480                    # widthxheight format
max_fps: 15
```

KwadFan, 2022-11-09:
Yes of course it does...

You didnt  set a name for the camera ;)

INeDiA, 2022-11-09:
> Yes of course it does...
> 
> You didnt set a name for the camera ;)
> 
> EDIT: To be a bit more precise here.
> 
> [cam foo]
> 
> Internally I use the name of the section and ripping out the "cam" from the string. Think of like in OOP approach, which obviously not works in bash, so foo.resolution, foo.device and so on... If there is no foo, crowsnest cant know what to do with an empty string/class.
> 
> I think I should implement a warnig or smt like that 🤔

Actually I just removed it from the log 😬

KwadFan, 2022-11-09:
So:

[cam]
mode: mjpg                              # mjpg/rtsp
port: 8080                              # Port
device: /dev/video0                     # See Log for available ...
resolution: 640x480                    # widthxheight format
max_fps: 15

That isnt what your config looks like?

INeDiA, 2022-11-09:
> So:
> 
> [cam] mode: mjpg # mjpg/rtsp port: 8080 # Port device: /dev/video0 # See Log for available ... resolution: 640x480 # widthxheight format max_fps: 15
> 
> That isnt what your config looks like?

No I set a name for the printer, I just removed it because it's silly 😅
Sorry for that

KwadFan, 2022-11-09:
Ok then, I dont know why it failes. This would be the only reason I could imagine

INeDiA, 2022-11-09:
> Ok then, I dont know why it failes. This would be the only reason I could imagine

of course you were right, config should be [cam sillyname] while I just used [sillyname]
up & running, thanks a lot!

KwadFan, 2022-11-09:
> > Ok then, I dont know why it failes. This would be the only reason I could imagine
> 
> of course you were right, config should be [cam sillyname] while I just used [sillyname] up & running, thanks a lot!

You're welcome :heart:

closed 2022-11-09 by KwadFan