mainsail-crew/crowsnest #55: INFO: No usable CSI Devices found.

KaloyanR, 2022-11-15:
### What happened

But I have usb cam.

What I have to do?

### What did you expect to happen

what should i do when the camera is usb

### How to reproduce

.

### Additional information

[11/15/22 07:17:25] crowsnest: INFO: Found 1 available camera(s)
[11/15/22 07:17:25] crowsnest: /dev/v4l/by-id/usb-Jieli_Technology_USB_PHY_2.0-video-index0 -> /dev/video0
[11/15/22 07:17:25] crowsnest: Supported Formats:
[11/15/22 07:17:25] crowsnest: 		[0]: 'MJPG' (Motion-JPEG, compressed)
[11/15/22 07:17:25] crowsnest: 		Size: Discrete 1920x1080
[11/15/22 07:17:25] crowsnest: 		Interval: Discrete 0.033s (30.000 fps)
[11/15/22 07:17:25] crowsnest: 		Size: Discrete 1280x720
[11/15/22 07:17:25] crowsnest: 		Interval: Discrete 0.033s (30.000 fps)
[11/15/22 07:17:25] crowsnest: 		Size: Discrete 640x480
[11/15/22 07:17:25] crowsnest: 		Interval: Discrete 0.033s (30.000 fps)
[11/15/22 07:17:25] crowsnest: 		Size: Discrete 640x360
[11/15/22 07:17:25] crowsnest: 		Interval: Discrete 0.033s (30.000 fps)
[11/15/22 07:17:25] crowsnest: 		Size: Discrete 352x288
[11/15/22 07:17:25] crowsnest: 		Interval: Discrete 0.033s (30.000 fps)
[11/15/22 07:17:25] crowsnest: 		Size: Discrete 320x240
[11/15/22 07:17:25] crowsnest: 		Interval: Discrete 0.033s (30.000 fps)
[11/15/22 07:17:25] crowsnest: 		[1]: 'YUYV' (YUYV 4:2:2)
[11/15/22 07:17:25] crowsnest: 		Size: Discrete 1280x720
[11/15/22 07:17:25] crowsnest: 		Interval: Discrete 0.200s (5.000 fps)
[11/15/22 07:17:25] crowsnest: 		Size: Discrete 640x480
[11/15/22 07:17:25] crowsnest: 		Interval: Discrete 0.040s (25.000 fps)
[11/15/22 07:17:25] crowsnest: 		Size: Discrete 640x360
[11/15/22 07:17:25] crowsnest: 		Interval: Discrete 0.033s (30.000 fps)
[11/15/22 07:17:25] crowsnest: 		Size: Discrete 352x288
[11/15/22 07:17:25] crowsnest: 		Interval: Discrete 0.033s (30.000 fps)
[11/15/22 07:17:25] crowsnest: 		Size: Discrete 320x240
[11/15/22 07:17:25] crowsnest: 		Interval: Discrete 0.033s (30.000 fps)
[11/15/22 07:17:25] crowsnest: Supported Controls:
[11/15/22 07:17:25] crowsnest: 		white_balance_temperature 0x0098091a (int) : min=0 max=255 step=1 default=128 value=128
[11/15/22 07:17:25] crowsnest: INFO: No usable CSI Devices found.
[11/15/22 07:17:25] crowsnest: V4L2 Control:
[11/15/22 07:17:26] crowsnest: No parameters set for [cam 1]. Skipped.
[11/15/22 07:17:26] crowsnest: Try to start configured Cams / Services...
[11/15/22 07:17:27] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue...
[11/15/22 07:17:27] crowsnest: Starting ustreamer with Device /dev/video0 ...
[11/15/22 07:17:28] crowsnest: ERROR: Start of ustreamer [cam 1] failed!
[11/15/22 07:17:29] crowsnest: ... Done!

KwadFan, 2022-11-15:
To be honest, I cant follow you, what is a Bug here? INFO is a short for Informational message. It only lets you know that there is no CSI device is attached.

In first place you should describe the issue in `What happend`

Your answer was:

`But i have usb...`

That tells me nothing about your Issue.
So, please review your issue and provide at least some useful informations.

Regards Kwad

KaloyanR, 2022-11-15:
Sorry I don't post here often, I don't know the platform.
I have a USB Camera that does not stream.
...
[11/15/22 07:17:28] crowsnest: ERROR: Start of ustreamer [cam 1] failed!
[11/15/22 07:17:29] crowsnest: ... Done!

KwadFan, 2022-11-15:
> Sorry I don't post here often, I don't know the platform. I have a USB Camera that does not stream. ... [11/15/22 07:17:28] crowsnest: ERROR: Start of ustreamer [cam 1] failed! [11/15/22 07:17:29] crowsnest: ... Done!

Well, then we have to determine why ustreamer refuses to do its Job...
Please open your crowsnest.conf and set in section `[crowsnest]`
two parameters:
`log_level: debug
delete_log: true`
after that restart crowsnest and wait for the stream to show up, if it doesnt show up, grab the File and post it here.

KaloyanR, 2022-11-16:
Thanks KwadFan!
For some reason port 8080 was busy, I put webcam2 on 8081 and it worked.

But what is the reason the 8080 is being used?

KwadFan, 2022-11-16:
> Thanks KwadFan!
> For some reason port 8080 was busy, I put webcam2 -> 8081 and cam works.
> 
> But what is the reason the 8080 is being used?

Relative simple to determine....

`sudo apt udate`
`sudo apt install lsof`
`lsof -i:8080`

This will show you the name of the process thats blocking the port

KaloyanR, 2022-11-16:
It shows nothing. I may have had a default defined mainsail camera at that point...

Thanks for the Help ...is closed.

closed 2022-11-16 by KaloyanR