mainsail-crew/crowsnest #58: Non-explicit error message (Error 1 occured on line 45)

Hexalyse, 2022-11-20:
### What happened

I installed MainsailOS on a Raspberry Pi 2 ver B.

When crowsnest tries to start, here is the debug log:

'''
Nov 20 13:24:15 printer crowsnest[542]: crowsnest - A webcam Service for multiple Cams and Stream Services.
Nov 20 13:24:15 printer crowsnest[369]: [11/20/22 13:24:15] crowsnest: Version: v3.0.3
Nov 20 13:24:16 printer crowsnest[554]: Version: v3.0.3
Nov 20 13:24:16 printer crowsnest[369]: [11/20/22 13:24:16] crowsnest: Prepare Startup ...
Nov 20 13:24:16 printer crowsnest[570]: Prepare Startup ...
Nov 20 13:24:23 printer crowsnest[369]: [11/20/22 13:24:23] crowsnest: INFO: Host information:
Nov 20 13:24:24 printer crowsnest[617]: INFO: Host information:
Nov 20 13:24:25 printer crowsnest[369]: [11/20/22 13:24:24] crowsnest: Host Info: Distribution: Raspbian GNU/Linux 10 (buster)
Nov 20 13:24:25 printer crowsnest[633]: Host Info: Distribution: Raspbian GNU/Linux 10 (buster)
Nov 20 13:24:25 printer crowsnest[369]: [11/20/22 13:24:25] crowsnest: Host Info: Release: MainsailOS release 0.7.1 (buster)
Nov 20 13:24:26 printer crowsnest[645]: Host Info: Release: MainsailOS release 0.7.1 (buster)
Nov 20 13:24:26 printer crowsnest[369]: [11/20/22 13:24:26] crowsnest: Host Info: Kernel: Linux 5.10.103+ armv6l
Nov 20 13:24:27 printer crowsnest[656]: Host Info: Kernel: Linux 5.10.103+ armv6l
Nov 20 13:24:27 printer crowsnest[369]: [11/20/22 13:24:27] crowsnest: Host Info: Model: Raspberry Pi Model B Rev 2
Nov 20 13:24:27 printer crowsnest[663]: Host Info: Model:  Raspberry Pi Model B Rev 2
Nov 20 13:24:28 printer crowsnest[369]: [11/20/22 13:24:28] crowsnest: Host Info: Available CPU Cores: 1
Nov 20 13:24:28 printer crowsnest[672]: Host Info: Available CPU Cores: 1
Nov 20 13:24:29 printer crowsnest[369]: [11/20/22 13:24:28] crowsnest: Host Info: Available Memory: 440344 kB
Nov 20 13:24:29 printer crowsnest[679]: Host Info: Available Memory: 440344 kB
Nov 20 13:24:29 printer crowsnest[369]: [11/20/22 13:24:29] crowsnest: Host Info: Diskspace (avail. / total): 53G / 59G
Nov 20 13:24:29 printer crowsnest[686]: Host Info: Diskspace (avail. / total): 53G / 59G
Nov 20 13:24:30 printer crowsnest[369]: [11/20/22 13:24:30] crowsnest: INFO: Checking Dependencys
Nov 20 13:24:30 printer crowsnest[698]: INFO: Checking Dependencys
Nov 20 13:24:31 printer crowsnest[369]: [11/20/22 13:24:31] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
Nov 20 13:24:31 printer crowsnest[711]: Dependency: 'crudini' found in /usr/bin/crudini.
Nov 20 13:24:33 printer crowsnest[369]: [11/20/22 13:24:32] crowsnest: Dependency: 'find' found in /usr/bin/find.
Nov 20 13:24:33 printer crowsnest[722]: Dependency: 'find' found in /usr/bin/find.
Nov 20 13:24:34 printer crowsnest[369]: [11/20/22 13:24:34] crowsnest: Dependency: 'logger' found in /usr/bin/logger.
Nov 20 13:24:34 printer crowsnest[732]: Dependency: 'logger' found in /usr/bin/logger.
Nov 20 13:24:36 printer crowsnest[369]: [11/20/22 13:24:35] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
Nov 20 13:24:36 printer crowsnest[746]: Dependency: 'xargs' found in /usr/bin/xargs.
Nov 20 13:24:37 printer crowsnest[369]: [11/20/22 13:24:37] crowsnest: Dependency: 'ffmpeg' found in /usr/bin/ffmpeg.
Nov 20 13:24:37 printer crowsnest[758]: Dependency: 'ffmpeg' found in /usr/bin/ffmpeg.
Nov 20 13:24:38 printer crowsnest[369]: [11/20/22 13:24:38] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
Nov 20 13:24:38 printer crowsnest[768]: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
Nov 20 13:24:39 printer crowsnest[369]: [11/20/22 13:24:39] crowsnest: Dependency: 'rtsp-simple-server' found in bin/rtsp-simple-server/rtsp-simple-server.
Nov 20 13:24:39 printer crowsnest[778]: Dependency: 'rtsp-simple-server' found in bin/rtsp-simple-server/rtsp-simple-server.
Nov 20 13:24:43 printer crowsnest[369]: ~/crowsnest/bin/ustreamer /
Nov 20 13:24:44 printer crowsnest[369]: [11/20/22 13:24:44] crowsnest: Version Control: ustreamer is up to date. (v4.13)
Nov 20 13:24:44 printer crowsnest[799]: Version Control: ustreamer is up to date. (v4.13)
Nov 20 13:24:44 printer crowsnest[369]: /
Nov 20 13:24:44 printer crowsnest[369]: ~/crowsnest/bin/rtsp-simple-server /
Nov 20 13:24:45 printer crowsnest[816]: ERROR: Error 132 occured on line 45
Nov 20 13:24:46 printer crowsnest[827]: ERROR: Stopping crowsnest.
Nov 20 13:24:46 printer crowsnest[836]: Goodbye...
Nov 20 13:24:47 printer crowsnest[369]: [11/20/22 13:24:47] crowsnest: ERROR: Error 1 occured on line 45
Nov 20 13:24:47 printer crowsnest[846]: ERROR: Error 1 occured on line 45
Nov 20 13:24:48 printer crowsnest[369]: [11/20/22 13:24:47] crowsnest: ERROR: Stopping crowsnest.
Nov 20 13:24:48 printer crowsnest[857]: ERROR: Stopping crowsnest.
Nov 20 13:24:48 printer crowsnest[369]: [11/20/22 13:24:48] crowsnest: Goodbye...
Nov 20 13:24:48 printer crowsnest[866]: Goodbye...
```

As you can see, there is an "Error 1" and an "Error 132" on line 45. But I have no idea in what file is the line 45 problematic, nor did I find an explanation of what is error 1 or error 132 online.

### What did you expect to happen

Service should start without error

### How to reproduce

Install MainsailOS. Check Crowsnest log.

KwadFan, 2022-11-21:
Hey,

to be honest I see this error the first time but I never tested crowsnest on a RPI2B.
But, a quick google-foo tells me that Error 132 is an exit code that a program is terminated due SIGKILL with an illegal core instruction (128+4).

As you said if you set to quiet, the version check will be skipped at all. That tells me that this error occurs in version checking of the rtsp server binary.

I guess that this precompiled binary is not working on this type of CPU (should be ARMv6, IRC).

To confirm that I need output of 
```uname -m```
and please try to run 
```~/crowsnest/bin/rtsp-simple-server/rtsp-simple-server --version```
this should spit out ```v0.19.1``` without any error.

If there is an error message please post that.

Regards Kwad

wassereimer86, 2022-11-21:
I have the same error. Everything was working fine until the update I had a few minutes ago. I have a RPI4B.

`[11/21/22 19:16:27] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[11/21/22 19:16:27] crowsnest: Version: v3.0.3-10-g77f23e0
[11/21/22 19:16:27] crowsnest: Prepare Startup ...
[11/21/22 19:16:27] crowsnest: INFO: Host information:
[11/21/22 19:16:27] crowsnest: Host Info: Distribution: Raspbian GNU/Linux 10 (buster)
[11/21/22 19:16:27] crowsnest: Host Info: Release: MainsailOS release 0.7.1 (buster)
[11/21/22 19:16:27] crowsnest: Host Info: Kernel: Linux 5.10.103-v7l+ armv7l
[11/21/22 19:16:27] crowsnest: Host Info: Model: Raspberry Pi 4 Model B Rev 1.1
[11/21/22 19:16:27] crowsnest: Host Info: Available CPU Cores: 4
[11/21/22 19:16:27] crowsnest: Host Info: Available Memory: 3748168 kB
[11/21/22 19:16:27] crowsnest: Host Info: Diskspace (avail. / total): 24G / 30G
[11/21/22 19:16:27] crowsnest: INFO: Checking Dependencys
[11/21/22 19:16:27] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[11/21/22 19:16:27] crowsnest: Dependency: 'find' found in /usr/bin/find.
[11/21/22 19:16:27] crowsnest: Dependency: 'logger' found in /usr/bin/logger.
[11/21/22 19:16:27] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[11/21/22 19:16:27] crowsnest: Dependency: 'ffmpeg' found in /usr/bin/ffmpeg.
[11/21/22 19:16:27] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[11/21/22 19:16:27] crowsnest: Dependency: 'rtsp-simple-server' found in bin/rtsp-simple-server/rtsp-simple-server.
[11/21/22 19:16:27] crowsnest: ERROR: Error 127 occured on line 31
[11/21/22 19:16:27] crowsnest: ERROR: Stopping crowsnest.
[11/21/22 19:16:27] crowsnest: Goodbye...
[11/21/22 19:16:27] crowsnest: ERROR: Error 1 occured on line 31
[11/21/22 19:16:27] crowsnest: ERROR: Stopping crowsnest.
[11/21/22 19:16:27] crowsnest: Goodbye...`

KwadFan, 2022-11-23:
@Hexalyse 
Any updates on this?

Hexalyse, 2022-11-23:
@KwadFan Sorry for the late reply. I've indeed also tried enabling rtsp (with the logging still in quiet mode) and got another error 132, on a different line, so you're right on the fact it probably comes from rtsp crashing on the RPi2.

Here is the `uname -m` : `armv6l` 
And the `rtsp-simple-server` indeed returns :
```
pi@printer:~$ ~/crowsnest/bin/rtsp-simple-server/rtsp-simple-server --version
Illegal instruction
```
again confirming your theory about CPU/architecture compatibility.

KwadFan, 2022-11-23:
Ok, rtsp-simple-server has armv6 packages so it must be something wrong in my scripts.

I will do some changes today, maybe tommorow. Could you test that branch for me, during the week?

Regards Kwad

Hexalyse, 2022-11-23:
> Could you test that branch for me, during the week?

Sure. Let me know when it's done and needs testing.

KwadFan, 2022-11-24:
Well,

I could pull out may hair, with that.

As you can see here: https://github.com/mainsail-crew/crowsnest/blob/f7cf2260f28901f33f5bc979778afeade8129c10/bin/Makefile#L22

I simply cut down `armv6l` to `armv6`.
I also tried to download the package with that resulting url, which also presents the armv6 package.

So, I am guessing that even the kernel reports an armv6, the Raspi 2B uses a CPU with an armv7 Instruction set.

To confirm that please download the armv7 package: https://github.com/aler9/rtsp-simple-server/releases/download/v0.19.1/rtsp-simple-server_v0.19.1_linux_armv7.tar.gz

```bash
cd /tmp
curl -jLO https://github.com/aler9/rtsp-simple-server/releases/download/v0.19.1/rtsp-simple-server_v0.19.1_linux_armv7.tar.gz
tar -xvzf rtsp-simple-server_v0.19.1_linux_armv7.tar.gz
./rtsp-simple-server --version
```
If it turns out that this will work I guess I brute force the armv7 package and hoping for the best :man_shrugging: 

Regards Kwad

Hexalyse, 2022-11-24:
Well.... I'm starting to wonder what's going on. Same problem... I get an illegal instruction.

But then I tried with the last release (v0.20.2), in armv6, and guess what.... it works.
```shell
pi@printer:/tmp $ tar xvzf rtsp-simple-server_v0.20.2_linux_armv6.tar.gz
rtsp-simple-server
rtsp-simple-server.yml
LICENSE
pi@printer:/tmp $ ./rtsp-simple-server --version
v0.20.2
pi@printer:/tmp $ 
```

I guess the solution would be to update to this version, then ? I have no idea what could be wrong with the v0.19.1, but I've found this issue on another piece of software: https://github.com/alphacep/vosk-api/issues/247
I didn't check rtsp-simple-server in detail, but is it using Python ? Could it be something similar ? And maybe the last version is compiled to work with the version of Python I have, but not the 0.19 ?

KwadFan, 2022-11-24:
Oh f... SHOOT!

No, rtsp-simple-server is written in GO. Now python involved as far as I know.

You this template thingi is my fault. :facepalm: 

Could you please open an issue for that? Will fix that first.

Hexalyse, 2022-11-24:
I opened an issue for the config file location.

But forget about the RPi 2 problem. I just tried RTSP. The CPU of the RPI is so bad that it can't even stream live (probably because by default it encodes with h264 and it's way too intensive for the poor old SoC of the RPi2). I'll go find a cheap second-hand mini-computer for around 100€, since RPi are out of stock everywhere on the planet, and it will be muuuuch easier to just install Mainsail, crowsnest etc. myself.

KwadFan, 2022-11-24:
> PS: hello fellow fpv pilot. Well... I used to be, cause early 2020 I gave all my drone related gear away to a friend cause I was moving abroad and couldn't bring it. Covid brought me back home, and I've been missing FPV for 2 years now :( I wish I hadn't given everything away.

Ouch, you shouldnt have done that :grimacing: , get an TinyWhoop ASAP!!! Covid throws me off flying outdoor but a Whoop runs fine in my appartment LOL

Back to topic.
Encoding of h264 is neccessary because it is the standart protocol for rtsp spec.
But, I have to do some test around it because it was implemented in time where ffmpeg supports GPU encoding due h264_omx lib, which isnt available any more, but lacking of time I couldnt patch it.
Or it uses h264 nativly comme from camera hardware inbuilt h264 encoder.
ffmpeg uses a simple copy command if neither is available what ends mostly in CPU encoding...

As 0.20.2 of rtsp-simple-server seems to be working I will bump the version and hoping that the config.yml will match ^^ 

Regards Kwad

closed 2022-11-24 by KwadFan

megabitsenmzq, 2023-03-26:
I had this issue on my pi zero gen 1. After changing the log option to quiet, everything went fine. Thank you for the solution.

gandalf44, 2023-09-02:
I appreciate the help. I also had this issue on a pi zero w. Adjusting the log option in crowsnest.conf to quiet immediately fixed the issue.

commit 53b4d52f (2022-11-24): chore: bump rtsp-simple-server to v0.20.2

This should also fix #58 (fixes #58)

As described in issue #58 the version v0.19.1 isnt working correctly on
armv6 devices like an RPI 2B
commit 305f5a0d (2022-11-24): chore: bump rtsp-simple-server to v0.20.2

This should also fix #58 (fixes #58)

As described in issue #58 the version v0.19.1 isnt working correctly on
armv6 devices like an RPI 2B