mainsail-crew/crowsnest #62: Install error when using Linux Mint OS

Beachviolence, 2022-11-22:
### Is your feature request related to a problem? Please describe

When running the command `sudo make install` i get the error "logname: no login name". In order to fix it i had to edit the "install.sh" file line 22:
`[[ -n "${BASE_USER}" ]] || BASE_USER="$(logname)"`
and replace "logname" with my username.

### Describe the solution you'd like

Able to install on Linux Mint OS without doing the fix. Unsure if it affects other operating systems.

### Describe alternatives you've considered

Alternativly update documentation to describe the fix.

KwadFan, 2022-11-23:
Just wondering...
How did you run a command when you dont have a "login name"?

You shouldnt edit any files, because this will lead to an "dirty" message in moonrakers update manager.

To fix that simply use `make config` and run Installer afterwards.
If I have some spare time I will try to reproduce that, but changing this would be affecting RPI Images which is the main use case for crowsnest :)

btw. it is tested on Ubuntu Server 20.04, so Mint is ubuntu based, it should work flawless.

Regards Kwad

KwadFan, 2022-11-24:
@Beachviolence 

Could you please test branch fix/fix62 ?

Its an kinda hacky approach but it should do the job.

Regards Kwad

Beachviolence, 2022-11-24:
> 
@KwadFan 

I was just following the instructions and got an error message every time. I fixed the "dirty" message with a soft fix in Mainsail afterwards, and its running just fine now. I will try to uninstall and try out both `make config` first, and try the branch if i figure out how and when my print is done :)

- Simon

Beachviolence, 2022-11-24:
@KwadFan  

I have now unintalled and tried using the `make config`, but it resulted in the same error after completing the prompts
`logname: no login name`
`make: *** [Makefile:37: install] Error 1`

I am not that skilled with git, but when i type `git checkout fix/fix62` it says "your branch is up to date with 'origin/fix/fix62"`, so does that mean that i have tried it? 

- Simon

KwadFan, 2022-11-24:
> @KwadFan
>
> I have now unintalled and tried using the `make config`, but it resulted in the same error after completing the prompts `logname: no login name` `make: *** [Makefile:37: install] Error 1` [...]

run `git fetch && git pull`

Beachviolence, 2022-11-24:
> 
Sorry, but no luck still.

[image]

Let me know if there is anything else i should try as i would like to contribute in making crowsnest work on all platforms of linux

KwadFan, 2022-11-24:
> > 
> 
> Sorry, but no luck still.
> 
> [image]
> 
> Let me know if there is anything else i should try as i would like to contribute in making crowsnest work on all platforms of linux

Could please show output of
`grep "logname" ~/crowsnest/tools/install.sh`

Beachviolence, 2022-11-24:
This is the line i edited in order to make it work. I changed `logname` to my username, `simon` then it would install just fine. 

`[[ -n "${BASE_USER}" ]] || BASE_USER="$(logname)"`

[image]

KwadFan, 2022-11-24:
Ok, something went wrong during the checkout. Should look like this:

```[[ -n "${BASE_USER}" ]] || BASE_USER="$(logname 2> /dev/null || echo "${PWD}" | cut -d'/' -f3)"```

Beachviolence, 2022-11-24:
> 
ah yes, i tried to delete everything and try again, and forgot to do the checkout. I did the checkout and ran the install, and it worked just fine. Sorry!

KwadFan, 2022-11-24:
> > 
> 
> ah yes, i tried to delete everything and try again, and forgot to do the checkout. I did the checkout and ran the install, and it worked just fine. Sorry!

Great! One last thing please provide output of ```lsb_release -a```

Beachviolence, 2022-11-24:
Now it says "invalid" in the update manager in Mainsail, but i assume that is because its not from the master branch.

Here is the output:

```
simon@simon-octoklipper:~/crowsnest$ lsb_release -a
No LSB modules are available.
Distributor ID:	Linuxmint
Description:	Linux Mint 21
Release:	21
Codename:	vanessa
```

KwadFan, 2022-11-24:
> Now it says "invalid" in the update manager in Mainsail, but i assume that is because its not from the master branch.

Exactly, wait a couple Hours I will merge that to master, then checkout master branch again. I will ping you here when its done and thanks for being patient and testing!

KwadFan, 2022-11-24:
@Beachviolence Please switch to branch "master" again...

commit 1c7fe61d (2022-11-24): fix: fixes install error if not via ssh

This should fix #62

If install is run on a Terminal using a Graphical Environment
there is no "logname" using bash global var $USER instead
commit a8ec4dc9 (2022-11-24): fix: fixes install error if not via ssh

This should fix #62

If install is run on a Terminal using a Graphical Environment
there is no "logname" using path and extract user name from path
commit 43c5f697 (2022-11-24): fix: fixes install error if not via ssh

This should fix #62

If install is run on a Terminal using a Graphical Environment
there is no "logname" using path and extract user name from path
commit 0e112711 (2022-11-24): fix: fixes install error on Linux Mint

This should fix #62 (fixes #62)
closed 2022-11-24 by KwadFan