mainsail-crew/crowsnest #67: ayufan/camera-streamer Arducam (libcamera) device (camera-path)

viappidu, 2022-12-07:
### What happened

Working from feat/ayucamstream branch I managed to get it working using the following configuration:

```
...
[cam Arducam_16MP]
mode: multi
port: 8080
device: /base/soc/i2c0mux/i2c@1/imx519@1a
...
custom_flags: -camera-type=libcamera ...
```

Which throws a log entry
`crowsnest: WATCHDOG: Lost Device: '/base/soc/i2c0mux/i2c@1/imx519@1a'`

My configuration also include a USB camera which use the /dev/video0 device entry.

My kernel is 5.15.76-v7l+ #1597 SMP Fri Nov 4 12:14:58 GMT 2022 armv7l GNU/Linux
My /boot/config.txt has the following lines
```
[all]
...
dtoverlay=imx519
dtoverlay=imx519,cam0
...
```

### What did you expect to happen

Device path should be "/dev/video2"
OR maybe "/sys/firmware/devicetree/base/soc/i2c0mux/i2c@1/imx519@1a"

### How to reproduce

First follow the instructions to install Arducam libcamera and modfied imx519 overlay from https://docs.arducam.com/Raspberry-Pi-Camera/Native-camera/Libcamera-User-Guide/ 
Then.
git clone https://github.com/mainsail-crew/crowsnest.git
cd crowsnest
git checkout feat/ayucamstream
./tools/configure.sh (and follow through)
sudo make install
Restart and finish configuration as above

### Additional information

I am starting to think it's possibly an ayufan/camera-streamer bug and I'm going to open an issue also there.

viappidu, 2022-12-07:
@KwadFan 
Yep! Exactly my point of view. I lost few hours looking into it and I can't understand it much either. I didn't even open the issue with @ayufan as it might even be the way libcamera works, which in this case is even modified by Arducam. My eyes were crossing in the code and I (temporarily) stopped. I will  resume asap but I totally agree this is a wip and not a bug. Thanks!
FYI, I can tell you that (previous tests, no real data though) camera streamer seems to be using much (much!) less resources on my pi compared to both ustreamer and (even worse) mjpg-streamer.

ayufan, 2022-12-07:
@viappidu I will try to help. Will get back to it :)

Kcdc3d, 2022-12-08:
Arducam is so touchy with stuff like mainsail because it only seems to accurately work with a specific driver that only works with a specific kernel that no one really uses anymore, or something close to that effect. I believe that is why it's so cheap for what you're getting. There was a huge amount of complaining at the kickstarter launch because they decided to hold that information until they made sales, including mine, so it will be the last product I buy from them.

 Octoprint has the ability to use it, but how they can implement 3rd party drivers and hardware is different.

THat was why I started the search for another way to stream things separately from my klipper raspi, 

Camera streamer does not require this finnicky driver as it literally bypasses anything requiring software to access the Arducam, treating it as an OEM vendor sensor at the hardware level, hence the lack of latency and a good framrate for the resolution provided your network supports it.  This is if you're using the CSI implementation only. YOu have options to then run it through software stuff, but I prefer the raw hardware output as it doesnt affect the stream at all and Mainsail reads the stream fine. No libcamera needed.

It's just a bit of a pain since without a driver, the tools are more "basic" and not very user friendly. But, once you set up autoload scripts, youll never really deal with it. Seems that he added more streaming options since I used it last, in march I think i used it last. I haven t tried them yet.

Kcdc3d, 2022-12-08:
> @Kcdc3d 
> > There was a huge amount of complaining at the kickstarter launch because they decided to hold that information until they made sales, including mine, so it will be the last product I buy from them.
> 
> Not sure if I should laugh or cry :) Been there, done that :(
> 
> Myself I tested camera and software earlier this year, and actually not very much, as my printer is sitting next to me and never running unattended (at the moment at least). Possibly I'll have more use of it when some sort of (free) spaghetti plugin gets out. At the moment the camera was sitting in a bin so I plugged it in to see if I could actually make it work (sic). Never managed to run it without libcamera though. I might give it another try, this chat made me curious. Are you still using it? What's your config like?

No, please laugh, it was at the height of my klipper/printing first phase and was taking things a bit too seriously. Looking back I laugh. They were cheap for large sensors, so why complain .

Well, since they only work with very little, I may set one up on its own little stream for the nozzle since you can get a nice quality image. It got thrown in my "put it back up" bin after I started my first relationship break-up with my printer. Only just started back up now, so I plan to get it on there this weekend. 

Unfortunately I lost my run script, so I will recreate it and check again to see if anything other than c streamer has emerged

viappidu, 2022-12-09:
@Kcdc3d
I'll be curious to see (whenever) your setup. I tried to run it WITHOUT the "-camera-type=libcamera" option but I always get a series of
`device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1): Can't queue buffer.`

@ayufan
I found a closed issue at https://github.com/ayufan/camera-streamer/issues/3 but I can't understand how (if) it is solved.
With libcamera all is working fine. Maybe you need to update the README.md?
(Do we need need libcamera at all? Do we need Arducam specific libcamera and imx519 drivers?)

(sorry for dragging the issue this much)

KwadFan, 2022-12-11:
> (sorry for dragging the issue this much)

All fine, at least it means progress in regard of camera-streamer.

ayufan, 2022-12-17:
@viappidu The `libcamera` is Raspberry PI way of handling camera now. The older `raspicams` might still be accessed directly via `v4l2`, but it is likely to break. Unfortunately, I don't have `raspicam`, only `ArduCams`.

The problem with `ArduCAMs` is that they require to use their own build of `libcamera` for which they do not share sources, because that they claim to have some proprietary stuff there (like camera focus). The `ArduCAMs` are stripped to be working with their cameras :( The Raspberry PI Foundation on the other hand ships `libcamera` also with `USB` support. So this can actually be used as solution for all.

> What did you expect to happen

In your case: it is expected to use `/base/soc/i2c0mux/i2c@1/imx519@1a` as this is how `libcamera` expects paths. If you use `/dev/video2` and camera sensor returns data in bayer format you are expected to configure also ISP. This is what `libcamera` is meant for, it maintains ISP sensor to ensure right color balance, exposure depending on lux hitting sensor.

kaibsora, 2023-01-11:
I also want to add to this and say that i'm using an imx-219 sensor from waveshare and experience the same issue. watchdog loses connection. I can get the camera to work fine using camera-streamer by itself, using the default daemon from Ayufans repo, but that obviously doesn't help with accessing via mainsail, which with this branch i get the same error in crowsnest.log

The new raspi cams (v3) also uses the imx-219 sensor, which is meant for libcamera. 

Thanks to everyone currently working on this.

Zylvo, 2023-01-11:
@kaibsora 
I've tried default config and your config with changed device, but both are not working on camera module v3 :/
btw. module 3 is using imx708 sensor device: `/base/soc/i2c0mux/i2c@1/imx708@1a`

KwadFan, 2023-01-19:
Please keep in mind that branch is absolutly experimental and behaviors like that have to be taken to account!

But thanks for reporting. Maybe it will be useful at some point :)

Regards Kwad

michaelmyc, 2023-04-10:
I was able to get the ArduCam IMX519 to work with Mainsail. I wrote a writeup on the settings you need to tweak to get it working. https://github.com/michaelmyc/mainsail-arducam-imx519

meteyou, 2023-04-10:
This branch is not for production use... So it's not "good" to write a guide for it...

michaelmyc, 2023-04-10:
@meteyou Good point. Decided to write it up as I was scouring the internet for hours trying to get the ArduCam to work. None of the current solutions is good (using pre-production code for crowsnest, using the modified libcamera binaries from ArduCam with no available source code, or hack the autofocus algorithm into system config), but the end result is definitely serviceable, I'd imagine that people might want a temporary solution if they're willing to take the risk and alpha test things. 

I'll update the readme to put giant red letters in front that nothing in the guide is fully tested and should be treated as a hacky workaround rather than a proper fix.

meteyou, 2023-04-10:
@michaelmyc sounds good! but i think this code is already merged in developer branch (also not for production), but as soon as kwad finish the installers, this will be part of the main branch.

closed 2023-05-22 by KwadFan