mainsail-crew/crowsnest #69: boot/config.txt errors

LazeMSS, 2022-12-13:
### What happened

After running the installer my boot config now has these entries:
```
start_=

[pi3]
gpu_me=256

[pi0]
gpu_me=60
```

### What did you expect to happen

the expected result should be gpu_mem=256 ie

### How to reproduce

I did a clean install on raspbian lite with klipper and mailsailos installed using kiauh 
Using a raspicam

KwadFan, 2022-12-15:
Oh shoot....
Thanks for reporting!
I will fix that as soon I have some spare time. 
I added a workaround to get rid of spaces between parameter and value, but it is obvious I made it worse with the sed command I used.
Regards Kwad

LazeMSS, 2022-12-15:
Yes there seems to be a bit of problem there.. using a V2 RPI cam I also had to use legacy mode. It's the error with wrong parameter. But there is already an issue for that

KwadFan, 2022-12-15:
Should be fixed now. Please remove messed up entry from config.txt before trying to install.

Sorry for inconvinience.

Regards Kwad

closed 2022-12-15 by KwadFan

LazeMSS, 2022-12-16:
Tested okay!

commit 5e95c0d3 (2022-12-15): fix(install): wrong syntax in config.txt

This should be fix #69

I used the wrong regex to get rid of blanks introduced by crudini