mainsail-crew/crowsnest #71: Rotate camera feed

NilsNippe, 2023-01-02:
### Is your feature request related to a problem? Please describe

When mounting cameras in cramped enclosures, the camera can end up sideways, meaning the video stream is on its side.
Crowsnest only gives the option to flip the image.

### Describe the solution you'd like

Additional parameter to rotate video stream by a number of degrees (just +-90° should work in most cases).

smwoodward, 2023-01-15:
So it's not possible to rotate the image as it comes in?  Mainsail and Fluidd has the option to rotate the image in the interface. There are some times when I want to use the stream outside of fluidd and mainsail. My camera I had to mount it upside down and it makes the stream useless.

KwadFan, 2023-01-19:
Unfortunatly, no, if the cam doesnt support rotate by its v4l2 driver, there is no chance.
ustreamer does not support image post processing which would be needed here.

Fluidd and Mainsail simply rotating the box where the Image is displayed but its dispatched by ustreamer "as is".

Maybe one day I find the ultimate solution to all our wishes, then it will, state now means "Nope, no chance, sorry Mate ;)"

dweenz, 2023-04-06:
ok I see the response now thank you.