mainsail-crew/crowsnest #76: Rotate camera image

smwoodward, 2023-01-14:
### Is your feature request related to a problem? Please describe

I would like to be able to flip the image based on if I need to mount the camera upside down.

### Describe the solution you'd like

Have options in the config to rotate the image.

KwadFan, 2023-01-19:
Please see https://github.com/mainsail-crew/crowsnest/issues/71

Regards Kwad

closed 2023-01-19 by KwadFan

dweenz, 2023-04-06:
Hi I am also trying to use an inverted camera mount but having issues. I can't seem to flip my cameras output using the parameter --flip-vertical 1 --flip-horizontal 1 or the V4L2CTL --set-ctrl=rotate=180 in crowsnest.cfg. I can flip the image in fluidd but when using timelapse was having a glitch that would show the thumbnails of the timelapse inverted. Also another likely timelapse only  problem was timelapse would take a timelapse of the wrong camera not associated with the printer I have it enabled on. I really  appreciate all your work Kwad.