mainsail-crew/crowsnest #79: Low FPS in stream

djObsidian, 2023-01-22:
### What happened

I am experiencing very low fps in stream, around 1-5 fps. I'm using logitech c920 and my config is the following:
```
[cam logitech]
mode: mjpeg
port: 8080                              # Port
device: /dev/video0
resolution: 1920x1080                   # widthxheight format
max_fps: 30                             # If Hardware Supports this it will be forced, ohterwise ignored/coerced.
v4l2ctl: focus_auto=0,focus_absolute=15,sharpness=130,zoom_absolute=300,power_line_frequency=1
#custom_flags:                          # You can run the Stream Services with custom flags.
```
FPS is low both in fluidd menu and on stream itself

### What did you expect to happen

I expected to have 30 fps as set, or at least something like 20

### How to reproduce

Just use same webcam config with 1920x1080 resolution and 30 fps. 

### Additional information

Here's VLC codec information
[image]
[image]

This is clearly not a hardware issue, i'm not using RPi, but an old nettop with i3-3217U CPU and no, it isn't loaded fully, CPU utilization is lower than 10%.
This also should not be a network problem. Machine itself has a pretty bad signal strength but LAN speed is around 20-25 Mbps which should be more than enough. 
Input bitrate is somewhat suspicious. My experience tells me that 3500 kbps is a good bitrate for FHD 30fps streaming and 5000+kbps is usually overkill, but here we are dealing with ~10000 kbps and 02-05 fps?

meteyou, 2023-01-22:
Full hd with 30fps needs 50-60mb/s to transfer.

djObsidian, 2023-01-22:
There's no stream encoding? H.264 high quality needs only like 4.5 Mbps, according to online calcs

meteyou, 2023-01-22:
We are working on an webrtc/h256 streamer and client.

djObsidian, 2023-01-24:
Okay, thanks. Closing issue, since that's expected behaivour for now

closed 2023-01-24 by djObsidian

laptop-laptop, 2025-04-10:
Hello, I am having the same issue. Tried various services available under Edit Webcam. FPS is usually 0 or 01. I also cannot find webrtc/h256

mryel00, 2025-04-10:
> I am having the same issue

We do not give support in our issues. Open a [Discussion](https://github.com/orgs/mainsail-crew/discussions) or join our [Discord](https://discord.gg/mainsail) for help.

> Tried various services available under Edit Webcam. FPS is usually 0 or 01

Try this: #282 

> I also cannot find webrtc/h256

I have no idea what _exactly_ you mean, but here are our [docs](https://crowsnest.mainsail.xyz/configuration/cam-section#a-small-note-on-webrtc)