mainsail-crew/crowsnest #82: Octoprint video issue

robustini, 2023-01-29:
### What happened

I am ascertaining that unfortunately this system does not work with Octoprint, while with Mainsail and Repetier Server it does, and it is quite strange.
Only snaophot works correctly.
Tried with:

192.168.0.220:8080/?action=stream
127.0.0.1:8080/?action=stream

### What did you expect to happen

I cannot rule out the possibility that it is an Octoprint problem, but with MJPG-Streamer it works.

### How to reproduce

I think it is easily reproducible, I read that someone else has the same problem on the Octoprint forum.

### Additional information

Video stream
[image]

Snapshop
[image]

robustini, 2023-01-29:
Also with Repetier Server work fine!
[image]

KwadFan, 2023-01-30:
I am unsure about octoprint or repetier server, but in Mainsail and Fluidd we use nginx as reverse proxy.
Afaik, octoprint uses HAProxy as reverse Proxy.

My guess is that the ports ar simply not redirected to it. Try to add the `no_proxy: true` option to section `[crowsnest]`

Maybe that helps

robustini, 2023-01-30:
> I am unsure about octoprint or repetier server, but in Mainsail and Fluidd we use nginx as reverse proxy. Afaik, octoprint uses HAProxy as reverse Proxy.
> 
> My guess is that the ports ar simply not redirected to it. Try to add the `no_proxy: true` option to section `[crowsnest]`
> 
> Maybe that helps

Thanks for the reply!
Correct, I use haproxy with Octoprint and this is its configuration file.
I can't interpret the final part, with the webcam port on 8084, i.e. what do you mean by that port.

`global
        maxconn 4096
        user haproxy
        group haproxy
        daemon
        log 127.0.0.1 local0 debug

defaults
        log     global
        mode    http
        option  httplog
        option  dontlognull
        retries 3
        option redispatch
        option http-server-close
        option forwardfor
        maxconn 2000
        timeout connect 5s
        timeout client  15min
        timeout server  15min

frontend public
        bind :::80 v4v6
        use_backend webcam if { path_beg /webcam/ }
        default_backend octoprint

backend octoprint
        reqrep ^([^\ :]*)\ /(.*)     \1\ /\2
        option forwardfor
        server octoprint1 127.0.0.1:5000

backend webcam
        reqrep ^([^\ :]*)\ /webcam/(.*)     \1\ /\2
        server webcam1  127.0.0.1:8084`

KwadFan, 2023-01-30:
Well, hard to tell what that exactly does but it looks to me, that it uses the URL /webcam/ for it.

So if you set, port 8084 in your crowsnest.conf and use the URL http://<printerip>/webcam/?action=stream it might work.

robustini, 2023-01-31:
Same issue without the proxy.

KwadFan, 2023-01-31:
Well, I cant tell how octo or repetier work, I will let this Issue open, but if you set no_proxy then it shouldnt have any restriction.

I think would be an option to ask a octo dev how they are setup. If you install crowsnest on a blank RaspiOS, you could use that with noproxy out of the box.

joshua76y, 2023-04-30:
> no_proxy: true

Great work! I have the same issues and that helps, thanks a lot!

KwadFan, 2023-05-22:
I think due the fact Octoprint uses the same backend as the current "release" this could be closed.
Regards Kwad