mainsail-crew/crowsnest #83: Raspberry Pi Cam 3 "No signal"

Hely0n, 2023-01-31:
### What happened

I bought the new Raspberry Pi Camera Model 3 and realized, it wasn't detected. I found out that the cam was just released 3 weeks ago, so i made an "apt update && apt full-upgrade" and it indeed updated some libcamera stuff.

crowsnest stopped working afterwards and after running the uninstall.sh and then "make install" again (with freshly pulled as from today), it was detected and shows some FPS in mainsail, but it says "no Signal" on the image.

While the crowsnest.log seems to act like everything is fine, 

"systemctl status crowsnest.service"

says this:

> Jan 31 17:41:33 Ender3 crowsnest[2865]: -- INFO  [48.408    stream] -- Requested 3 device buffers, got 3
> Jan 31 17:41:33 Ender3 crowsnest[2871]: -- ERROR [48.414    stream] -- Can't start capturing: Invalid argument
> Jan 31 17:41:34 Ender3 crowsnest[2883]: -- INFO  [48.419    stream] -- Sleeping 1 seconds before new stream init ...
> Jan 31 17:41:35 Ender3 crowsnest[2901]: -- INFO  [49.421    stream] -- Using input channel: 0
> Jan 31 17:41:36 Ender3 crowsnest[2907]: -- INFO  [49.422    stream] -- Using resolution: 1920x1080
> Jan 31 17:41:36 Ender3 crowsnest[2913]: -- INFO  [49.422    stream] -- Using format: MJPEG
> Jan 31 17:41:36 Ender3 crowsnest[2919]: -- INFO  [49.422    stream] -- Using HW FPS: 15
> Jan 31 17:41:37 Ender3 crowsnest[2925]: -- ERROR [49.422    stream] -- Device doesn't support setting of HW encoding quality parameters
> Jan 31 17:41:37 Ender3 crowsnest[2931]: -- INFO  [49.422    stream] -- Using IO method: MMAP
> Jan 31 17:41:39 Ender3 crowsnest[2951]: -- INFO  [49.442    stream] -- Device fd=8 closed

### What did you expect to happen

The new Raspberry Pi camera works plug & play like always, at least after system upgrade.

### How to reproduce

Just try to use a raspberry pi camera 3. 

Maybe the error is somewhere else but as I ordered the cam because my old Raspicam v2.1 broke by trying to fit it in an enclosue (so not software related damage) and I can't test it now.

### Additional information

crowsnest.log, just in case:

> [01/31/23 17:40:37] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[01/31/23 17:40:37] crowsnest: Version: v3.0.6-23-g00fe721
[01/31/23 17:40:37] crowsnest: Prepare Startup ...
[01/31/23 17:40:38] crowsnest: INFO: Host information:
[01/31/23 17:40:38] crowsnest: Host Info: Distribution: Debian GNU/Linux 11 (bullseye)
[01/31/23 17:40:38] crowsnest: Host Info: Release: MainsailOS release 1.0.1 (bullseye)
[01/31/23 17:40:38] crowsnest: Host Info: Kernel: Linux 5.15.84-v8+ aarch64
[01/31/23 17:40:38] crowsnest: Host Info: Model: Raspberry Pi 3 Model B Rev 1.2
[01/31/23 17:40:38] crowsnest: Host Info: Available CPU Cores: 4
[01/31/23 17:40:38] crowsnest: Host Info: Available Memory: 751596 kB
[01/31/23 17:40:38] crowsnest: Host Info: Diskspace (avail. / total): 1.3G / 7.1G
[01/31/23 17:40:38] crowsnest: INFO: Checking Dependencys
[01/31/23 17:40:38] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[01/31/23 17:40:39] crowsnest: Dependency: 'find' found in /usr/bin/find.
[01/31/23 17:40:39] crowsnest: Dependency: 'logger' found in /usr/bin/logger.
[01/31/23 17:40:39] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[01/31/23 17:40:39] crowsnest: Dependency: 'ffmpeg' found in /usr/bin/ffmpeg.
[01/31/23 17:40:39] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[01/31/23 17:40:39] crowsnest: Dependency: 'rtsp-simple-server' found in bin/rtsp-simple-server/rtsp-simple-server.
[01/31/23 17:40:40] crowsnest: Version Control: ustreamer is up to date. (v5.37)
[01/31/23 17:40:41] crowsnest: Version Control: rtsp-simple-server is up to date. (v0.20.2)
[01/31/23 17:40:46] crowsnest: Version Control: ffmpeg is up to date. (4.3.5-0+deb11u1+rpt3)
[01/31/23 17:40:46] crowsnest: INFO: Print Configfile: '/home/pi/printer_data/config/crowsnest.conf'
[01/31/23 17:40:46] crowsnest:          [crowsnest]
[01/31/23 17:40:46] crowsnest:          log_path: ~/printer_data/logs/crowsnest.log
[01/31/23 17:40:46] crowsnest:          log_level: verbose
[01/31/23 17:40:46] crowsnest:          delete_log: false
[01/31/23 17:40:46] crowsnest:
[01/31/23 17:40:46] crowsnest:          [cam 1]
[01/31/23 17:40:46] crowsnest:          mode: mjpg
[01/31/23 17:40:46] crowsnest:          port: 8080
[01/31/23 17:40:46] crowsnest:          device: /dev/video0
[01/31/23 17:40:46] crowsnest:          resolution: 1920x1080
[... 153 lines cut ...]
[01/31/23 17:40:52] crowsnest:          4: 800000 (0xc3500)
[01/31/23 17:40:52] crowsnest:          iso_sensitivity_auto 0x009a0918 (menu) : min=0 max=1 default=1 value=1
[01/31/23 17:40:52] crowsnest:          0: Manual
[01/31/23 17:40:52] crowsnest:          1: Auto
[01/31/23 17:40:52] crowsnest:          exposure_metering_mode 0x009a0919 (menu) : min=0 max=3 default=0 value=0
[01/31/23 17:40:52] crowsnest:          0: Average
[01/31/23 17:40:52] crowsnest:          1: Center Weighted
[01/31/23 17:40:52] crowsnest:          2: Spot
[01/31/23 17:40:52] crowsnest:          3: Matrix
[01/31/23 17:40:52] crowsnest:          scene_mode 0x009a091a (menu) : min=0 max=13 default=0 value=0
[01/31/23 17:40:52] crowsnest:          0: None
[01/31/23 17:40:52] crowsnest:          8: Night
[01/31/23 17:40:52] crowsnest:          11: Sports
[01/31/23 17:40:52] crowsnest:
[01/31/23 17:40:52] crowsnest:          JPEG Compression Controls
[01/31/23 17:40:52] crowsnest:
[01/31/23 17:40:52] crowsnest:          compression_quality 0x009d0903 (int) : min=1 max=100 step=1 default=30 value=30
[01/31/23 17:40:52] crowsnest: INFO: No usable CSI Devices found.
[01/31/23 17:40:52] crowsnest: V4L2 Control:
[01/31/23 17:40:53] crowsnest: No parameters set for [cam 1]. Skipped.
[01/31/23 17:40:54] crowsnest: INFO: Blockyfix: Setting video_bitrate_mode to constant.
[01/31/23 17:40:54] crowsnest: Try to start configured Cams / Services...
[01/31/23 17:41:13] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue...
[01/31/23 17:41:15] crowsnest: ... Done!
[01/31/23 17:41:16] crowsnest: Starting ustreamer with Device /dev/video0 ...

Hely0n, 2023-01-31:
I did a fresh reinstall on another Pi and it's the same and I found out it's a issue from ustreamer:
https://github.com/pikvm/ustreamer/issues/198

Is there some workaround to use until they fix it? Maybe an alternative to ustreamer or the whole crowsnest that's compatible to Mainsail/Moonraker

Hely0n, 2023-01-31:
Thanks to the swarm intelligence on the discord server, we have a workaround!!

https://github.com/roamingthings/spyglass

KwadFan, 2023-01-31:
That one will come, one day... I am bit busy right now but yeah, there is already an experimental branch ...

But..

THERE IS ABSOLUTLY NO SUPPORT FOR OTHER THAN THE STABLE VERSION!

Thanks in Advance

JD-Harding, 2023-01-31:
Talk about timing - I've been trying to get my Pi Cam 3 working for two days with no luck. This might explain it

Hely0n, 2023-01-31:
Yeah for me it was this whole day, until i looked at the discord channel. Just try out https://github.com/roamingthings/spyglass, it's very easy. I added it as a systemctl service for autostarting. One day Crowsnest is fixed and we can switch back, as there are some drawbacks like no adaptive mjpeg.

JD-Harding, 2023-01-31:
I might give that a try but, until then, I think I might just reinstall my trusty logitech until crowsnest is back on its feet.

kaibsora, 2023-02-01:
Please note, this is not an issue with crowsnest by any means. this is an issue with the newer cameras that arent able to use the old libraries(sony sensor based cameras like the pi cam 3) usb cameras should not have any issues, even using a sony sensor. Its a flaw in the way buster handles camera devices on csi now, specifically related to raspberry. this is why libcamera works. Libcamera was created to be able to use these new cams, by the raspberry pi foundation. 

Credits to @ayucam for finding this issue with the system itself, and working to implement ways around it. 
Credits to the mainsail team for their understanding of it and trying to work around it.

Caouette1988, 2023-02-08:
on this website, these guy susceffuly make work the pi cam 3 by installing a streamer in bulleye.
You can give it a try. The final working method is the post # 72

https://community.octoprint.org/t/pi-camera-v3-imx-chipset-based-cameras-not-working/49022/72

KwadFan, 2023-02-08:
There is already a sideproject for raspicam v3 ...

https://github.com/roamingthings/spyglass

KwadFan, 2023-02-08:
But at some point it will find its way to crowsnest ...

bodak, 2023-03-11:
Thanks for the workaround for raspicam V3.

I found one instruction missing from this thread and https://github.com/roamingthings/spyglass:
You have to re-enable libcamera in boot config
`sudo vi /boot/config.txt`
Under ## Disable libcamera (interferes with ustreamer, when using raspicams) change to 1
`camera_auto_detect=1`

ubrider, 2023-04-16:
Any news on an update for it ? i can't get it to work in spyglass... and want multi cam

KwadFan, 2023-04-17:
Yes, I am almost done with develop branch using camera-streamer. We found 2 little bugs, that I will fix during this week. But you could try it if you are interessted. Cannt give support yet, but it works somehow...

stabias, 2023-04-27:
I'm having problems with spyglass too. Any updates on solving this issue?

rizeddd, 2023-05-03:
Hi , any news on this ? I just bought a Pi Cam 3 and didnt check it wouldnt work :(

stabias, 2023-05-04:
> I'm having problems with spyglass too. Any updates on solving this issue?

As for now - no

bobdivx, 2023-05-04:
hello,  look this:

https://robertmarks.cloud/posts/2023/02/setting-up-pi-camera-module-3-in-mainsailos/

stabias, 2023-05-06:
`pi@mainsailos:~ $ libcamera-hello`
`Preview window unavailable`
`[0:01:57.680285034] [1433]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70`
`[0:01:57.699442216] [1434] ERROR V4L2 v4l2_device.cpp:352 'imx708': Unable to set controls: Device or resource busy`
`[0:01:57.712827110] [1434]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to ``Unicam device /dev/media4 and ISP device /dev/media1`
`[0:01:57.713162200] [1433]  INFO Camera camera.cpp:841 Pipeline handler in use by another process`
`ERROR: *** failed to acquire camera /base/soc/i2c0mux/i2c@1/imx708@1a ***`
I got this

TacoLococo, 2023-05-17:
> `pi@mainsailos:~ $ libcamera-hello` `Preview window unavailable` `[0:01:57.680285034] [1433] INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70` `[0:01:57.699442216] [1434] ERROR V4L2 [...]

Did you find a solution for this? I've got the same issue... "Device or resource busy".

meteyou, 2023-05-17:
```
Pipeline handler in use by another process
```
shutdown/stop all other services which try to open your webcam...

closed 2023-05-22 by KwadFan

Hely0n, 2023-05-22:
Just updated crowsnest to let it replace spyglass again, but it doesn't work. I managed to get it working by logs (it says ... DONE! Starting camera-streamer with Device /dev/video0 ...) But there is no image either in mainsail, nor on the mobileraker App and if I call the page manually, it says "server error", while the logs don't change and it seems like everything works

Hely0n, 2023-05-22:
Okay you have to replace the "device" parameter in the config to the one that gets listed in the log manually.  Now it works, but I think this should happen automatically and the logs shouldn't say that everything is okay 😁.

But otherwise, well done guys! I can imagine how much work was involved to make this update.

TacoLococo, 2023-05-24:
I reinstalled Mainsail OS from scratch to get mine to work. Just saved my config files, reimaged, installed Spyglass and it was all hunky-dory.

meteyou, 2023-05-25:
@TacoLococo you don't need spyglass with the new mainsailOS, because camera-streamer also support v3 cams...

Chalbo-Learning, 2023-06-02:
Hi, I had been following this closely and had been impacted by this.

There is now a fix in Crowsnest that supports Pi Camera Module 3. I prefer to use this option as it fully supports Obico where the spyglass option doesn't seem to.

**FIX**: Update Crowsnest as per guide here: https://crowsnest.mainsail.xyz/faq/upgrade-from-v3-to-v4

_NOTE_ if you upgrade from Mainsail GUI, you will have to manually uninstall the old version, read the guide very carefully!

Hope this helps someone!

commit 95c1dca1 (2023-03-11): feat!: adds camera-streamer to crowsnest

This introduces camera-streamer as streamer option
via `mode: multi`

camera-streamer is a feature packed stream service.
It is capable to deliver mjpg/snapshots/webrtc and rtsp

Limited to raspberry pi sbc's for now.

This should also resolv
Feature request #51
Feature request #37
Fixes #83
Closes #85
Fixes #89

BREAKING CHANGES:

Dropping support for Debian Buster based images and kernels older than
5.15y

Dropping RTSP support due aler9/simple-rtsp-server

Dropping usage of ffmpeg

No support anymore for Raspicam V1 (EOL)