mainsail-crew/crowsnest #84: External RTSP over network possible or planned?

erazor666, 2023-02-01:
### Is your feature request related to a problem? Please describe

Is it currently possible or planned for Crowsnest to use networked cameras? I’m no expert on required system resources for something like this to opeate, but isnt it less taxing to get a stream from a external device?

### Describe the solution you'd like

If possible support for external cameras over RTSP or the likes

KwadFan, 2023-02-02:
Currently it isn't planned, but I always look for improvements regarding the used backends.
I recently started playing around with a new one, this one is able to restream rtsp sources.

Can't promise it will come yet, but also didnt want to say "No" in general.
For now, it is a hunky fat "Maybe one day" ...

Regards Kwad

closed 2023-02-02 by KwadFan

erazor666, 2023-02-06:
Fair enough response, it may be a small number or users wanting/needing it. But then again there are many good value rtsp cams and if it doesnt tax the system much it could be good. Thank you for reply non the less. 🙂