mainsail-crew/crowsnest #85: Breaking changes on its way ....

KwadFan, 2023-02-05:
### What happened

Hey there,

I am currently working on the switch to camera-streamer as default backend Raspberry Pi SBC.

What does that mean for future releases?

1. Crowsnest is able to handle Arducams and probably the new Raspicam V3 module.
2. We have webrtc streams (Yay, finally) in future releases of mainsail.
3. We are dropping completly the support for OMX therefor "buster based images", it has to be bullseye from now on.
4. Non Rpi Devices are limited to MJPG/snapshots as previously due well known ustreamer, Install may fail because I cant test everything yet, but promised I will fix it asap, if reported here as an Issue.

If you want to stay with Version 3 please change branch to `legacy/v3` this is a snapshot of current `master` branch.
Please be aware that this branch will not recieve updates in future. It is only for those who need backwards compatibility

Stay tuned....

Regards Kwad

### What did you expect to happen

Breaking things

### How to reproduce

We will see how to reproduce ;)

JD-Harding, 2023-02-07:
Tested with RPI Cam v3 and still no joy. Good job on the update though!

jonp92, 2023-02-08:
This will break proper Obico streaming FYI. They are still dependent on 32bit RP images due to them using the OMX encoder in FFMPEG plus a 32bit version of Janus.

KwadFan, 2023-02-08:
Well, then time to switch for Obico ... OMX is not supported anymore in newer images of RaspiOS.
But FYI, we are in contact with Obico and at some point we eill join forces to use the already existing webrtc feature of camera-streamer ...

KwadFan, 2023-02-08:
> Tested with RPI Cam v3 and still no joy. Good job on the update though!

Yes, Implementation of raspicams is on my todo. My V3 Module arrived today, so only a mater of time ;)

JD-Harding, 2023-02-24:
Hey @KwadFan - just wondering if you had any update on the pi cam v3? Really excited to get it working! Thanks!

KwadFan, 2023-02-25:
> Hey @KwadFan - just wondering if you had any update on the pi cam v3? Really excited to get it working! Thanks!

Please headover to https://github.com/roamingthings/spyglass for now. This streamer will find its way also to crowsnest.
Use the method to install as Service from their README and disable crowsnest or manage settings with different ports, then they can run side by side

camera-streamer needs currently some ayufan, which is working hard on it, to fully be integrated. Also we have an issue with compiling on older Models of the pi ( due lag of ram and/or compute power) that has to be solved.

Hopefully we can get it managed in the next weeks.

JD-Harding, 2023-02-25:
I'm in no rush and happy to wait (although I am excited to get it working). Just curious as to how you guys were getting on!

KwadFan, 2023-03-11:
Hey there :wave: ,

today I merged the camera-streamer branch to develop, so feel free to test and report back if any bugs are found, I want to iron them out completly before merging to master.

This is BREAKING!!!! uninstall crowsnest completly before testing!!!

Will work on documentation, for now you have to use `mode: multi` to use camera-streamer, mjpg stays ustreamer.

**NOTE:** Raspicam V1 and OMX support are completly dropped! You need to use the legacy/v3 branch to stay compatible, also this will only work on bullseye images with an 5.15.84 kernel at least!

Regards Kwad

JD-Harding, 2023-03-15:
Thanks! Not going to have much time to test this but looking forward to it hitting master!

benphelps, 2023-03-17:
Running under Ubuntu 22.04.2 LTS I had to add the focal universe repo to install the `liblivemedia-dev` package, it seems to not be part of the jammy universe repo,

```sh
sudo add-apt-repository 'deb http://ports.ubuntu.com/ubuntu-ports focal universe'
```

other than that it built fine.  In mainsail, I updated the URL Stream value to `/webcam/webrtc` and everything is working great.

benphelps, 2023-03-17:
Leaving the URL Stream as the default `/webcam/?action=stream` does not work for me, and I get a 502 Bad Gateway error from Nginx.  The upstream server sends back an empty response, the POST action needs to be sent to `/webcam/webrtc`.

I'm not sure if this is expected behaviour, but I made a pull request that allows the default Mainsail values to continue working here https://github.com/ayufan/camera-streamer/pull/49

BreeOge, 2023-03-19:
> Leaving the URL Stream as the default `/webcam/?action=stream` does not work for me, and I get a 502 Bad Gateway error from Nginx. The upstream server sends back an empty response, the POST action needs to be sent to `/webcam/webrtc`.
> 
> I'm not sure if this is expected behaviour, but I made a pull request that allows the default Mainsail values to continue working here [ayufan/camera-streamer#49](https://github.com/ayufan/camera-streamer/pull/49)

Check your v4l2ctl settings.  Something changed just the other day and it has affected some of the settings.  The setting names changed as well.  Mine was doing the same thing till I noticed it.

benphelps, 2023-03-20:
> Check your v4l2ctl settings. Something changed just the other day and it has affected some of the settings. The setting names changed as well. Mine was doing the same thing till I noticed it.

Empty for me, the C270 can't be helped with any of the settings, it's just all around poor quality 😂

My issue was just to do with using WebRTC as the service, and leaving the other settings as their defaults.  If camera-streamer merges that PR, then it'll work like in the image, otherwise users will have to change the stream URL to `/webcam/webrtc`.

[image]

meteyou, 2023-03-20:
@benphelps users have to change the url to /webcam/webrtc every time, when they want to use camera-streamer with webrtc. the merge will noch redirect anything of this streamer.

benphelps, 2023-03-20:
> @benphelps users have to change the url to /webcam/webrtc every time, when they want to use camera-streamer with webrtc. the merge will noch redirect anything of this streamer.

It just extends the already existing compatibility layer to also handle POST requests to that endpoint.  The user would not have to change anything if it's accepted.

I'm just saying it doesn't _have_ to be that way, it can work without needing to change the URL every time, and it's a very simple patch to camera-streamer.

benphelps, 2023-03-20:
Just to expand a little, `/webcam/?action=stream` is already a redirect to `/webcam/stream`.  So adding another redirect to handle the WebRTC case isn't going against already existing behaviour.

KwadFan, 2023-03-23:
> Just to expand a little, `/webcam/?action=stream` is already a redirect to `/webcam/stream`.  So adding another redirect to handle the WebRTC case isn't going against already existing behaviour.

This is out of my scope. Please head to ayufan/camera-streamer to get that feature.

@ayufan would this be possible?

benphelps, 2023-03-23:
> This is out of my scope. Please head to ayufan/camera-streamer to get that feature.
>
> @ayufan would this be possible?

I already made a pull request to add that (https://github.com/ayufan/camera-streamer/pull/49), I was just updating this thread on the problems I ran into setting up the new branch.

KwadFan, 2023-04-10:
> Running under Ubuntu 22.04.2 LTS I had to add the focal universe repo to install the `liblivemedia-dev` package, it seems to not be part of the jammy universe repo,
> 
> ```shell
> sudo add-apt-repository 'deb http://ports.ubuntu.com/ubuntu-ports focal universe'
> ```
> 
> other than that it built fine. In mainsail, I updated the URL Stream value to `/webcam/webrtc` and everything is working great.

Good point, could you please share some output? Need some hooks to identify board and distro, then I could do a "workaround" inside the installer...

`grep "ubuntu" /etc/os-release` and path to config.txt(or an equivalent) to it.
Need at least to hooks to clearly differentiate from a Buntu x86

Regards Kwad

KwadFan, 2023-04-10:
As a notice. Merged new installer to develop.

Things to do are:
1. Adding function to bump `gpu_mem`
2. Changing `mode` to more appropriate `streamer` keyword
3. "Ubuntu ARM Workaround"

Tested install on Debian 11 VM and Raspberrpi OS Lite.
Has to be tested on Ubuntu ARM.

Regards Kwad

KwadFan, 2023-04-13:
I did it test by myself with Ubuntu Server ARM 22.10 and the last version that contains `liblivemedia-dev` is focal aka 20.04.

Which in the end leads me to say sorry but there will be no camera-streamer support in Ubuntu ARM, because I dont want to add repositorys from outdated versions to actual versions.

Regards Kwad

ayufan, 2023-04-13:
@KwadFan 

> Which in the end leads me to say sorry but there will be no camera-streamer support in Ubuntu ARM, because I dont want to add repositorys from outdated versions to actual versions.

`camera-streamer` does not require `liblivemedia-dev` to be present. If it will be not found no RTSP backend will be compiled.

KwadFan, 2023-04-13:
> @KwadFan
> 
> > Which in the end leads me to say sorry but there will be no camera-streamer support in Ubuntu ARM, because I dont want to add repositorys from outdated versions to actual versions.
> 
> `camera-streamer` does not require `liblivemedia-dev`. If it will be not found no RTSP backend will be compiled in.

Yes, but camera-streamer is now also my RTSP Server :) So, hard decision, but somehow I have to ensure its working.
And I simply dont want to install outdated Repos to the distro.

ayufan, 2023-04-13:
@KwadFan 

I could possible compile in `liblivemedia`. Unsure if this is worth though :)

KwadFan, 2023-04-13:
We could keep that in mind for future, for sure, for now lets get this thing public asap :P

gifter77, 2023-04-20:
I was testing camera-streamer in the develop branch yesterday on my MainsailOS 1.1.1 on pi 4 with pi camera module v2. The webrtc stream hangs unless I comment out the --camera-format=YUYV option FYI.

KwadFan, 2023-04-20:
> I was testing camera-streamer in the develop branch yesterday on my MainsailOS 1.1.1 on pi 4 with pi camera module v2. The webrtc stream hangs unless I comment out the --camera-format=YUYV option.
> 
> This is my crowsnet config:
> 
> ```
> [crowsnest]
> log_path: /home/pi/printer_data/logs/crowsnest.log
> log_level: verbose                      # Valid Options are quiet/verbose/debug
> delete_log: false                       # Deletes log on every restart, if set to true
> no_proxy: false
> 
> [cam 1]
> mode: camera-streamer                   # ustreamer - Provides mjpg and snapshots. (All devices)
>                                         # camera-streamer - Provides webrtc, mjpg and snapshots. (rpi + Raspi OS based only)
> enable_rtsp: true                       # If camera-streamer is used, this enables also usage of an rtsp server
> rtsp_port: 8554                         # Set different ports for each device!
> port: 8080                              # HTTP/MJPG Stream/Snapshot Port
> device: /base/soc                       # See Log for available ...
> resolution: 1640x1232                   # widthxheight format
> max_fps: 30                             # If Hardware Supports this it will be forced, otherwise ignored/coerced.
> #custom_flags:                          # You can run the Stream Services with custom flags.
> #v4l2ctl:                               # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.
> ```
> 
> Obviously I have changed my boot config to enable libcamera.

I bet its because your misconfiguration ...

```device: /base/soc``` 

This is not a valid device path, please see your log for the correct path.
Tested it right now with Raspicam v1.3 v2 and v3, cant reproduce..
Regards Kwad

gifter77, 2023-04-20:
Hi Kwad,

Thanks for the reply.

Correcting the device path doesn't fix it. Only commenting out `start_param+=( --camera-format=YUYV )` in `libs/camera-streamer.sh` fixes it somehow...

gifter77, 2023-04-20:
More details, with the camera format as YUYV, I get the following in debug logs non stop and no webrtc video:
```
[04/20/23 19:27:18] crowsnest: DEBUG: camera-streamer [cam 1]: device/links.c: VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 446654326, vs 438681259.
[04/20/23 19:27:20] crowsnest: DEBUG: camera-streamer [cam 1]: device/links.c: VIDEO:capture:mplane:buf1: Capture image is outdated. Skipped. Now: 448654399, vs 442680568.
[04/20/23 19:27:22] crowsnest: DEBUG: camera-streamer [cam 1]: device/links.c: VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 450654376, vs 444680217.
[04/20/23 19:27:24] crowsnest: DEBUG: camera-streamer [cam 1]: device/links.c: VIDEO:capture:mplane:buf1: Capture image is outdated. Skipped. Now: 452654396, vs 446679869.
[04/20/23 19:27:26] crowsnest: DEBUG: camera-streamer [cam 1]: device/links.c: VIDEO:capture:mplane:buf0: Capture image is outdated. Skipped. Now: 454654413, vs 448679521.
[04/20/23 19:27:28] crowsnest: DEBUG: camera-streamer [cam 1]: device/links.c: VIDEO:capture:mplane:buf1: Capture image is outdated. Skipped. Now: 456654404, vs 450679171.
```

KwadFan, 2023-04-20:
> More details, with the camera format as YUYV, I get the following in debug logs non stop and no webrtc video:
>
> ```
> [04/20/23 19:27:18] crowsnest: DEBUG: camera-streamer [cam 1]: device/links.c: [...]

@ayufan Is that something to worry about?

KwadFan, 2023-04-20:
> Hi Kwad,
> 
> Thanks for the reply.
> 
> Correcting the device path doesn't fix it. Only commenting out `start_param+=( --camera-format=YUYV )` in `libs/camera-streamer.sh` fixes it somehow...

Dont know why it doesnt work for you, but this is the recommended setup for your cam by camera-streamer
https://github.com/ayufan/camera-streamer/blob/master/service/camera-streamer-raspi-v2-8MP.service

And as already said, cant reproduce it, without my V2 Cam does not work properly

gifter77, 2023-04-20:
Thanks, it works properly if I add  `--camera-format=NV12` to custom_flags. But that seems very odd that you need YUYV for your V2 CAM to work and I need NV12...

ayufan, 2023-04-20:
Bandwidth issue? YUYV requires 2x NV12 BW.

What does `tools/dump_cameras.sh` from `camera-streamer` says?

gifter77, 2023-04-20:
<details>
  <summary>
    dump_camera.sh output
  </summary>
<code>
++ cat /sys/firmware/devicetree/base/model
++ tr -d '\0'
+ echo 'Raspberry Pi 4 Model B Rev 1.5'
Raspberry Pi 4 Model B Rev 1.5
+ uname -a
Linux voron 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
+ v4l2-ctl --list-devices
v4l2-ctl | bcm2835-codec-decode (platform:bcm2835-codec):
v4l2-ctl | 	/dev/video10
v4l2-ctl | 	/dev/video11
v4l2-ctl | 	/dev/video12
v4l2-ctl | 	/dev/video18
v4l2-ctl | 	/dev/video31
v4l2-ctl | 	/dev/media2
v4l2-ctl | 
v4l2-ctl | bcm2835-isp (platform:bcm2835-isp):
v4l2-ctl | 	/dev/video13
v4l2-ctl | 	/dev/video14
v4l2-ctl | 	/dev/video15
v4l2-ctl | 	/dev/video16
v4l2-ctl | 	/dev/video20
v4l2-ctl | 	/dev/video21
v4l2-ctl | 	/dev/video22
v4l2-ctl | 	/dev/video23
v4l2-ctl | 	/dev/media1
v4l2-ctl | 	/dev/media3
v4l2-ctl | 
v4l2-ctl | unicam (platform:fe801000.csi):
v4l2-ctl | 	/dev/video0
v4l2-ctl | 	/dev/video1
v4l2-ctl | 	/dev/media4
v4l2-ctl | 
v4l2-ctl | rpivid (platform:rpivid):
v4l2-ctl | 	/dev/video19
v4l2-ctl | 	/dev/media0
v4l2-ctl | 
+ libcamera-hello --list-cameras
libcamera | Available cameras
libcamera | -----------------
libcamera | 0 : imx219 [3280x2464] (/base/soc/i2c0mux/i2c@1/imx219@10)
libcamera |     Modes: 'SRGGB10_CSI2P' : 640x480 [103.33 fps - (1000, 752)/1280x960 crop]
libcamera |                              1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
libcamera |                              1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
libcamera |                              3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
libcamera |            'SRGGB8' : 640x480 [103.33 fps - (1000, 752)/1280x960 crop]
libcamera |                       1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
libcamera |                       1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
libcamera |                       3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
====================================
DEVICE: /dev/video0
====================================
+ v4l2-ctl -d /dev/video0 --info --list-formats-ext --list-fields --list-formats-out --list-fields-out
/dev/video0 | Driver Info:
/dev/video0 | 	Driver name      : unicam
/dev/video0 | 	Card type        : unicam
/dev/video0 | 	Bus info         : platform:fe801000.csi
/dev/video0 | 	Driver version   : 6.1.21
/dev/video0 | 	Capabilities     : 0xa5a00001
/dev/video0 | 		Video Capture
/dev/video0 | 		Metadata Capture
/dev/video0 | 		Read/Write
/dev/video0 | 		Streaming
/dev/video0 | 		Extended Pix Format
/dev/video0 | 		Device Capabilities
/dev/video0 | 	Device Caps      : 0x25200001
[... 1087 lines cut ...]
/dev/video31 | 	  Link 0x02000040: to remote pad 0x100003c of entity 'bcm2835-codec-encode_image-proc': Data, Enabled, Immutable
/dev/video31 | ioctl: VIDIOC_ENUM_FMT
/dev/video31 | 	Type: Video Capture Multiplanar
/dev/video31 | 
/dev/video31 | 	[0]: 'JPEG' (JFIF JPEG, compressed)
/dev/video31 | 		Size: Stepwise 32x32 - 1920x1920 with step 2/2
/dev/video31 | Supported Video Fields:
/dev/video31 | 	None
/dev/video31 | ioctl: VIDIOC_ENUM_FMT
/dev/video31 | 	Type: Video Output Multiplanar
/dev/video31 | 
/dev/video31 | 	[0]: 'RGBP' (16-bit RGB 5-6-5)
/dev/video31 | 	[1]: 'RGB3' (24-bit RGB 8-8-8)
/dev/video31 | 	[2]: 'BGR3' (24-bit BGR 8-8-8)
/dev/video31 | 	[3]: 'AB24' (32-bit RGBA 8-8-8-8)
/dev/video31 | 	[4]: 'BGR4' (32-bit BGRA/X 8-8-8-8)
/dev/video31 | 	[5]: 'YU12' (Planar YUV 4:2:0)
/dev/video31 | 	[6]: 'YUYV' (YUYV 4:2:2)
/dev/video31 | 	[7]: 'YVYU' (YVYU 4:2:2)
/dev/video31 | 	[8]: 'VYUY' (VYUY 4:2:2)
/dev/video31 | 	[9]: 'UYVY' (UYVY 4:2:2)
/dev/video31 | Supported Video Output Fields:
/dev/video31 | 	None
</code>
</details>

ayufan, 2023-04-20:
Hmm. I don't know. Maybe we should apply default to be `YU12` or `NV12`? They will work fine as well for camera output.

gifter77, 2023-04-20:
Also I forgot to mention my installation is 64 bits.

ayufan, 2023-04-20:
Nah. Impossible. Interesting. Can you try `YUV420` ?

gifter77, 2023-04-20:
YUV420 works. What is impossible?

KwadFan, 2023-04-20:
I dont mind to change, based on your recommendation @ayufan

gifter77, 2023-04-20:
Also @ayufan why does the video output automatically gets configured to 1437x1080 when the h264 encode should be able to handle more than that?

ayufan, 2023-04-20:
@gifter77 

The `1640x1232` is larger than the default sets via `--camera-snapshot.height=1080`.

closed 2023-05-22 by KwadFan

ytugarev, 2023-05-24:
I thought it would be useful for ppl with old raspi cams [https://github.com/mainsail-crew/crowsnest/issues/112#issuecomment-1560417914](https://github.com/mainsail-crew/crowsnest/issues/112#issuecomment-1560417914) @KwadFan thanks a lot - my old v1 raspberri pi camera (and Pi itself is also quite old - v2 B) works again on a new stack.

`[cam 1]
mode: camera-streamer                     # ustreamer - Provides mjpg and snapshots. (All devices)
                                          # camera-streamer - Provides webrtc, mjpg and snapshots. (rpi + Raspi OS based only)
enable_rtsp: false                        # If camera-streamer is used, this enables also usage of an rtsp server
rtsp_port: 8554                           # Set different ports for each device!
port: 8080                                # HTTP/MJPG Stream/Snapshot Port
device: /base/soc/i2c0mux/i2c@1/ov5647@36 # See Log for available ...
resolution: 2592x1944                     # widthxheight format
max_fps: 15                               # If Hardware Supports this it will be forced, otherwise ignored/coerced.
#custom_flags:                            # You can run the Stream Services with custom flags.
#v4l2ctl:                                 # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.`

KwadFan, 2023-05-24:
> I thought it would be useful for ppl with old raspi cams [...]

Will update the docs, and use your setup as example, if that is ok.

MikeEitel, 2023-05-24:
HI
I have a strange issue with Manta-8M CM4 and Raspicam V3
Tried different configs with    device:  /dev/video0  # and also /dev/v4l/by-path/platform-fe801000.csi-video-index0  
Other device: xxxx  show complete errors to device is probably addressed correctly in both versions.

Also locking for the camera in system seems ok
pi@MeVoron:~ $ v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12
        /dev/video18
        /dev/video31
        /dev/media3

bcm2835-isp (platform:bcm2835-isp):
        /dev/video13
        /dev/video14
        /dev/video15
        /dev/video16
        /dev/video20
        /dev/video21
        /dev/video22
        /dev/video23
        /dev/media1
        /dev/media2

unicam (platform:fe801000.csi):
        /dev/video0
        /dev/video1
        /dev/media4

rpivid (platform:rpivid):
        /dev/video19
        /dev/media0

AND HERE THE long DEBUG LOG:
05/24/23 13:22:19] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[05/24/23 13:22:19] crowsnest: Version: v4.0.2
[05/24/23 13:22:19] crowsnest: Prepare Startup ...
[05/24/23 13:22:19] crowsnest: INFO: Host information:
[05/24/23 13:22:19] crowsnest: Host Info: Distribution: Raspbian GNU/Linux 11 (bullseye)
[05/24/23 13:22:19] crowsnest: Host Info: Release: MainsailOS release 1.0.1 (bullseye)
[05/24/23 13:22:19] crowsnest: Host Info: Kernel: Linux 6.1.21-v8+ aarch64
[05/24/23 13:22:19] crowsnest: Host Info: Model:  Raspberry Pi Compute Module 4 Rev 1.0
[05/24/23 13:22:19] crowsnest: Host Info: Available CPU Cores: 4
[05/24/23 13:22:19] crowsnest: Host Info: Available Memory: 3705504 kB
[05/24/23 13:22:19] crowsnest: Host Info: Diskspace (avail. / total): 20G / 29G
[05/24/23 13:22:19] crowsnest: INFO: Checking Dependencys
[05/24/23 13:22:19] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[05/24/23 13:22:19] crowsnest: Dependency: 'find' found in /usr/bin/find.
[05/24/23 13:22:19] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[05/24/23 13:22:19] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[05/24/23 13:22:19] crowsnest: Dependency: 'camera-streamer' found in bin/camera-streamer/camera-streamer.
[05/24/23 13:22:20] crowsnest: Version Control: ustreamer is up to date. (v5.38)
[05/24/23 13:22:20] crowsnest: Version Control: camera-streamer is up to date. ((9106d2c))
[05/24/23 13:22:20] crowsnest: INFO: Print Configfile: '/home/pi/printer_data/config/crowsnest.conf'
[05/24/23 13:22:20] crowsnest:		[crowsnest]
[05/24/23 13:22:20] crowsnest:		log_path: /home/pi/printer_data/logs/crowsnest.log
[05/24/23 13:22:20] crowsnest:		log_level: debug
[05/24/23 13:22:20] crowsnest:		delete_log: true
[05/24/23 13:22:20] crowsnest:		no_proxy: false
[05/24/23 13:22:20] crowsnest:		
[05/24/23 13:22:20] crowsnest:		[cam 1]
[05/24/23 13:22:20] crowsnest:		mode: camera-streamer
[05/24/23 13:22:20] crowsnest:		
[05/24/23 13:22:20] crowsnest:		enable_rtsp: false
[05/24/23 13:22:20] crowsnest:		rtsp_port: 8554
[05/24/23 13:22:20] crowsnest:		port: 8080
[05/24/23 13:22:20] crowsnest:		device:  /dev/video0
[... 54 lines cut ...]
[05/24/23 13:22:24] crowsnest: DEBUG: camera-streamer [cam 1]: device/buffer_list.c: VIDEO:capture:mplane: Streaming started... Was 0 of 3 enqueud
[05/24/23 13:22:24] crowsnest:  ... Done!
[05/24/23 13:22:37] crowsnest: DEBUG: camera-streamer [cam 1]: util/http/http.c: HTTP8080/0: Client connected 127.0.0.1 (fd=4).
[05/24/23 13:22:37] crowsnest: DEBUG: camera-streamer [cam 1]: util/http/http.c: HTTP8080/0: Request 'GET' '/' 'action=stream'
[05/24/23 13:22:37] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
[05/24/23 13:22:38] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
[05/24/23 13:22:38] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
[05/24/23 13:22:38] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
[05/24/23 13:22:38] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
[05/24/23 13:22:38] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
[05/24/23 13:22:38] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
[05/24/23 13:22:38] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
[05/24/23 13:22:38] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
[05/24/23 13:22:38] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
[05/24/23 13:22:38] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
[05/24/23 13:22:39] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
[... cut ...]
[05/24/23 13:22:39] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
[05/24/23 13:22:39] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
[05/24/23 13:22:39] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/buffer.c: CAMERA:capture:buf0: ioctl(ret=-1, errno=22): Can't queue buffer.
[05/24/23 13:22:39] crowsnest: DEBUG: camera-streamer [cam 1]: util/http/http.c: HTTP8080/0: Client disconnected 127.0.0.1.

ytugarev, 2023-05-24:
> HI I have a strange issue with Manta-8M CM4 and Raspicam V3 Tried different configs with device: /dev/video0 # and also /dev/v4l/by-path/platform-fe801000.csi-video-index0 Other device: xxxx show [...]

Try /base/soc/i2c0mux/i2c@1/imx708@1a as a device and use "mode: camera-streamer "

ytugarev, 2023-05-24:
> > I thought it would be useful for ppl with old raspi cams [#112 (comment)](https://github.com/mainsail-crew/crowsnest/issues/112#issuecomment-1560417914) @KwadFan thanks a lot - my old v1 raspberri pi camera (and Pi itself is also quite old - v2 B) works again on a new stack.
> > ```
> > [cam 1]
> > mode: camera-streamer              # ustreamer - Provides mjpg and snapshots. (All devices)
> >                                                     # camera-streamer - Provides webrtc, mjpg and snapshots. (rpi + Raspi OS based only)
> > enable_rtsp: false                        # If camera-streamer is used, this enables also usage of an rtsp server
> > rtsp_port: 8554                           # Set different ports for each device!
> > port: 8080                                # HTTP/MJPG Stream/Snapshot Port
> > device: /base/soc/i2c0mux/i2c@1/ov5647@36  # See Log for available ...
> > resolution: 2592x1944                     # widthxheight format
> > max_fps: 15                               # If Hardware Supports this it will be forced, otherwise ignored/coerced.
> > #custom_flags:                            # You can run the Stream Services with custom flags.
> > #v4l2ctl:                                 # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.
> > ```
> 
> Will update the docs, and use your setup as example, if that is ok.

Sure! This one below is better from CPU utilization (Again RPi v2 B). With the one below ~10% CPU load when watching camera, and if 2592x1944 - it is easely goes up to 30%.
```
[cam 1]
mode: camera-streamer                     # ustreamer - Provides mjpg and snapshots. (All devices)
                                          # camera-streamer - Provides webrtc, mjpg and snapshots. (rpi + Raspi OS based only)
enable_rtsp: false                        # If camera-streamer is used, this enables also usage of an rtsp server
rtsp_port: 8554                           # Set different ports for each device!
port: 8080                                # HTTP/MJPG Stream/Snapshot Port
device: /base/soc/i2c0mux/i2c@1/ov5647@36 # See Log for available ...
resolution: 1280x720                      # widthxheight format
max_fps: 15                               # If Hardware Supports this it will be forced, otherwise ignored/coerced.
#custom_flags:                            # You can run the Stream Services with custom flags.
#v4l2ctl:                                 # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.
```
Also as a pre-check has to work correctly
```
libcamera-still -o img.jpg -v
```
I do not update Rasp OS due to the fact it bricks my setup but I also use these settings (not sure if everything absolutely necessary but it works)
```
## Disable libcamera (interferes with ustreamer, when using raspicams)
camera_auto_detect=0
dtoverlay=ov5647

## Enable VideoCore at boot, needed for Crowsnest (Raspicams and DSI devices).
#start_x=1

```

BlackNet, 2023-05-25:
this is the compile errors.

`make[1]: Leaving directory '/home/pi/crowsnest/bin/camera-streamer'
make: *** [Makefile:57: all] Error 2
make: *** [Makefile:17: install] Error 2
`

going in scrollback here is the errors.

`make[1]: Leaving directory '/home/pi/crowsnest/bin/ustreamer'
Build 'ustreamer' ... [SUCCESS]
Build 'camera-streamer' using 4 Cores ...
fatal: No names found, cannot describe anything.
echo "#define GIT_VERSION \"\"\n#define GIT_REVISION \"f0634f9\"" > version.h.tmp; \
        diff -u version.h version.h.tmp || mv version.h.tmp version.h; \
        rm -f version.h.tmp
diff: version.h: No such file or directory
`

`cc -std=gnu17 -MMD -Werror -Wall -g -I/home/pi/crowsnest/bin/camera-streamer -D_GNU_SOURCE -DUSE_FFMPEG -DUSE_LIBCAMERA -I/usr/include/libcamera  -DUSE_LIBDATACHANNEL -Ithird_party/libdatachannel/include -Ithird_party/libdatachannel/deps/json/include -c -o util/http/http_methods.o util/http/http_methods.c
util/ffmpeg/remuxer.c: In function ‘ffmpeg_remuxer_open’:
util/ffmpeg/remuxer.c:97:33: error: initialization discards ‘const’ qualifier from pointer target type [-Werror=discarded-qualifiers]
   97 |   AVInputFormat *input_format = av_find_input_format(remuxer->input_format);
      |                                 ^~~~~~~~~~~~~~~~~~~~
`

the ~~~~~~~~~~~~~~~~~~~ is under av_find_input_format  fyi.

thetic, 2023-05-30:
I cannot get my picam working after the update.

```ini
[crowsnest]
log_path: /home/pi/printer_data/logs/crowsnest.log
log_level: verbose                      # Valid Options are quiet/verbose/debug
delete_log: false                       # Deletes log on every restart, if set to true
no_proxy: false

[cam pi]
mode: camera-streamer
enable_rtsp: false
rtsp_port: 8554
port: 8080
device: /base/soc/i2c0mux/i2c@1/imx219@10
resolution: 1640x1232
max_fps: 15

[cam angry]
mode: camera-streamer
port: 8081
device: /dev/v4l/by-id/usb-Sonix_Technology_Co.__Ltd._USB_2.0_1080P_Camera_SN0001-video-index0
resolution: 1600x1200
max_fps: 15
v4l2ctl: focus_automatic_continuous=0,focus_absolute=30
```

```
$ libcamera-hello --list-cameras
Available cameras
-----------------
0 : imx219 [3280x2464] (/base/soc/i2c0mux/i2c@1/imx219@10)
    Modes: 'SRGGB10_CSI2P' : 640x480 [103.33 fps - (1000, 752)/1280x960 crop]
                             1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                             1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                             3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
           'SRGGB8' : 640x480 [103.33 fps - (1000, 752)/1280x960 crop]
                      1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                      1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                      3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
```

```
$ libcamera-still -o img.jpg -v
Options:
    verbose: 2
    info_text:#%frame (%fps fps) exp %exp ag %ag dg %dg
    timeout: 5000
    width: 0
    height: 0
    output: img.jpg
    post_process_file:
    rawfull: 0
    preview: default
    qt-preview: 0
    transform: identity
    roi: all
    metering: centre
    exposure: normal
    ev: 0
    awb: auto
    flush: false
    wrap: 0
    brightness: 0
    contrast: 1
    saturation: 1
    sharpness: 1
    framerate: 30
    denoise: auto
    viewfinder-width: 0
    viewfinder-height: 0
    tuning-file: (libcamera)
    lores-width: 0
    lores-height: 0
    autofocus-range: normal
    autofocus-speed: normal
    autofocus-window: all
    mode: unspecified
    viewfinder-mode: unspecified
    metadata:
    metadata-format: json
    encoding: jpg
    quality: 93
    raw: 0
    restart: 0
    timelapse: 0
    framestart: 0
    datetime: 0
    timestamp: 0
    keypress: 0
    signal: 0
    thumbnail width: 320
    thumbnail height: 240
    thumbnail quality: 70
    latest:
    immediate 0
    AF on capture: 0
Preview window unavailable
Running without preview window
Opening camera...
[0:06:52.587079820] [2610]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
[0:06:52.627759024] [2614]  WARN RPI raspberrypi.cpp:1357 Mismatch between Unicam and CamHelper for embedded data usage!
[0:06:52.629342753] [2614]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media2 and ISP device /dev/media0
Acquired camera /base/soc/i2c0mux/i2c@1/imx219@10
Configuring viewfinder...
Viewfinder size chosen is 1640x1232
[0:06:52.632676414] [2610]  INFO Camera camera.cpp:1028 configuring streams: (0) 1640x1232-YUV420
[0:06:52.633194188] [2614]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
Camera streams configured
Available controls:
    AeEnable : [false..true]
    AeMeteringMode : [0..3]
    AeConstraintMode : [0..3]
    ExposureValue : [-8.000000..8.000000]
    ScalerCrop : [(0, 0)/128x128..(0, 0)/3280x2464]
    Sharpness : [0.000000..16.000000]
    Contrast : [0.000000..32.000000]
    NoiseReductionMode : [0..4]
    FrameDurationLimits : [23894..11767556]
    Brightness : [-1.000000..1.000000]
    ExposureTime : [75..11766829]
    AwbMode : [0..7]
    ColourGains : [0.000000..32.000000]
    AwbEnable : [false..true]
    AnalogueGain : [1.000000..10.666667]
    ColourCorrectionMatrix : [-16.000000..16.000000]
    AeExposureMode : [0..3]
    Saturation : [0.000000..32.000000]
[0:06:52.644651050] [2614] ERROR V4L2 v4l2_videodevice.cpp:1248 /dev/video14[14:cap]: Not enough buffers provided by V4L2VideoDevice
Closing Libcamera application(frames displayed 0, dropped 0)
Camera stopped!
Tearing down requests, buffers and configuration
Camera closed
ERROR: *** failed to allocate capture buffers ***
```

pl4nkt0n, 2023-06-01:
Experiencing  the same as @thetic on 
Version: v0.11.0-210-g0539e9f2-dirty
OS: Debian GNU/Linux 11 (bullseye)
Distro: MainsailOS 1.2.0 (bullseye):

-  same camera and same config
- pi CM4
- the pi camera can be found with libcamera-hello --list-camera
- crowsnest log reports that it found a pi cam /base/soc/i2c0mux/i2c@1/imx219@10, 
---but fails to start:
----crowsnest: DEBUG: camera-streamer [cam raspi]: xargs: /home/pi/crowsnest/bin/camera-streamer/camera-streamer: exited with status 255; aborting
----crowsnest: ERROR: Start of camera-streamer [cam raspi] failed!

also libcamera-still fails.

USB Cam on the same setup works fine.
before, both cams (USB + raspi simultaneously) worked fine - raspi cam also still works on another setup with a pi4.
so it's not a hardware problem.

/boot/config.txt was edited as described in crowsnest FAQ:
dtoverlay=vc4-fkms-v3d
camera_auto_detect=1
#start_x=1

anyone else struggling with this setup or can find a mistake from myself?

MrDananas, 2023-06-24:
> @gifter77
> 
> The `1640x1232` is larger than the default set via `--camera-snapshot.height=1080`.

Hello! How do I go about changing the default? I am trying the enable the full resolution of my RPi HQ camera (4056x3040@10fps) and not having much luck. I tried adding "--camera-snapshot.height=3040" as a flag but nothing. Everything works without it of course but it seems the resolutions gets limited despite "resolution: 4056x3040". I could get the full resolution when running OctoPrint + Klipper.
This is my config:
```
[crowsnest]
log_path: /home/pi/printer_data/logs/crowsnest.log
log_level: debug                                          # Valid Options are quiet/verbose/debug
delete_log: false                                         # Deletes log on every restart, if set to true
no_proxy: false

[cam 1]
mode: camera-streamer                                     # ustreamer - Provides mjpg and snapshots. (All devices)
                                                          # camera-streamer - Provides webrtc, mjpg and snapshots. (rpi + Raspi OS based only)
enable_rtsp: false                                        # If camera-streamer is used, this enables also usage of an rtsp server
rtsp_port: 8554                                           # Set different ports for each device!
port: 8080                                                # HTTP/MJPG Stream/Snapshot Port
device: /base/soc/i2c0mux/i2c@1/imx477@1a                 # See Log for available ...
resolution: 4056x3040                                     # widthxheight format
max_fps: 10                                               # If Hardware Supports this it will be forced, otherwise ignored/coerced.
custom_flags: --camera-snapshot.height=3040               # You can run the Stream Services with custom flags.
#v4l2ctl:                                                 # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.
```
Here is the log:
```
[06/24/23 16:15:55] crowsnest: Shutdown or Killed by User!
[06/24/23 16:15:55] crowsnest: Please come again :)
[06/24/23 16:15:55] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[06/24/23 16:15:55] crowsnest: Version: v4.0.2-2-g8b54eae
[06/24/23 16:15:55] crowsnest: Prepare Startup ...
[06/24/23 16:15:55] crowsnest: INFO: Host information:
[06/24/23 16:15:55] crowsnest: Host Info: Distribution: Debian GNU/Linux 11 (bullseye)
[06/24/23 16:15:55] crowsnest: Host Info: Release: MainsailOS release 1.2.1 (bullseye)
[06/24/23 16:15:55] crowsnest: Host Info: Kernel: Linux 6.1.21-v8+ aarch64
[06/24/23 16:15:55] crowsnest: Host Info: Model:  Raspberry Pi 4 Model B Rev 1.2
[06/24/23 16:15:55] crowsnest: Host Info: Available CPU Cores: 4
[06/24/23 16:15:55] crowsnest: Host Info: Available Memory: 3705524 kB
[06/24/23 16:15:55] crowsnest: Host Info: Diskspace (avail. / total): 107G / 117G
[06/24/23 16:15:55] crowsnest: INFO: Checking Dependencys
[06/24/23 16:15:55] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[06/24/23 16:15:55] crowsnest: Dependency: 'find' found in /usr/bin/find.
[06/24/23 16:15:56] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[06/24/23 16:15:56] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[06/24/23 16:15:56] crowsnest: Dependency: 'camera-streamer' found in bin/camera-streamer/camera-streamer.
[06/24/23 16:15:56] crowsnest: Version Control: ustreamer is up to date. (v5.41)
[06/24/23 16:15:56] crowsnest: Version Control: camera-streamer is up to date. ((ef8f6a4))
[06/24/23 16:15:56] crowsnest: INFO: Print Configfile: '/home/pi/printer_data/config/crowsnest.conf'
[06/24/23 16:15:56] crowsnest:		[crowsnest]
[06/24/23 16:15:56] crowsnest:		log_path: /home/pi/printer_data/logs/crowsnest.log
[06/24/23 16:15:56] crowsnest:		log_level: debug
[06/24/23 16:15:56] crowsnest:		delete_log: false
[06/24/23 16:15:56] crowsnest:		no_proxy: false
[06/24/23 16:15:56] crowsnest:		
[06/24/23 16:15:56] crowsnest:		[cam 1]
[06/24/23 16:15:56] crowsnest:		mode: camera-streamer
[06/24/23 16:15:56] crowsnest:		
[06/24/23 16:15:56] crowsnest:		enable_rtsp: false
[06/24/23 16:15:56] crowsnest:		rtsp_port: 8554
[06/24/23 16:15:56] crowsnest:		port: 8080
[06/24/23 16:15:56] crowsnest:		device: /base/soc/i2c0mux/i2c@1/imx477@1a
[06/24/23 16:15:56] crowsnest:		resolution: 4056x3040
[06/24/23 16:15:56] crowsnest:		max_fps: 10
[06/24/23 16:15:56] crowsnest:		custom_flags: --camera-snapshot.height=3040
[06/24/23 16:15:56] crowsnest: INFO: Detect available Devices
[06/24/23 16:15:56] crowsnest: INFO: Found 1 total available Device(s)
[06/24/23 16:15:56] crowsnest: Detected 'libcamera' device -> /base/soc/i2c0mux/i2c@1/imx477@1a
[06/24/23 16:15:56] crowsnest: Try to start configured Cams / Services...
[06/24/23 16:15:57] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue ...
[... 31 lines cut ...]
[06/24/23 16:15:59] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/device_options.c: VIDEO: Configuring option h264iframeperiod (00990a66) = 30
[06/24/23 16:15:59] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/device_options.c: VIDEO: Configuring option h264level (00990a67) = 11
[06/24/23 16:15:59] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/device_options.c: VIDEO: Configuring option h264profile (00990a6b) = 4
[06/24/23 16:15:59] crowsnest: DEBUG: camera-streamer [cam 1]: device/v4l2/device_options.c: VIDEO: Configuring option h264minimumqpvalue (00990a61) = 16
[... cut ...]
[06/24/23 16:16:00] crowsnest: DEBUG: camera-streamer [cam 1]: [1:28:04.693663947] [6151] [1;31mERROR [1;37mV4L2 [1;34mv4l2_videodevice.cpp:1241 [0;32m/dev/video16[28:cap]: [0mUnable to request 0 buffers: Device or resource busy
[06/24/23 16:16:00] crowsnest: DEBUG: camera-streamer [cam 1]: terminate called without an active exception
[06/24/23 16:16:00] crowsnest: DEBUG: camera-streamer [cam 1]: xargs: /home/pi/crowsnest/bin/camera-streamer/camera-streamer: terminated by signal 6
[06/24/23 16:16:00] crowsnest: ERROR: Start of camera-streamer [cam 1] failed!
```
Any help is really appreciated. I need those sexy timelapses! Thanks!

commit 95c1dca1 (2023-03-11): feat!: adds camera-streamer to crowsnest

This introduces camera-streamer as streamer option
via `mode: multi`

camera-streamer is a feature packed stream service.
It is capable to deliver mjpg/snapshots/webrtc and rtsp

Limited to raspberry pi sbc's for now.

This should also resolv
Feature request #51
Feature request #37
Fixes #83
Closes #85
Fixes #89

BREAKING CHANGES:

Dropping support for Debian Buster based images and kernels older than
5.15y

Dropping RTSP support due aler9/simple-rtsp-server

Dropping usage of ffmpeg

No support anymore for Raspicam V1 (EOL)