mainsail-crew/crowsnest #86: Streaming broken on current version

jonp92, 2023-02-08:
### What happened

When trying to start Crowsnest the following error occurs.

crowsnest[27139]: [02/08/23 13:43:12] crowsnest: ERROR: Error 1 occured on line 147
mainsailos crowsnest[27141]: ERROR: Error 1 occured on line 147

### What did you expect to happen

Crowsnest should launch successfully.

### How to reproduce

Configure Crowsnest with USB Camera.
Try to Launch.

### Additional information

This can be fixed by commenting out the call for blockyfix in the crowsnest executable.

,#### crowsnest.conf
#### This is a typical default config.
#### Also used as default in mainsail / MainsailOS
#### See:
#### https://github.com/mainsail-crew/crowsnest/blob/master/README.md
#### for details to configure to your needs.

#####################################################################
####                                                            #####
####      Information about ports and according URL's           #####
####                                                            #####
#####################################################################
####                                                            #####
####    Port 8080 equals /webcam/?action=[stream/snapshot]      #####
####    Port 8081 equals /webcam2/?action=[stream/snapshot]     #####
####    Port 8082 equals /webcam3/?action=[stream/snapshot]     #####
####    Port 8083 equals /webcam4/?action=[stream/snapshot]     #####
####                                                            #####
#####################################################################

[crowsnest]
log_path: ~/printer_data/logs/crowsnest.log
log_level: verbose                      # Valid Options are quiet/verbose/debug
delete_log: false                       # Deletes log on every restart, if set to true

[cam 1]
mode: mjpg                              # mjpg/rtsp
port: 8080                              # Port
device: /dev/v4l/by-id/usb-046d_Logitech_Webcam_C930e_E44F92BE-video-index0                     # See Log for available ...
resolution: 800x600                   # widthxheight format
max_fps: 15                             # If Hardware Supports this it will be forced, ohterwise ignored/coerced.
#custom_flags:                          # You can run the Stream Services with custom flags.
v4l2ctl: focus_auto=0,focus_absolute=35,backlight_compensation=1                              # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.

KwadFan, 2023-02-08:
Without a full log I cant help, sorry

jonp92, 2023-02-09:
[crowsnest.log](https://github.com/mainsail-crew/crowsnest/files/10698368/crowsnest.log)

KwadFan, 2023-02-16:
After reviewing your log twice, there are some hickups, I guess they are caused by a bad config file.

Before the stream starts I let print the conf file into the log ( for obvious reasons ).

```[02/08/23 01:33:57] crowsnest: 		,
[02/08/23 01:33:57] crowsnest: 		
[02/08/23 01:33:57] crowsnest: 		
[02/08/23 01:33:57] crowsnest: 		
[02/08/23 01:33:57] crowsnest: 		
[02/08/23 01:33:57] crowsnest: 		[crowsnest]
[02/08/23 01:33:57] crowsnest: 		log_path: ~/printer_data/logs/crowsnest.log
[02/08/23 01:33:57] crowsnest: 		log_level: verbose
[02/08/23 01:33:57] crowsnest: 		delete_log: false
[02/08/23 01:33:57] crowsnest: 		
[02/08/23 01:33:57] crowsnest: 		[cam 1]
[02/08/23 01:33:57] crowsnest: 		mode: mjpg
[02/08/23 01:33:57] crowsnest: 		port: 8080
[02/08/23 01:33:57] crowsnest: 		device: /dev/v4l/by-id/usb-046d_Logitech_Webcam_C930e_E44F92BE-video-index0
[02/08/23 01:33:57] crowsnest: 		resolution: 800x600
[02/08/23 01:33:57] crowsnest: 		max_fps: 15
[02/08/23 01:33:57] crowsnest: 		v4l2ctl: focus_auto=0,focus_absolute=35,backlight_compensation=1```

This shows that the config file begins with a comma `,` right at the start. Also it shows later on
```[02/08/23 01:34:06] crowsnest: No parameters set for [cam DEFAULT]. Skipped.
[02/08/23 01:34:11] crowsnest: Device: [cam 1]```

Which are not present in your config.
So, please use our discord for further help, this is not an code Issue, thats purely a configuration issue.

Thanks in advance, regards Kwad

closed 2023-02-16 by KwadFan