mainsail-crew/crowsnest #88: flip-vertical and flip-horizontal flags seem to have no effect

kieraneglin, 2023-02-18:
### What happened

My camera has to be mounted 180º to fit on my printer. The `--flip-vertical=1 --flip-horizontal=1` flags in my `crowsnest.conf`'s `custom_flags` section has no effect on the vertical/horizontal flipping of the camera signal

### What did you expect to happen

A combination of the `--flip-vertical=1` and `--flip-horizontal=1` should effectively rotate my camera's signal by 180º

### How to reproduce

See above

### Additional information

**Camera**: Logitech C270 (which doesn't seem to support rotation through `V4L2`)
**SBC**: BTT CB1 mounted in their Manta M4P mainboard
**OS**: BTT-CB1 2.3.1 Bullseye
**Crowsnest version**: v3.0.7-1-g43b86a3
**Logs**: [crowsnest.log](https://github.com/mainsail-crew/crowsnest/files/10774853/crowsnest.log)
**crowsnest.conf**:
```
[crowsnest]
log_path: ~/printer_data/logs/crowsnest.log
log_level: debug
delete_log: false

[cam 1]
mode: mjpg
port: 8080
device: /dev/video0
resolution: 1280x720 
max_fps: 20
# Increasing device timeout as troubleshooting step per https://github.com/mainsail-crew/crowsnest/issues/39#issuecomment-1271729882
custom_flags: --device-timeout=8 --flip-vertical=1 --flip-horizontal=1
#v4l2ctl:             
```

meteyou, 2023-02-18:
```
[02/18/23 17:28:10] crowsnest: DEBUG: ustreamer [cam 1]: -- ERROR [1675.039 stream] -- Changing control flip_vertical is unsupported
[02/18/23 17:28:10] crowsnest: DEBUG: ustreamer [cam 1]: -- ERROR [1675.039 stream] -- Changing control flip_horizontal is unsupported
```

where do you have these commands? these are no ustreamer commands.

closed 2023-02-18 by meteyou

kieraneglin, 2023-02-18:
@meteyou I found these commands both on the [reference in this repo](https://github.com/mainsail-crew/crowsnest/blob/master/ustreamer_manpage.md) and the [official man page](https://manpages.ubuntu.com/manpages/jammy/man1/ustreamer.1.html)

DarkBuf0n, 2024-10-15:
> @meteyou I found these commands both on the [reference in this repo](https://github.com/mainsail-crew/crowsnest/blob/master/ustreamer_manpage.md) and the [official man page](https://manpages.ubuntu.com/manpages/jammy/man1/ustreamer.1.html)

same problem. did you finally get it?
Thanks

mryel00, 2024-10-15:
@DarkBuf0n The manpage is for version 4.9 and we use version 6.10. This issue is over one year old and has nothing to do with Crowsnest. That error is from ustreamer and you have to check if your camera supports rotation. If you need help join our [Discord](https://discord.gg/mainsail) or use our [Discussions](https://github.com/orgs/mainsail-crew/discussions)

DarkBuf0n, 2024-10-16:
> @DarkBuf0n The manpage is for version 4.9 and we use version 6.10. This issue is over one year old and has nothing to do with Crowsnest. That error is from ustreamer and you have to check if your [...]

Thank you. My camera is a Logitech c922 how can I check if it is compatible?

mryel00, 2024-10-16:
Like I wrote, if you need help you should use Discord or the Discussions!
All available controls are in the log. For everything further go to our Discord or the Discussions.