mainsail-crew/crowsnest #89: Getting "Error 1 occured on line 50" after update to v3.0.7-2-ga5a36e5

vash2695, 2023-02-20:
### What happened

Ran an update via KIAUH from v3.0.6-25-g5418550 to v3.0.7-2-ga5a36e5 and now I'm getting this error and crowsnest fails to start. No previous errors.

### What did you expect to happen

Normal startup

### How to reproduce

na

### Additional information

[02/20/23 17:53:59] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[02/20/23 17:53:59] crowsnest: Version: v3.0.7-2-ga5a36e5
[02/20/23 17:53:59] crowsnest: Prepare Startup ...
[02/20/23 17:53:59] crowsnest: INFO: Host information:
[02/20/23 17:53:59] crowsnest: Host Info: Distribution: Ubuntu 22.04.2 LTS
[02/20/23 17:53:59] crowsnest: Host Info: Kernel: Linux 5.19.0-32-generic x86_64
[02/20/23 17:53:59] crowsnest: Host Info: Model: Intel(R) Core(TM) i5-7500T CPU @ 2.70GHz
[02/20/23 17:53:59] crowsnest: Host Info: Available CPU Cores: 4
[02/20/23 17:53:59] crowsnest: Host Info: Available Memory: 16266492 kB
[02/20/23 17:53:59] crowsnest: Host Info: Diskspace (avail. / total): 207G / 234G
[02/20/23 17:53:59] crowsnest: INFO: Checking Dependencys
[02/20/23 17:53:59] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[02/20/23 17:53:59] crowsnest: Dependency: 'find' found in /usr/bin/find.
[02/20/23 17:53:59] crowsnest: Dependency: 'logger' found in /usr/bin/logger.
[02/20/23 17:53:59] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[02/20/23 17:53:59] crowsnest: Dependency: 'ffmpeg' found in /usr/bin/ffmpeg.
[02/20/23 17:54:00] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[02/20/23 17:54:00] crowsnest: Dependency: 'rtsp-simple-server' found in bin/rtsp-simple-server/rtsp-simple-server.
[02/20/23 17:54:00] crowsnest: Version Control: ustreamer new version available: v5.37-1-ga10df2f (v5.37).
[02/20/23 17:54:00] crowsnest: Version Control: rtsp-simple-server is up to date. (v0.20.2)
[02/20/23 17:54:00] crowsnest: Version Control: ffmpeg is up to date. (4.4.2-0ubuntu0.22.04.1)
[02/20/23 17:54:00] crowsnest: INFO: Print Configfile: '/home/bobby/printer_data/config/crowsnest.conf'
[02/20/23 17:54:00] crowsnest: 		[crowsnest]
[02/20/23 17:54:00] crowsnest: 		log_path: ~/printer_data/logs/crowsnest.log
[02/20/23 17:54:00] crowsnest: 		log_level: verbose
[02/20/23 17:54:00] crowsnest: 		delete_log: true
[02/20/23 17:54:00] crowsnest: 		
[02/20/23 17:54:00] crowsnest: 		[cam 1]
[02/20/23 17:54:00] crowsnest: 		mode: mjpg
[02/20/23 17:54:00] crowsnest: 		port: 8080
[02/20/23 17:54:00] crowsnest: 		device: /dev/video0
[02/20/23 17:54:00] crowsnest: 		resolution: 1920x1080
[02/20/23 17:54:00] crowsnest: 		max_fps: 60
[02/20/23 17:54:00] crowsnest: 		custom_flags:
[02/20/23 17:54:00] crowsnest: 		v4l2ctl: white_balance_automatic=0,white_balance_temperature=5500,focus_auto=0,hue=-10,exposure_auto=1,exposure_auto_priority=1,exposure_absolute=100
[02/20/23 17:54:00] crowsnest: 		
[02/20/23 17:54:00] crowsnest: INFO: Detect available Devices
[02/20/23 17:54:00] crowsnest: INFO: Found 1 total available Device(s)
[02/20/23 17:54:00] crowsnest: INFO: Found 1 available camera(s)
[02/20/23 17:54:00] crowsnest: /dev/v4l/by-id/usb-Sonix_Technology_Co.__Ltd._NexiGo_HD_Webcam_SN0001-video-index0 -> /dev/video0
[02/20/23 17:54:00] crowsnest: Supported Formats:
[02/20/23 17:54:00] crowsnest: 		[0]: 'MJPG' (Motion-JPEG, compressed)
[02/20/23 17:54:00] crowsnest: 		Size: Discrete 640x480
[02/20/23 17:54:00] crowsnest: 		Interval: Discrete 0.017s (60.000 fps)
[02/20/23 17:54:00] crowsnest: 		Interval: Discrete 0.033s (30.000 fps)
[02/20/23 17:54:00] crowsnest: 		Size: Discrete 1600x896
[02/20/23 17:54:01] crowsnest: 		Interval: Discrete 0.017s (60.000 fps)
[02/20/23 17:54:01] crowsnest: 		Interval: Discrete 0.033s (30.000 fps)
[02/20/23 17:54:01] crowsnest: 		Size: Discrete 1280x720
[02/20/23 17:54:01] crowsnest: 		Interval: Discrete 0.017s (60.000 fps)
[02/20/23 17:54:01] crowsnest: 		Interval: Discrete 0.033s (30.000 fps)
[02/20/23 17:54:01] crowsnest: 		Size: Discrete 1024x576
[02/20/23 17:54:01] crowsnest: 		Interval: Discrete 0.017s (60.000 fps)
[02/20/23 17:54:01] crowsnest: 		Interval: Discrete 0.033s (30.000 fps)
[02/20/23 17:54:01] crowsnest: 		Size: Discrete 960x720
[02/20/23 17:54:01] crowsnest: 		Interval: Discrete 0.017s (60.000 fps)
[... 46 lines cut ...]
[02/20/23 17:54:01] crowsnest: 		gamma 0x00980910 (int) : min=72 max=500 step=1 default=95 value=95
[02/20/23 17:54:01] crowsnest: 		gain 0x00980913 (int) : min=0 max=100 step=1 default=0 value=0
[02/20/23 17:54:01] crowsnest: 		power_line_frequency 0x00980918 (menu) : min=0 max=2 default=2 value=2 (60 Hz)
[02/20/23 17:54:01] crowsnest: 		0: Disabled
[02/20/23 17:54:02] crowsnest: 		1: 50 Hz
[02/20/23 17:54:02] crowsnest: 		2: 60 Hz
[02/20/23 17:54:02] crowsnest: 		white_balance_temperature 0x0098091a (int) : min=2800 max=6500 step=1 default=4600 value=5500
[02/20/23 17:54:02] crowsnest: 		sharpness 0x0098091b (int) : min=0 max=6 step=1 default=2 value=2
[02/20/23 17:54:02] crowsnest: 		backlight_compensation 0x0098091c (int) : min=0 max=2 step=1 default=1 value=0
[02/20/23 17:54:02] crowsnest: 		
[02/20/23 17:54:02] crowsnest: 		Camera Controls
[02/20/23 17:54:02] crowsnest: 		
[02/20/23 17:54:02] crowsnest: 		auto_exposure 0x009a0901 (menu) : min=0 max=3 default=3 value=1 (Manual Mode)
[02/20/23 17:54:02] crowsnest: 		1: Manual Mode
[02/20/23 17:54:02] crowsnest: 		3: Aperture Priority Mode
[02/20/23 17:54:02] crowsnest: 		exposure_time_absolute 0x009a0902 (int) : min=1 max=5000 step=1 default=157 value=100
[02/20/23 17:54:02] crowsnest: 		exposure_dynamic_framerate 0x009a0903 (bool) : default=0 value=1
[02/20/23 17:54:02] crowsnest: 		focus_automatic_continuous 0x009a090c (bool) : default=1 value=0
[02/20/23 17:54:02] crowsnest: 		privacy 0x009a0910 (bool) : default=0 value=0
[02/20/23 17:54:02] crowsnest: INFO: No usable CSI Devices found.
[02/20/23 17:54:02] crowsnest: V4L2 Control: Device: [cam 1]
[02/20/23 17:54:02] crowsnest: V4L2 Control: Options: white_balance_automatic=0,white_balance_temperature=5500,focus_auto=0,hue=-10,exposure_auto=1,exposure_auto_priority=1,exposure_absolute=100
[02/20/23 17:54:02] crowsnest: ERROR: Error 1 occured on line 50
[02/20/23 17:54:02] crowsnest: ERROR: Stopping crowsnest.
[02/20/23 17:54:02] crowsnest: Goodbye...

KwadFan, 2023-02-24:
Well, I really try to figure out where this might come from.

I didnt change the part where this happens.
Could you please share the real log file, git rips the formatting appart, I have a clue what it might be, but need the original log to verify.
You can simply drag and drop file to the comment box.

Your error happends here: https://github.com/mainsail-crew/crowsnest/blob/a5a36e5c3d153f0b568ea731b7e3b02badb139f4/libs/v4l2_control.sh#L50

This commands tries to set a value and errors out if not successful.

Regards Kwad

KwadFan, 2023-02-24:
I added a error message if command fails. Now you should see `"Failed to set parameter: '${param}' ..."` in your log.
`${param}` is replaced by the parameter it tries to set.

If you see a message like this let me know and I'll try to fix it.

Regards Kwad

DasBurninator, 2023-02-24:
Been having the same issue for a few days as well on a machine. If I comment out the vl2ctl line in my config it works fine. With the new you implemented, it just ignores that part of the config entirely. Updated a few minutes ago to v3.0.7-3-g20ed6a8.

Have a machine on v3.0.6-23-g00fe721 and this is working fine. Same camera, same configs, just different version of crowsnest.

KwadFan, 2023-02-25:
> 4l2ctl: focus_auto=0,focus_absolute=42

This will not work ;) the section is named ```v4l2ctl:```

vash2695, 2023-02-25:
Can confirm that after the most recent update we're back to normal! Now if I could just get my stream above 15fps in mainsail it'd be perfect.

commit 95c1dca1 (2023-03-11): feat!: adds camera-streamer to crowsnest

This introduces camera-streamer as streamer option
via `mode: multi`

camera-streamer is a feature packed stream service.
It is capable to deliver mjpg/snapshots/webrtc and rtsp

Limited to raspberry pi sbc's for now.

This should also resolv
Feature request #51
Feature request #37
Fixes #83
Closes #85
Fixes #89

BREAKING CHANGES:

Dropping support for Debian Buster based images and kernels older than
5.15y

Dropping RTSP support due aler9/simple-rtsp-server

Dropping usage of ffmpeg

No support anymore for Raspicam V1 (EOL)
closed 2023-05-22 by KwadFan