mainsail-crew/crowsnest #90: How to make Crowsnest to switch off/on when you power off/on your 3d printer

Huyaki, 2023-02-22:
### Is your feature request related to a problem? Please describe

I wanted to make my 3d printer's webcam switching off when I power off my printer so webcam is working only when 3d printer is powered. I power off my printer when I finish printing however my mini-pc with Klipper on it keeps working 24/7 as I have some other service running on it.

### Describe the solution you'd like

I've spent a day solving this issue and I succeeded. So now I want to share my result to other people who want make their 3d printing setup in the same way.

So the solution is following:

1) You need to know your Klipper mcu device id in systemd unit format. In order to do it make sure your printer is powered on and execute:
`systemctl list-units --type=device`

you will need to find it in the list and save it. in my case it was:
`dev-serial-by\x2did-usb\x2dKlipper_stm32g0b1xx_280049000550414235393120\x2dif00.device`

2) You need to modify Crowsnest systemd service unit file which should be located in:
`sudo nano /etc/systemd/system/crowsnest.service`

In the [Unit] section you need to have following fields to be like:
```
After=udev.service network-online.target nss-lookup.target <YOUR-MCU-ID>
BindsTo=<YOUR-MCU-ID>
Wants=udev.service network-online.target <YOUR-MCU-ID>
```

As well as you need to add addition field (in the end):
`StopWhenUnneeded=true`

In my case [Unit] section looks like:
```
[Unit]
Description=crowsnest - Multi Webcam/Streamer Control Deamon
Documentation=https://github.com/mainsail-crew/crowsnest
After=udev.service network-online.target nss-lookup.target dev-serial-by\x2did-usb\x2dKlipper_stm32g0b1xx_280049000550414235393120\x2dif00.device
BindsTo=dev-serial-by\x2did-usb\x2dKlipper_stm32g0b1xx_280049000550414235393120\x2dif00.device
Wants=udev.service network-online.target dev-serial-by\x2did-usb\x2dKlipper_stm32g0b1xx_280049000550414235393120\x2dif00.device
Before=nginx.service
StartLimitBurst=10
StartLimitIntervalSec=180
StopWhenUnneeded=true
```

[Install] section remains like this:
```
[Install]
WantedBy=multi-user.target
```

In the [Service] section I am not sure about "RemainAfterExit" field, but I've commented it out so its value goes to default (No):
```
[Service]
Type=simple
User=jack
#RemainAfterExit=No
WorkingDirectory=/home/jack/crowsnest
EnvironmentFile=/home/jack/printer_data/systemd/crowsnest.env
ExecStart= /usr/local/bin/crowsnest $CROWSNEST_ARGS
Restart=on-failure
RestartSec=5 
```

After making these  changed you'll probably need to reload daemon and reboot:
```
sudo systemctl daemon-reload
sudo reboot
```

And that's it. Working great on Debian 11.

### Describe alternatives you've considered

I've tried to make it using udev rules. I made it working when switching on but failed on switching off. As well as people in the internet said that it's not really good to run long-term processes using udev rules and suggested to modify systemd service.

KwadFan, 2023-02-24:
Hey there :wave: 

Thank you for your effort, but this section, as mentioned in the foreword, to keep track of issues related to my code.
This is not a wiki! So, please dont thread it like one. This makes it harder to keep track of real issues. Also this is a very specific setup, tailored to your needs and I dont want to provide solutions of that kind. At least mainsail gives you an option to stop the service if you dont want to run crowsnest.

I will close this issue but feel free to share your solution, where ever you want.

Regards Kwad

closed 2023-02-24 by github-actions[bot]

MohammedSamy, 2024-01-02:
There is another way that works. Especially if you rely on the printer MCU connected directly to raspbery pi pins and not the USB, the above method wont work for you. As the klipper device wont see a difference between printer of and on in the systemd.

In the `moonraker.conf,` you should have a config for contronlling the printer power (how to enable the mcu printer or disable it). If you dont have it , follow the moonraker config to create it according to your situation. I am controlling my mcu printer with a gpio on my raspberry pi connected to a relay. Here is mine:

```
[power printer]
type: gpio
pin: !gpiochip0/gpio26
off_when_shutdown: True
restart_klipper_when_powered: true
on_when_job_queued: True
initial_state: on
```

Now just add the following line to the config: 

`bound_services: crowsnest`

crowsnest servive (which controls the webcam) will be disabled wen printer is off.

D3ltorohd, 2024-03-30:
I will shut down my cam to. On my printer there is a plug i can control ofer my smarthome. So i shutdown the printer when he is done. I have this state , so how can i tell crowsnets tu shutdown the cam, like i can do in webinterface ? I need some commands for the command line, that i can send,

mryel00, 2024-03-30:
> so how can i tell crowsnets tu shutdown the cam

You can't! Crowsnest cannot do something like that. There isn't even any reason for Crowsnest to be able to shutdown your camera, as it doesn't know anything about your printer. If you want to shutdown your camera, shutdown Crowsnest.

Like Kwad already told in the beginning, this is no forum or wiki to discuss such things. Furthermore this isn't any kind of support forum. If you want to discuss such things, please use our [Github discussions](https://github.com/orgs/mainsail-crew/discussions). Just open a post there.

P.S. `sudo systemctl stop crowsnest` will stop the service.