mainsail-crew/crowsnest #94: Ability to turn off webcam

jorins, 2023-03-05:
### Is your feature request related to a problem? Please describe

The webcam I'm using with Crowsnest in conjunction with Mainsail is streaming at all times and I've found no simple way of controlling its power in software. This raises two concerns for me:

1. The camera's lifespan may be affected negatively.
2. More footage than necessary is being streamed. In the unfortunate event that my camera feed reaches the public web, I'd much prefer to minimise the amount of footage that could get out.

### Describe the solution you'd like

The ability to control the state of the camera feed and programmatically turning it on and off in conjunction with prints, in the same vein that video-conferencing software will only activate your camera when you're in a call.

### Describe alternatives you've considered

1. The camera could be forcefully powered off using a relay controlled over GPIO and [Moonraker's `[power]` configuration](https://moonraker.readthedocs.io/en/latest/configuration/#power). This would require extra hardware and Crowsnest would not have a meaningful understanding of why the camera is unavailable.
2. Unwanted leakage of footage can be mitigated through choosing a different camera angle and physically blocking the camera's view.

### Additional information

Thank you for your wonderful work :)

KwadFan, 2023-03-08:
Hi there :wave:,
thanks for your request. I dont want to deny this at all, but there are some technical restriction that has to be taken to account.

Switching of USB on a pi requires either some quirky udev rules and/or some sudo rights.
Besides that its quite complicated to "hit" the right USB and in some models only all USB are powerd off then, which leads to complications with klipper because its will loose connection to the MCU.
Those are effects I dont want to deal with to be honest.

A better approach would be "to teach" moonraker to stop crowsnest due a special gcode like `TIMELAPSE_TAKE_FRAME` for example. But my experience is limited to bash, or better said my py foo are good like a months old child .. I eat sand and memorized it doesnt taste good :rofl:

So maybe one day I can deliver that, for now I have to deny that request even I like the Idea.

Sorry & Regards Kwad

closed 2023-03-08 by KwadFan

jorins, 2023-03-09:
I figured the simplest approach would be to just stop consuming the video feed. At least, that's what I'm assuming is happening with most video chat-capable apps. I might see if I can hack something together on my own nevertheless, thank you! :)

KwadFan, 2023-03-11:
> I figured the simplest approach would be to just stop consuming the video feed. At least, that's what I'm assuming is happening with most video chat-capable apps. I might see if I can hack something together on my own nevertheless, thank you! :)

Yes, stopping is the easiest apptoach and can be done via moonrakers service management manually.In mainsail simply stop it via "Power Button" dropdown menu. I really like the automated stop/start idea. But I have to implement a component to moonraker to start/stop via gcode command, and thats out of my skills currently. I keep that idea in mind, because I want to learn py anyways so, maybe one day I am able to deliver.

Regards Kwad