mainsail-crew/crowsnest #96: error with v4l2ctl focus settings (Logitech C920)

dmnkhhn, 2023-03-09:
### What happened

If I use the examples from the manual to control my Logitech C920 Camera I get this:
```
pi@voron2:~ $ v4l2-ctl -d /dev/video0 -c focus_auto=0
unknown control 'focus_auto'
pi@voron2:~ $ v4l2-ctl -d /dev/video0 -c focus_absolute=30
VIDIOC_S_EXT_CTRLS: failed: Input/output error
focus_absolute: Input/output error
``` 
With this line in my config I get this output in the log:
```
[03/09/23 12:00:48] crowsnest: V4L2 Control: Device: [cam 1]
[03/09/23 12:00:48] crowsnest: V4L2 Control: Options: focus_auto=0,focus_absolute=30
[03/09/23 12:00:48] crowsnest: V4L2 Control: Failed to set parameter: 'focus_auto=0' ...
[03/09/23 12:00:48] crowsnest: V4L2 Control: Failed to set parameter: 'focus_absolute=30' ...
[03/09/23 12:00:49] crowsnest: Try to start configured Cams / Services...
[03/09/23 12:00:50] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue...
[03/09/23 12:00:52] crowsnest: ... Done!
[03/09/23 12:00:52] crowsnest: Starting ustreamer with Device /dev/video0 ...
[03/09/23 12:00:53] crowsnest: WARN: Detected 'brokenfocus' device.
[03/09/23 12:00:53] crowsnest: INFO: Trying to set to configured Value.
[03/09/23 12:00:53] crowsnest: ERROR: Error 255 occured on line 114
[03/09/23 12:00:53] crowsnest: ERROR: Stopping crowsnest.
[03/09/23 12:00:53] crowsnest: Goodbye...
``` 

Maybe a firmware update changed those parameters?! The (newly?) available controls are:
```
pi@voron2:~ $ v4l2-ctl -d /dev/video0 --list-ctrls

User Controls

                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=128
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=128 value=128
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=128 value=128
        white_balance_automatic 0x0098090c (bool)   : default=1 value=1
                           gain 0x00980913 (int)    : min=0 max=255 step=1 default=0 value=0
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2
      white_balance_temperature 0x0098091a (int)    : min=2800 max=7500 step=1 default=5000 value=4472 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=255 step=1 default=128 value=128
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=0 value=0

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
         exposure_time_absolute 0x009a0902 (int)    : min=3 max=2047 step=1 default=156 value=156 flags=inactive
     exposure_dynamic_framerate 0x009a0903 (bool)   : default=0 value=1
                   pan_absolute 0x009a0908 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
                  tilt_absolute 0x009a0909 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
                 focus_absolute 0x009a090a (int)    : min=0 max=250 step=1 default=0 value=59
     focus_automatic_continuous 0x009a090c (bool)   : default=1 value=0
                  zoom_absolute 0x009a090d (int)    : min=100 max=500 step=1 default=100 value=100
``` 

If I update my configuration and set `v4l2ctl: focus_automatic_continuous=0,focus_absolute=30`, which works when setting it manually from the command line, the following appears in my log:
```
[03/09/23 11:50:50] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue...
[03/09/23 11:50:52] crowsnest: ... Done!
[03/09/23 11:50:52] crowsnest: Starting ustreamer with Device /dev/video0 ...
[03/09/23 11:50:53] crowsnest: WARN: Detected 'brokenfocus' device.
[03/09/23 11:50:53] crowsnest: INFO: Trying to set to configured Value.
[03/09/23 11:50:54] crowsnest: ERROR: Error 255 occured on line 114
[03/09/23 11:50:54] crowsnest: ERROR: Stopping crowsnest.
[03/09/23 11:50:54] crowsnest: Goodbye...
```
Now crowsnest will start again, but it gets the same error and it goes on repeat and after a short while the Pi is almost overheating.

TL;DR
As soon as I set the focus parameter (new or old) the `brokenfocus` device warning comes up and crowsnest continuously restarts.

### What did you expect to happen

I would expect the `v4l2ctl` setting to work properly with the updated parameters for my Logitech C920 

### How to reproduce

- connect the camera via USB
- get the available parameters `v4l2-ctl -d /dev/video0 --list-ctrls`
- set the `focus_automatic_continuous` parameter to `0`
- set the `focus_absolute` parameter to something between 0 and 250
- check the log

Additionally you could set those parameters from the command line to see if they work at all:
`v4l2-ctl -d /dev/video0 -c focus_automatic_continuous=0`
`v4l2-ctl -d /dev/video0 -c focus_absolute=125` 

### Additional information

```
crowsnest: Version: v3.0.7-3-g20ed6a8
crowsnest: Host Info: Distribution: Raspbian GNU/Linux 11 (bullseye)
crowsnest: Host Info: Kernel: Linux 6.1.14-v7+ armv7l
crowsnest: Host Info: Model: Raspberry Pi 3 Model B Plus Rev 1.3
crowsnest: Version Control: ustreamer is up to date. (v4.13)
crowsnest: Version Control: rtsp-simple-server is up to date. (v0.20.2)
crowsnest: Version Control: ffmpeg is up to date. (4.3.5-0+deb11u1+rpt3)
```

KwadFan, 2023-03-11:
Hey there,

First of all, as I do not own this camera it is hard to reproduce for me, which makes it impossible to fix.

But...

```
[03/09/23 12:00:48] crowsnest: V4L2 Control: Failed to set parameter: 'focus_auto=0' ...
[03/09/23 12:00:48] crowsnest: V4L2 Control: Failed to set parameter: 'focus_absolute=30' ...
``` 
these to lines already tell you what is going on here.

These Logitech Cams (C920, C922, C930e) are known to have issues in firmware and there where several tries fron linux kernel devs to workaround those issues.

The camera simply refuses to set those parameters at runtime. Currently those parameters are set before ustreamer starts, and the "brokenfocus" fix already tries to re-set those values after ustreamer is started. But if the camera wont let you set these values, for what ever reason, it fails which is intended.

So its hard to say but I cant help you here, simply because I need the exact same model and firmware of your cam to reproduce and develop a workaround.

Maybe @zellneralex has an idea here.
Regards Kwad

zellneralex, 2023-03-11:
No I have no other idea. We would need to have that cam to do further investigations.

@KwadFan is v4l2ctl: focus_automatic_continuous=0,focus_absolute=30 maybe the problem here.
Would we see more if he uses the debug option?

KwadFan, 2023-03-11:
So,
the only thing we can do is a bit a guessing game here.

@dmnkhhn 
Please set `log_level: debug` and `delete_log: true`. Restart crowsnest and upload the complete new log.

time-tm, 2023-03-14:
I have the same C920. Had the exact same issue. Changing focus_auto = 0 to focus_automatic_continuous = 0 solved it for me.

TheHipsterGandalf, 2023-03-18:
> I have the same C920. Had the exact same issue. Changing focus_auto = 0 to focus_automatic_continuous = 0 solved it for me.

Thank you for this, it has solved my problem!

Imburr, 2023-03-21:
This also fixed it for me, thank you. For the next person here, there is no spacing in the config file, i.e.

v4l2ctl: focus_automatic_continuous=0,focus_absolute=20

zao, 2023-04-14:
My experience with the C920 camera is that the firmware in the camera refuses to set the exposure and focus values if the corresponding feature is configured to be automatic. This includes the case where one command line tries to change those features from auto to manual mode _and_ adjust the value at the same time.

Results can vary over time in a session if you've managed to change the mode beforehand after which the value setting is likely to take.

Before moving to Klipper+crowsnest I had to run two distinct commands in `rc.local`, one to first configure the mode toggles followed by one to actually adjust the manual values:
```bash
v4l2-ctl --set-ctrl=auto_exposure=1,focus_automatic_continuous=0,white_balance_automatic=0,power_line_frequency=1
v4l2-ctl --set-ctrl=exposure_time_absolute=200,focus_absolute=40,white_balance_temperature=4395
```

I can't find any way in `crowsnest.conf` to do this kind of "two-phase" control setting. I could probably work around it by writing a janky `v4l2-ctl` wrapper that splits a command line into two based on the types of arguments but it'd be nice to have some sort of upstream support for this kind of horror.

One interesting side note is that the `focus_auto` parameter mentioned earlier in the issue is called `focus_automatic_continuous` on my model.
```
  idVendor           0x046d Logitech, Inc.
  idProduct          0x082d HD Pro Webcam C920
  bcdDevice            0.11
```

behold81, 2023-04-14:
> focus_automatic_continuous=0

V4L2 Control: Parameter 'focus_automatic_continuous=0' not available for '/dev/video0'. Skipped.

Not an option for me

behold81, 2023-04-14:
> My experience with the C920 camera is that the firmware in the camera refuses to set the exposure and focus values if the corresponding feature is configured to be automatic. This includes the case where one command line tries to change those features from auto to manual mode _and_ adjust the value at the same time.
> 
> Results can vary over time in a session if you've managed to change the mode beforehand after which the value setting is likely to take.
> 
> Before moving to Klipper+crowsnest I had to run two distinct commands in `rc.local`, one to first configure the mode toggles followed by one to actually adjust the manual values:
> 
> ```shell
> v4l2-ctl --set-ctrl=auto_exposure=1,focus_automatic_continuous=0,white_balance_automatic=0,power_line_frequency=1
> v4l2-ctl --set-ctrl=exposure_time_absolute=200,focus_absolute=40,white_balance_temperature=4395
> ```
> 
> I can't find any way in `crowsnest.conf` to do this kind of "two-phase" control setting. I could probably work around it by writing a janky `v4l2-ctl` wrapper that splits a command line into two based on the types of arguments but it'd be nice to have some sort of upstream support for this kind of horror.
> 
> One interesting side note is that the `focus_auto` parameter mentioned earlier in the issue is called `focus_automatic_continuous` on my model.
> 
> ```
>   idVendor           0x046d Logitech, Inc.
>   idProduct          0x082d HD Pro Webcam C920
>   bcdDevice            0.11
> ```

it seems to fail to apply when done to quickly. it works perfectly when edited in the shell. and never falters till I restart the Pi

KwadFan, 2023-04-15:
> > focus_automatic_continuous=0
> 
> V4L2 Control: Parameter 'focus_automatic_continuous=0' not available for '/dev/video0'. Skipped.
> 
> Not an option for me

There is now "option" if v4l2-ctl does not list that parameter for your device `/dev/video0` you simply cant use it, and thats what this line tells you. Not more not less...

KwadFan, 2023-04-15:
> My experience with the C920 camera is that the firmware in the camera refuses to set the exposure and focus values if the corresponding feature is configured to be automatic. This includes the case [...]

As you can see here
https://github.com/mainsail-crew/crowsnest/blob/20ed6a8b585a92e8a0e7d8333e81b6e8ca7044e1/libs/v4l2_control.sh#L41

It sets the values in a for loop, so each paramater after another. You could try to add a sleep state but it will delay its startup anyways. So, there is not much I can do if the Firmware is simply to slow to except settings in a small time frame. Besides, that for all users that are not affected by it have to wait even unnecessary longer to come up.

I cant test changes because I do not own any Logitech Cams. So feel free to add a PR with a fix.
It has to be done in the develop branch wich changes its behavio completly, because if you are using camera-streamer as backend its no longer done via v4l2-ctl, its handled by start parameters of camera-streamer. Therefor no possibilty to delay that commands.

Sidkrsk, 2023-04-21:
None of the above methods of solving this problem helped me, so I went ahead and found a way in which they add a file to the directory /etc/udev/rules.d/99-video-c920.rules, but the configurations described there also did not apply. I searched again and found this [archlinux](https://wiki.archlinux.org/title/Webcam_setup#Persisting_configuration_changes) I checked the device id and replaced it with the necessary parameters and hooray, everything worked for me! I hope this method will help someone)

chrishoage, 2023-04-30:
> I have the same C920. Had the exact same issue. Changing focus_auto = 0 to focus_automatic_continuous = 0 solved it for me.

For anyone coming from google this may not be just a C920 issue. I have a C615 and had the exact same issue, along with the same solution. 

This started happening after a dist-upgrade on the klipper host - so I'm guessing either the v4l2 binary changed or the driver interface. 

In any case - wanted to share that this solution may work for other Logitech cameras

KwadFan, 2023-05-05:
Thanks for reporting, the funny thing is I do nothing other than udev rule ^^
Dont know why Logitechs are that picky ...

Will think about implementing a custom_script thingy.
I have no real clue on how to solve that now, but will somehow add a feature to run a custom script before the streamer launches.

Hurry up, will take time...

cliffiard, 2023-05-19:
> I have the same C920. Had the exact same issue. Changing focus_auto=0 to focus_automatic_continuous=0 solved it for me.

How did you made the changes? Could you please direct me on the right way?

austinrdennis, 2023-09-10:
> > I have the same C920. Had the exact same issue. Changing focus_auto=0 to focus_automatic_continuous=0 solved it for me.
> 
> How did you made the changes? Could you please direct me on the right way?

Put this in the cam section of your crowsnest.conf file, specifically pay attention to the last line:

[cam 1]
mode: camera-streamer
	
enable_rtsp: false
rtsp_port: 8554
port: 8080
device: /dev/video0
resolution: 1366✕768
max_fps: 15
**v4l2ctl: focus_automatic_continuous=0, focus_absolute=45**

tastyratz, 2023-09-22:
Came here to create a report but it looks like people are finding the same thing I did.
If you focus_auto does not work but focus_automatic_continuous works on a c920.

Setting focus_absolute however is ignored when set on the same line in crowsnest:
v4l2ctl: focus_automatic_continuous=0,focus_absolute=55 

If I manually run  from SSH:
sudo v4l2-ctl --device=/dev/video0 -c focus_absolute=55

The camera comes into focus perfectly!
It looks like chaining these commands is not compatible when the focus state is already set to auto as shown in the SSH session below:

pi@mainsailos:~ $ sudo v4l2-ctl --device=/dev/video0 -c focus_automatic_continuous=0
pi@mainsailos:~ $ sudo v4l2-ctl --device=/dev/video0 -c focus_automatic_continuous=0,focus_absolute=55
pi@mainsailos:~ $ sudo v4l2-ctl --device=/dev/video0 -c focus_automatic_continuous=1
pi@mainsailos:~ $ sudo v4l2-ctl --device=/dev/video0 -c focus_automatic_continuous=0,focus_absolute=55
VIDIOC_S_EXT_CTRLS: failed: Invalid or incomplete multibyte or wide character
focus_absolute: Invalid or incomplete multibyte or wide character

I am guessing this is where crowsnest is dying. It's not able to turn off autofocus and THEN set a focal distance, it's trying to set a focal distance and disable autofocus simultaneously.

AlexR-maker, 2023-10-20:
Hi,
I was having the same issue with a Logitech C615, broken focus message.
I'm using crowsnest.
After reading this thread I've changed the settings on Crowsnest to:
v4l2ctl: focus_automatic_continuous=0,focus_absolute=119
but the trick was to use multiples of 17 for the focus_absolute (actually is what is suggested in crowsnest.log: "min=0 max=255 step=17 default=51 value=51 flags=inactive").

tastyratz, 2023-10-27:
For the C920 my log reveals the following:
 crowsnest:		focus_absolute 0x009a090a (int)    : min=0 max=250 step=5 default=0 value=0 flags=inactive

It looks like a step size of 5 . I was not having issues related to step size though and my final value would be 55 if it functioned. 
Looks like a good find though for the c615!

If there was some way to send multiple commands sequentially or with delay with crowsnest I think the problem could be solved that simply for many.

jwyse, 2023-11-10:
I'm not ready to submit a PR with a fix yet, but I have some more information about the issue and an approach that fixed it for me.

First, this issue isn't specific to Logitech cameras.  I'm using [this camera](https://www.amazon.com/gp/product/B0CG4BYL7Z/) and having the same issue trying to set the focus in one line via `v4l2ctl: focus_automatic_continuous=0,focus_absolute=10)`.

Here's what I see happening:  In `ustreamer.sh` it starts up ustreamer via the run_ustreamer function (each function call in its own thread) for each configured camera, and then calls the `brokenfocus` function to (re)apply the focus_absolute parameter.  

```sh
run_mjpg() {
    local cams
    v4l2_control
    cams="${1}"
    for instance in ${cams} ; do
        run_ustreamer "${instance}" &
    done
    blockyfix
    brokenfocus
    return
}
```

The issue is that `blockyfix`+`brokenfocus` get called immediately _after_ all the thread(s) of `run_ustreamer` function get started, but likely _before_ any/all of those threads have finished their checks and actually launched ustreamer process(es).

I added some logging output to `v4l2_control.sh` and I see that when `brokenfocus` gets the current value of `focus_absolute` to see if it matches the configured value, it WILL match initially -- but once the ustreamer process starts (in another thread), focus_absolute will be set to the default value.  For my camera/config, focus_absolute default value is 370, and I'm trying to set it to 10.  Note that I'm repeatedly reading cur_val and sleeping 0.1s, and note the value changes after the ustreamer process starts (in another thread).  My results are the same for different sleep durations; regardless of the timing, the value read will be updated after ustreamer starts.

```
[11/09/23 18:39:14] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue ...
[11/09/23 18:39:16] crowsnest: V4L2 Control: Device: [cam 1]
[11/09/23 18:39:16] crowsnest: V4L2 Control: Options: focus_automatic_continuous=0,focus_absolute=10
[...]
[11/09/23 18:39:16] crowsnest: DEBUG: v4l2ctl: focus_absolute 0x009a090a (int)    : min=0 max=1023 step=1 default=370 value=10
[11/09/23 18:39:16] crowsnest: DEBUG: v4l2ctl: focus_automatic_continuous 0x009a090c (bool)   : default=1 value=0
[11/09/23 18:39:16] crowsnest: V4L2 Control: (blockyfix...)
[11/09/23 18:39:18] crowsnest: V4L2 Control: (...blockyfix)

[11/09/23 18:39:18] crowsnest: V4L2 Control: (brokenfocus...)
[11/09/23 18:39:19] crowsnest: V4L2 Control: cur_val focus_absolute=10 (sleeping 0.1s ...)
[11/09/23 18:39:19] crowsnest: V4L2 Control: cur_val focus_absolute=10 (sleeping 0.1s ...)
[11/09/23 18:39:19] crowsnest: V4L2 Control: cur_val focus_absolute=10 (sleeping 0.1s ...)
[11/09/23 18:39:19] crowsnest: V4L2 Control: cur_val focus_absolute=10 (sleeping 0.1s ...)
[11/09/23 18:39:19] crowsnest: V4L2 Control: cur_val focus_absolute=10 (sleeping 0.1s ...)
[11/09/23 18:39:19] crowsnest: V4L2 Control: cur_val focus_absolute=10 (sleeping 0.1s ...)
[11/09/23 18:39:20] crowsnest: V4L2 Control: cur_val focus_absolute=10 (sleeping 0.1s ...)

[11/09/23 18:39:20] crowsnest: Starting ustreamer with Device /dev/v4l/by-id/usb-8MP_USB_Camera_8MP_USB_Camera_2022051301-video-index0 ...
[11/09/23 18:39:20] crowsnest: DEBUG: ustreamer [cam 1]: Parameters: --host 127.0.0.1 -p 8080 -d /dev/v4l/by-id/usb-8MP_USB_Camera_8MP_USB_Camera_2022051301-video-index0 --device-timeout=2 -m MJPEG --encoder=HW -r 1920x1080 -f 10 --allow-origin=* --static /home/jwyse/crowsnest/ustreamer-www
[...]
[11/09/23 18:39:20] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO  [787782.790    stream] -- Capturing ...

[11/09/23 18:39:20] crowsnest: V4L2 Control: cur_val focus_absolute=370 (sleeping 0.1s ...)
[11/09/23 18:39:20] crowsnest: V4L2 Control: cur_val focus_absolute=370 (sleeping 0.1s ...)
[11/09/23 18:39:21] crowsnest: V4L2 Control: cur_val focus_absolute=370 (sleeping 0.1s ...)
[11/09/23 18:39:21] crowsnest: V4L2 Control: cur_val focus_absolute=370 (sleeping 0.1s ...)
[11/09/23 18:39:21] crowsnest: V4L2 Control: cur_val focus_absolute=370 (sleeping 0.1s ...)
[11/09/23 18:39:21] crowsnest: V4L2 Control: cur_val focus_absolute=370 (sleeping 0.1s ...)
[11/09/23 18:39:21] crowsnest: V4L2 Control: cur_val focus_absolute=370 (sleeping 0.1s ...)
[11/09/23 18:39:21] crowsnest: V4L2 Control: cur_val focus_absolute=370 (sleeping 0.1s ...)
[11/09/23 18:39:22] crowsnest: V4L2 Control: cur_val focus_absolute=370 (sleeping 0.1s ...)
[11/09/23 18:39:22] crowsnest: V4L2 Control: cur_val focus_absolute=370 (sleeping 0.1s ...)
[11/09/23 18:39:22] crowsnest: V4L2 Control: cur_val focus_absolute=370 (sleeping 0.1s ...)
[11/09/23 18:39:22] crowsnest: V4L2 Control: cur_val focus_absolute=370 (sleeping 0.1s ...)

[11/09/23 18:39:23] crowsnest: V4L2 Control: cam 1 ; device /dev/v4l/by-id/usb-8MP_USB_Camera_8MP_USB_Camera_2022051301-video-index0 ; cur_val focus_absolute=370 ; conf_val focus_absolute=10

[11/09/23 18:39:23] crowsnest: WARN: Detected 'brokenfocus' device.
[11/09/23 18:39:23] crowsnest: INFO: Trying to set to configured Value.
[11/09/23 18:39:23] crowsnest: DEBUG: Value is now: focus_absolute=10
[11/09/23 18:39:23] crowsnest: V4L2 Control: (...brokenfocus)
[11/09/23 18:39:25] crowsnest:  ... Done!
```

[... 47 lines cut ...]
[11/09/23 19:53:19] crowsnest: Waiting for ustreamer process to start...
[11/09/23 19:53:19] crowsnest: Starting ustreamer with Device /dev/v4l/by-id/usb-8MP_USB_Camera_8MP_USB_Camera_2022051301-video-index0 ...
[... cut ...]
[11/09/23 19:53:30] crowsnest:  ... Done!

```

Additional ideas:

- `brokenfocus` could be refactored to be called within each `run_ustreamer` function, and only check/fix the corresponding camera instead of doing its own iteration over configured cameras.
- `pidof` could be used to get the PIDs of all ustreamer processes, and check to make sure the number of instances matches the number of configured cameras.  This may not work as intended if the machine has any other ustreamer processes running outside of Crowsnest.

mryel00, 2023-11-11:
Good to see that someone is working on this and having some good ideas on doing so😄❤️
Always hard to work on a fix for something you cannot reproduce. I want to give you some hints you might want to consider on implementing this.

> This may not work properly when multiple cameras are configured

This should be considered for a fix. I have already seen quite some setups with two cams with one  or more Logitech cams. So only checking for one instance can make problems there.

> or if any ustreamer process is running outside of Crowsnest.

Just ignore such a case. You cannot achieve a satisfying solution for all edge cases. The chances of someone running ustreamer and Crowsnest simultaneously should be considered pretty low or equal to zero.

> `brokenfocus` could be refactored to be called within each `run_ustreamer` function, and only check/fix the corresponding camera instead of doing its own iteration over configured cameras.

I don't think this would be possible. Have a look at the last few lines of [ustreamer.sh](https://github.com/mainsail-crew/crowsnest/blob/master/libs/ustreamer.sh#L76-L79). There it starts the streamer and if it fails it will log an error. So for this you would need some completely different method of catching an error with the streamer. Currently it's checking if line 76 & 77 are "finished". If they are finished there is a problem with the streamer and  the streamer failed.

> `pidof` could be used to get the PIDs of all ustreamer processes, and check to make sure the number of instances matches the number of configured cameras.

This might be the best solution for this. We already did some testing on detecting running streaming instances for some other thing. It seems like we have forgotten about `pidof`.
I also talked with @KwadFan about this and he had the further idea about saving the PID of Crowsnest and then only check the childs of the Crowsnest process wiuth `pidof`. This would handle that stuff with ustreamer instances outside of Crowsnest and would be for a "perfect" handling. But like I said earlier, such case can be ignored if it makes problems to implement something.

So overall the code you attached seems already very promising and we would be happy about a PR with the support for multiple instances.

Obagunceiro, 2024-01-14:
Not working here, apparently the brokenfocus function is never executed and so my focus is never set, and always ends up returning to the webcam's default focus, I never get the message WARN: Detected 'brokenfocus' device

mryel00, 2024-01-14:
@Obagunceiro What is not working for you? The "fix" provided by [jwyse](https://github.com/jwyse)? The `brokenfocus` function that you mentioned in your other issue #232? Information is key, otherwise no one knows what you are talking about.
Also don't start a new issue and then go to an old one. First search for old ones and if you cannot find one, then open a new one.

Obagunceiro, 2024-01-14:
> @Obagunceiro What is not working for you? The "fix" provided by [jwyse](https://github.com/jwyse)? The `brokenfocus` function that you mentioned in your other issue #232? Information is key, [...]

Both aren't working, the fix doesn't work and probably in my case it's because the function isn't called, and I searched all over the internet, tried to make small changes myself to solve the issue But without success, as this issue is still open, I thought I could try to discuss it here too, and considering that I believe the root of the problem is something else, I opened another issue, containing all the details that I think are important in principle about this new problem.

mryel00, 2024-01-14:
> in my case it's because the function isn't called

The function is called, as you can see in your log. Just at the moment of the function call it's still the value it got set too. So for this "fix" to work for you, you might need to increase the delay. But I won't give any support on this, as it's not my code. I can assure you that we will try to fix this in future.
Anyway, I will ask you some questions in the other issue as there is something weird in the log.

Adblu, 2024-01-20:
Hi. need help setting up focus value. 

`User Controls

                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=128
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=128 value=128
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=128 value=128
        white_balance_automatic 0x0098090c (bool)   : default=1 value=1
                           gain 0x00980913 (int)    : min=0 max=255 step=1 default=0 value=0
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2 (60 Hz)
      white_balance_temperature 0x0098091a (int)    : min=2000 max=7500 step=10 default=4000 value=4000 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=255 step=1 default=128 value=128
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=0 value=0

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=3 (Aperture Priority Mode)
         exposure_time_absolute 0x009a0902 (int)    : min=3 max=2047 step=1 default=127 value=156 flags=inactive
     exposure_dynamic_framerate 0x009a0903 (bool)   : default=0 value=1
                   pan_absolute 0x009a0908 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
                  tilt_absolute 0x009a0909 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
                 focus_absolute 0x009a090a (int)    : min=0 max=255 step=1 default=0 value=255 flags=inactive
     focus_automatic_continuous 0x009a090c (bool)   : default=1 value=1
                  zoom_absolute 0x009a090d (int)    : min=100 max=140 step=10 default=100 value=140
`

I have such option, but when setting:
`(base) n@n:~$ sudo v4l2-ctl --device=6 --set-ctrl=focus_absolute=255
VIDIOC_S_EXT_CTRLS: failed: Permission denied
Error setting controls: Permission denied
`

Could you pelase help guide me throu changes in the file you mentiond ? where is this file ?

mryel00, 2024-01-20:
@Adblu please don't ask for help in any issues. If you want some help, join our [Discord](https://discord.gg/mainsail) or use our [GitHub discussions](https://github.com/orgs/mainsail-crew/discussions). But before you go to get help read everything in this message.
For more informations you can read through this: https://docs.mainsail.xyz/faq/getting-help

As this code modification isn't official, we don't give support on it, and I don't reocmmend to do it, if you cannot even find the file it's in on your own.
Furthermore you issue isn't even connected to this one according to you message as you might using the command wrong.
For a tutorial on how to set them read this: https://crowsnest.mainsail.xyz/configuration/cam-section#v4l2ctl

ToddLorey, 2024-07-28:
FWIW, the fix by @jwyse worked for me too.  It's a no-name 8MP auto focus camera in the narrow "Angrycam" format from Amazon.

ToddLorey, 2024-08-19:
FWIW, I made use of gcode_shell_command and created a little ecosystem that automatically sets focus after startup (which is what wasn't working for me) and it also creates a macro with focus value parameter that you can send from the console.  If you're interested:
https://github.com/ToddLorey/v4l2-ctl-KlipperFocus.git

k4Zo12, 2025-03-24:
> FWIW, I made use of gcode_shell_command and created a little ecosystem that automatically sets focus after startup (which is what wasn't working for me) and it also creates a macro with focus value [...]

Unreal... I finally just stumbled across this late last night after fighting with this issue in multiple different contexts over the past couple months. Thanks so much @ToddLorey !! My intuition was telling me this was the issue, and I tried playing around with ways to 'fake' a delay via the commands in the crowsnest config, but no dice. This is a super clean and handy solution! I'll be implementing this twice for two similar webcam modules on my printer - one for a bed cam, and one for a nozzle cam - both of which were having the same issues. Also super handy to have the macro accessible to change the absolute focus on the fly. 

Thanks again!!

mryel00, 2025-03-24:
A small Disclaimer again:
This issue is not an issue with Crowsnest but an issue with the cams themself. They are reporting back wrong values, therefore we cannot detect, if the focus applied correctly or not.
This can be fixed by adding a delay shown [here](https://github.com/mainsail-crew/crowsnest/issues/96#issuecomment-1804916511) but this is not an official workaround.
I will consider this issue and workaround with #270 as I can detect the running process way easier with that rewrite.

Also worth mentioning, this issue might be already fixed in camera-streamer, at least for some cams, as of this commit: https://github.com/ayufan/camera-streamer/commit/8d06faffd905b0bc0ab5916a7951244915b82340d
It also mentions the detail, that it's a cam issue and that the cam seems to only accept focus parameters, once the stream started, therefore a delay can fix it. That commit does not only apply to 3DO cams (mentioned in the commit message), but all cameras with similar issues.

mryel00, 2026-04-17:
I think this should be solved now, if not please let me know.
I basically added a 0.5s sleep after a ustreamer cam starts and then apply the focus again. So the same thing as @jwyse, thanks again for that.