mainsail-crew/crowsnest #97: Crowsnest.service cannot run

MS1987, 2023-03-17:
### What happened

I installed crowsnest on my armbian system(based on rockchip rk3328), but it cannot run ,and when I type:
```
sudo service crowsnest status > log
cat log
```

It shows:
```
● crowsnest.service - crowsnest - Multi Webcam/Streamer Control Deamon
   Loaded: loaded (/etc/systemd/system/crowsnest.service; enabled; vendor preset: enabled)
   Active: **failed** (Result: exit-code) since Fri 2023-03-17 19:57:28 HKT; 37s ago
     Docs: https://github.com/mainsail-crew/crowsnest
  Process: 6063 ExecStart=/usr/local/bin/crowsnest $CROWSNEST_ARGS (code=exited, status=1/FAILURE)
 Main PID: 6063 (code=exited, status=1/FAILURE)

systemd[1]: crowsnest.service: Service RestartSec=5s expired, scheduling restart.
systemd[1]: crowsnest.service: Scheduled restart job, restart counter is at 10.
systemd[1]: Stopped crowsnest - Multi Webcam/Streamer Control Deamon.
systemd[1]: crowsnest.service: Start request repeated too quickly.
systemd[1]: crowsnest.service: Failed with result 'exit-code'.
systemd[1]: Failed to start crowsnest - Multi Webcam/Streamer Control Deamon.
```

And the crowsnest.log show:
```
……

[03/17/23 20:07:29] crowsnest: INFO: Print Configfile: '/home/mks/printer_data/config/crowsnest.conf'
[03/17/23 20:07:29] crowsnest:          [crowsnest]
[03/17/23 20:07:29] crowsnest:          log_path: ~/printer_data/logs/crowsnest.log
[03/17/23 20:07:29] crowsnest:          log_level: verbose
[03/17/23 20:07:29] crowsnest:          delete_log: false
[03/17/23 20:07:29] crowsnest:
[03/17/23 20:07:29] crowsnest:          [cam 1]
[03/17/23 20:07:29] crowsnest:          mode: mjpg
[03/17/23 20:07:29] crowsnest:          port: 8080
[03/17/23 20:07:29] crowsnest:          device: /dev/video4
[03/17/23 20:07:30] crowsnest:          resolution: 640x480
[03/17/23 20:07:30] crowsnest:          max_fps: 15
[03/17/23 20:07:30] crowsnest: INFO: Detect available Devices
[03/17/23 20:07:30] crowsnest: INFO: Found 1 total available Device(s)
[03/17/23 20:07:30] crowsnest: INFO: Found 1 available camera(s)
[03/17/23 20:07:30] crowsnest: /dev/v4l/by-id/usb-lihappe8_Corp._USB_2.0_Camera-video-index0 -> /dev/video4
[03/17/23 20:07:30] crowsnest: Supported Formats:
[03/17/23 20:07:30] crowsnest:
[03/17/23 20:07:30] crowsnest: ERROR: Error 1 occured on line 70
[03/17/23 20:07:30] crowsnest: ERROR: Stopping crowsnest.
[03/17/23 20:07:30] crowsnest: Goodbye...
[03/17/23 20:07:30] crowsnest: ERROR: Error 1 occured on line 70
[03/17/23 20:07:30] crowsnest: ERROR: Stopping crowsnest.
[03/17/23 20:07:30] crowsnest: Goodbye...
[03/17/23 20:07:30] crowsnest: ERROR: Error 1 occured on line 27
[03/17/23 20:07:30] crowsnest: ERROR: Stopping crowsnest.
[03/17/23 20:07:30] crowsnest: Goodbye...

```

But if I run the crowsnest manually:
`~/crowsnest/crowsnest -c ~/printer_data/config/crowsnest.conf`
It can work!

So what is the possibility?

### What did you expect to happen

crowsnest can run using service

### How to reproduce

armbian+rk3328

KwadFan, 2023-03-18:
Please show content of service file, maybe you tried to install it as root user and didnt run the `sudo make install` as user klipper

MS1987, 2023-03-19:
@KwadFan here is my service file:
```
#### crowsnest - A webcam Service for multiple Cams and Stream Services.
####
#### Written by Stephan Wendel aka KwadFan <me@stephanwe.de>
#### Copyright 2021 - 2022
#### https://github.com/mainsail-crew/crowsnest
####
#### This File is distributed under GPLv3
####

[Unit]
Description=crowsnest - Multi Webcam/Streamer Control Deamon
Documentation=https://github.com/mainsail-crew/crowsnest
After=udev.service network-online.target nss-lookup.target
Wants=udev.service network-online.target
Before=nginx.service
StartLimitBurst=10
StartLimitIntervalSec=180

[Install]
WantedBy=multi-user.target

[Service]
Type=simple
User=klipper
RemainAfterExit=Yes
WorkingDirectory=/home/klipper/crowsnest
EnvironmentFile=/home/klipper/printer_data/systemd/crowsnest.env
ExecStart= /usr/local/bin/crowsnest $CROWSNEST_ARGS
Restart=on-failure
RestartSec=5
```

MS1987, 2023-03-21:
I finally found the reason is I added the user to the group of root, remove it solve the problem.

closed 2023-03-21 by MS1987