mainsail-crew/crowsnest #98: Failed to start with error “Error: error 1 occurred on line 150. Error: Stopping crowsnest”

Adubs88, 2023-03-17:
### What happened

Crowsnest fails to start. Unknown reason why. Running newest versions of mainsail, Klipper, etc. updated today 3/17/23

### What did you expect to happen

Launch of crowsnest and video monitoring capabilities.

### How to reproduce

Attempting to launch crowsnest through mainsail os, or via ssh with the command

~/crowsnest/crowsnest -c ~/printer_data/config/crowsnest.con 

yields same results.

### Additional information

  [03/17/23 23:41:31] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[03/17/23 23:41:31] crowsnest: Version: v3.0.7-3-g20ed6a8
[03/17/23 23:41:31] crowsnest: Prepare Startup ...
[03/17/23 23:41:31] crowsnest: INFO: Host information:
[03/17/23 23:41:31] crowsnest: Host Info: Distribution: BTT-CB1 2.3.2 Bullseye
[03/17/23 23:41:31] crowsnest: Host Info: Kernel: Linux 5.16.17-sun50iw9 aarch64
[03/17/23 23:41:31] crowsnest: Host Info: Available CPU Cores: 4
[03/17/23 23:41:31] crowsnest: Host Info: Available Memory: 1010640 kB
[03/17/23 23:41:31] crowsnest: Host Info: Diskspace (avail. / total): 2.5G / 7.1G
[03/17/23 23:41:32] crowsnest: INFO: Checking Dependencys
[03/17/23 23:41:32] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[03/17/23 23:41:32] crowsnest: Dependency: 'find' found in /usr/bin/find.
[03/17/23 23:41:32] crowsnest: Dependency: 'logger' found in /usr/bin/logger.
[03/17/23 23:41:32] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[03/17/23 23:41:32] crowsnest: Dependency: 'ffmpeg' found in /usr/bin/ffmpeg.
[03/17/23 23:41:32] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/ustreamer.
[03/17/23 23:41:32] crowsnest: Dependency: 'rtsp-simple-server' found in bin/rtsp-simple-server/rtsp-simple-server.
[03/17/23 23:41:32] crowsnest: Version Control: ustreamer is up to date. (v5.38)
[03/17/23 23:41:32] crowsnest: Version Control: rtsp-simple-server is up to date. (v0.20.2)
[03/17/23 23:41:33] crowsnest: Version Control: ffmpeg is up to date. (4.3.5-0+deb11u1)
[03/17/23 23:41:33] crowsnest: INFO: Print Configfile: '/home/biqu/printer_data/config/crowsnest.conf'
[03/17/23 23:41:33] crowsnest: 		[crowsnest]
[03/17/23 23:41:33] crowsnest: 		log_path: ~/printer_data/logs/crowsnest.log
[03/17/23 23:41:33] crowsnest: 		log_level: debug
[03/17/23 23:41:33] crowsnest: 		delete_log: true
[03/17/23 23:41:33] crowsnest: 		no_proxy: true
[03/17/23 23:41:33] crowsnest: 		
[03/17/23 23:41:33] crowsnest: 		[cam Creality Cam]
[03/17/23 23:41:33] crowsnest: 		mode: mjpg
[03/17/23 23:41:33] crowsnest: 		port: 8080
[03/17/23 23:41:33] crowsnest: 		device: /dev/v4l/by-path/platform-5200000.usb-usb-0:1.1:1.0-video-index0
[03/17/23 23:41:33] crowsnest: 		resolution: 1920x1080
[03/17/23 23:41:33] crowsnest: 		max_fps: 30
[03/17/23 23:41:33] crowsnest: 		brightness 0x00980900 (int) : 128
[03/17/23 23:41:33] crowsnest: 		contrast 0x00980901 (int) : 128
[03/17/23 23:41:33] crowsnest: 		saturation 0x00980902 (int) : 128
[03/17/23 23:41:33] crowsnest: 		white_balance_automatic 0x0098090c (bool) : 1
[03/17/23 23:41:33] crowsnest: 		gain 0x00980913 (int) : 50
[03/17/23 23:41:33] crowsnest: 		power_line_frequency 0x00980918 (menu) : 2
[03/17/23 23:41:33] crowsnest: 		white_balance_temperature 0x0098091a (int) : 4650
[03/17/23 23:41:33] crowsnest: 		sharpness 0x0098091b (int) : 128
[03/17/23 23:41:33] crowsnest: 		uto_exposure 0x009a0901 (menu) : 0
[03/17/23 23:41:33] crowsnest: 		exposure_time_absolute 0x009a0902 (int) : 5
[03/17/23 23:41:34] crowsnest: 		exposure_dynamic_framerate 0x009a0903 (bool) : 0
[03/17/23 23:41:34] crowsnest: 		
[03/17/23 23:41:34] crowsnest: 		v4l2ctl: focus_auto=0,focus_absolute=30
[03/17/23 23:41:34] crowsnest: 		
[03/17/23 23:41:34] crowsnest: INFO: Detect available Devices
[03/17/23 23:41:34] crowsnest: INFO: Found 1 total available Device(s)
[03/17/23 23:41:34] crowsnest: INFO: Found 1 available camera(s)
[03/17/23 23:41:34] crowsnest: /dev/v4l/by-id/usb-Creality_3D_Technology_CREALITY_CAM_V20210304STZRF37_30-video-index0 -> /dev/video0
[03/17/23 23:41:34] crowsnest: Supported Formats:
[03/17/23 23:41:34] crowsnest: 		[0]: 'H264' (H.264, compressed)
[03/17/23 23:41:34] crowsnest: 		Size: Discrete 1280x720
[03/17/23 23:41:34] crowsnest: 		Interval: Discrete 0.033s (30.000 fps)
[03/17/23 23:41:34] crowsnest: 		Interval: Discrete 0.033s (30.000 fps)
[03/17/23 23:41:34] crowsnest: 		Size: Discrete 720x1080
[03/17/23 23:41:34] crowsnest: 		Interval: Discrete 0.033s (30.000 fps)
[03/17/23 23:41:34] crowsnest: 		Size: Discrete 640x480
[03/17/23 23:41:34] crowsnest: 		Interval: Discrete 0.033s (30.000 fps)
[03/17/23 23:41:34] crowsnest: 		Size: Discrete 480x640
[03/17/23 23:41:34] crowsnest: 		Interval: Discrete 0.033s (30.000 fps)
[03/17/23 23:41:34] crowsnest: 		Size: Discrete 1920x1080
[03/17/23 23:41:34] crowsnest: 		Interval: Discrete 0.033s (30.000 fps)
[03/17/23 23:41:34] crowsnest: 		Size: Discrete 1280x720
[03/17/23 23:41:34] crowsnest: 		Interval: Discrete 0.033s (30.000 fps)
[03/17/23 23:41:34] crowsnest: 		Interval: Discrete 0.033s (30.000 fps)
[03/17/23 23:41:34] crowsnest: 		[1]: '' (35363248-0000-0010-8000-00aa003, compressed)
[03/17/23 23:41:34] crowsnest: Supported Controls:
[03/17/23 23:41:34] crowsnest: 		
[03/17/23 23:41:34] crowsnest: 		User Controls
[03/17/23 23:41:34] crowsnest: 		
[03/17/23 23:41:34] crowsnest: 		brightness 0x00980900 (int) : min=1 max=255 step=1 default=128 value=128
[03/17/23 23:41:34] crowsnest: 		contrast 0x00980901 (int) : min=1 max=255 step=1 default=128 value=128
[03/17/23 23:41:34] crowsnest: 		saturation 0x00980902 (int) : min=1 max=255 step=1 default=128 value=128
[03/17/23 23:41:34] crowsnest: 		white_balance_automatic 0x0098090c (bool) : default=1 value=1
[03/17/23 23:41:35] crowsnest: 		gain 0x00980913 (int) : min=1 max=100 step=1 default=50 value=50
[03/17/23 23:41:35] crowsnest: 		power_line_frequency 0x00980918 (menu) : min=0 max=2 default=1 value=1
[03/17/23 23:41:35] crowsnest: 		0: Disabled
[03/17/23 23:41:35] crowsnest: 		1: 50 Hz
[03/17/23 23:41:35] crowsnest: 		2: 60 Hz
[03/17/23 23:41:35] crowsnest: 		white_balance_temperature 0x0098091a (int) : min=2800 max=6500 step=1 default=4650 value=4650 flags=inactive
[03/17/23 23:41:35] crowsnest: 		sharpness 0x0098091b (int) : min=1 max=255 step=1 default=128 value=128
[03/17/23 23:41:35] crowsnest: 		
[03/17/23 23:41:35] crowsnest: 		Camera Controls
[03/17/23 23:41:35] crowsnest: 		
[03/17/23 23:41:35] crowsnest: 		auto_exposure 0x009a0901 (menu) : min=0 max=3 default=0 value=0
[03/17/23 23:41:35] crowsnest: 		0: Auto Mode
[03/17/23 23:41:35] crowsnest: 		2: Shutter Priority Mode
[03/17/23 23:41:35] crowsnest: 		exposure_time_absolute 0x009a0902 (int) : min=5 max=2500 step=1 default=5 value=5 flags=inactive
[03/17/23 23:41:35] crowsnest: 		exposure_dynamic_framerate 0x009a0903 (bool) : default=0 value=0
[03/17/23 23:41:35] crowsnest: INFO: No usable CSI Devices found.
[03/17/23 23:41:36] crowsnest: V4L2 Control: No parameters set for [cam Creality]. Skipped.
[03/17/23 23:41:37] crowsnest: V4L2 Control: No parameters set for [cam Cam]. Skipped.
[03/17/23 23:41:39] crowsnest: ERROR: Error 1 occured on line 150
[03/17/23 23:41:39] crowsnest: ERROR: Stopping crowsnest.
[03/17/23 23:41:39] crowsnest: Goodbye...

suizhenmin, 2023-03-18:
Had the same problem yesterday after updating the system components

KwadFan, 2023-03-18:
Well, to be honest, in the past few weeks it seems there are more and more users with issues especcialy with Logitech ones.

Please try to set your v4l2ctl parameters via ssh. If you face something like https://github.com/mainsail-crew/crowsnest/issues/96 there isnt much I cand do, v4l2-ctl command simply refuses to set these values

Adubs88, 2023-03-18:
@KwadFan thanks for the reply. This is with a Wyze cam flashed to webcam firmware provided by the developers.  Not running a Logitech.. Unfortunately I am new to using Klipper, mainsail, and crowsnest, so it is difficult for me to determine if the issues are user error on my end, or software issues, and bugs that are out of my hand. Can you please elaborate on setting the V4L2ctL parameters via SSH? I read the post you linked, too, and it’s not 100% clear. Can you explain it to me like I’m five? I’ve spent so many hours trying to get this working and it seems like it may not have even been that I was doing something wrong. Any help would be appreciated.

KwadFan, 2023-03-18:
Ah! I see, sorry I literally flown over the log not realizing that it is not an Issue related to a Logitech cam.

So, after reading your Log snippet, some bad news. To use a camera with crowsnest it has to be capable to deliver MJPG or at least YUYV format. Your camera doesnt support either. So, I am sorry to say but crowsnest cant help you here.

These are the supported protocols by your cam:

```[03/17/23 23:41:34] crowsnest: /dev/v4l/by-id/usb-Creality_3D_Technology_CREALITY_CAM_V20210304STZRF37_30-video-index0 -> /dev/video0
[03/17/23 23:41:34] crowsnest: Supported Formats:
[03/17/23 23:41:34] crowsnest: [0]: 'H264' (H.264, compressed)
[03/17/23 23:41:34] crowsnest: Size: Discrete 1280x720
[03/17/23 23:41:34] crowsnest: Interval: Discrete 0.033s (30.000 fps)
[03/17/23 23:41:34] crowsnest: Interval: Discrete 0.033s (30.000 fps)
[03/17/23 23:41:34] crowsnest: Size: Discrete 720x1080
[03/17/23 23:41:34] crowsnest: Interval: Discrete 0.033s (30.000 fps)
[03/17/23 23:41:34] crowsnest: Size: Discrete 640x480
[03/17/23 23:41:34] crowsnest: Interval: Discrete 0.033s (30.000 fps)
[03/17/23 23:41:34] crowsnest: Size: Discrete 480x640
[03/17/23 23:41:34] crowsnest: Interval: Discrete 0.033s (30.000 fps)
[03/17/23 23:41:34] crowsnest: Size: Discrete 1920x1080
[03/17/23 23:41:34] crowsnest: Interval: Discrete 0.033s (30.000 fps)
[03/17/23 23:41:34] crowsnest: Size: Discrete 1280x720
[03/17/23 23:41:34] crowsnest: Interval: Discrete 0.033s (30.000 fps)
[03/17/23 23:41:34] crowsnest: Interval: Discrete 0.033s (30.000 fps)
[03/17/23 23:41:34] crowsnest: [1]: '' (35363248-0000-0010-8000-00aa003, compressed)```

A typical cam would report ```[03/18/23 14:03:37] crowsnest: 		[1]: 'MJPG' (Motion-JPEG, compressed)```

Also you misinterpreted the setup
```[03/17/23 23:41:33] crowsnest: [cam Creality Cam]
[03/17/23 23:41:33] crowsnest: mode: mjpg
[03/17/23 23:41:33] crowsnest: port: 8080
[03/17/23 23:41:33] crowsnest: device: /dev/v4l/by-path/platform-5200000.usb-usb-0:1.1:1.0-video-index0
[03/17/23 23:41:33] crowsnest: resolution: 1920x1080
[03/17/23 23:41:33] crowsnest: max_fps: 30
[03/17/23 23:41:33] crowsnest: brightness 0x00980900 (int) : 128
[03/17/23 23:41:33] crowsnest: contrast 0x00980901 (int) : 128
[03/17/23 23:41:33] crowsnest: saturation 0x00980902 (int) : 128
[03/17/23 23:41:33] crowsnest: white_balance_automatic 0x0098090c (bool) : 1
[03/17/23 23:41:33] crowsnest: gain 0x00980913 (int) : 50
[03/17/23 23:41:33] crowsnest: power_line_frequency 0x00980918 (menu) : 2
[03/17/23 23:41:33] crowsnest: white_balance_temperature 0x0098091a (int) : 4650
[03/17/23 23:41:33] crowsnest: sharpness 0x0098091b (int) : 128
[03/17/23 23:41:33] crowsnest: uto_exposure 0x009a0901 (menu) : 0
[03/17/23 23:41:33] crowsnest: exposure_time_absolute 0x009a0902 (int) : 5
[03/17/23 23:41:34] crowsnest: exposure_dynamic_framerate 0x009a0903 (bool) : 0
[03/17/23 23:41:34] crowsnest:
[03/17/23 23:41:34] crowsnest: v4l2ctl: focus_auto=0,focus_absolute=30
[03/17/23 23:41:34] crowsnest:
[03/17/23 23:41:34] crowsnest: INFO: Detect available Devices
[03/17/23 23:41:34] crowsnest: INFO: Found 1 total available Device(s)
[03/17/23 23:41:34] crowsnest: INFO: Found 1 available camera(s)
[03/17/23 23:41:34] crowsnest: /dev/v4l/by-id/usb-Creality_3D_Technology_CREALITY_CAM_V20210304STZRF37_30-video-index0 -> /dev/video0```

That contains how your config file looks like. Please read the README.md again on how to setup with a different cam. this one isnt supported.

Please avoid white spaces in camera names!! Use for ex. [cam Creality_cam], otherwise crowsnest will interprete them as two different configuration sections. as shown here

```[03/17/23 23:41:36] crowsnest: V4L2 Control: No parameters set for [cam Creality]. Skipped.
[03/17/23 23:41:37] crowsnest: V4L2 Control: No parameters set for [cam Cam]. Skipped.```

Due the fact that this cam hassnt the appropriate Formats I will close this issue, because it is not related to my code, simply it has no usable output format.
Sorry mate :man_shrugging: 

Regards Kwad

closed 2023-03-18 by github-actions[bot]

Adubs88, 2023-03-18:
@KwadFan this issue has been resolved. I see that the log I posted shows only the one camera, and I understand it does not meet the format requirements, so thank you for clarifying that regarding the Creality Cam but i was trying with multiple cameras, and still received the same same error 1 on line 150. That being said, I removed the the "white space" from the name, and it worked immediately. too, now the know what that error refers to!