mainsail-crew/crowsnest #99: PiCam V2.1 / MainsailOS : No image with crowsnest but it's OK with ustreamer

mikebzh44, 2023-03-21:
### What happened

I'm using a PiCam V2.1 on a Raspberry Pi 3 running MainsailOS.
Everything is up to date.
When I start Crowsnest with this conf file : 
`[crowsnest]
log_path: ~/printer_data/logs/crowsnest.log
log_level: debug 
delete_log: false 
[cam 1]
mode: mjpg
port: 8081
device: /dev/video0
resolution: 640x480
max_fps: 15
`
ustreamer is running : 
`[03/21/23 21:39:01] crowsnest: Starting ustreamer with Device /dev/video0 ...
[03/21/23 21:39:01] crowsnest: DEBUG: ustreamer [cam 1]: Parameters: --host 127.0.0.1 -p 8081 -m MJPEG --device-timeout=5 --buffers=3 -r 640x480 -f 15 --allow-origin=* --static /home/pi/crowsnest/ustreamer-www
[03/21/23 21:39:01] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [94206.718 main] -- Starting PiKVM uStreamer 5.38 ...
[03/21/23 21:39:02] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [94206.719 main] -- Using internal blank placeholder
[03/21/23 21:39:02] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [94206.719 main] -- Enabling HTTP file server: /home/pi/crowsnest/ustreamer-www
[03/21/23 21:39:02] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [94206.720 main] -- Listening HTTP on [127.0.0.1]:8081
[03/21/23 21:39:02] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [94206.720 stream] -- Using V4L2 device: /dev/video0
[03/21/23 21:39:02] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [94206.720 stream] -- Using desired FPS: 15
[03/21/23 21:39:02] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [94206.720 http] -- Starting HTTP eventloop ...
[03/21/23 21:39:02] crowsnest: DEBUG: ustreamer [cam 1]: ================================================================================
[03/21/23 21:39:02] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [94206.721 stream] -- Device fd=8 opened
[03/21/23 21:39:02] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [94206.721 stream] -- Using input channel: 0
[03/21/23 21:39:02] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [94206.723 stream] -- Using resolution: 640x480
[03/21/23 21:39:02] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [94206.723 stream] -- Using format: MJPEG
[03/21/23 21:39:02] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [94206.723 stream] -- Using HW FPS: 15
[03/21/23 21:39:02] crowsnest: DEBUG: ustreamer [cam 1]: -- ERROR [94206.723 stream] -- Device doesn't support setting of HW encoding quality parameters
[03/21/23 21:39:02] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [94206.723 stream] -- Using IO method: MMAP
[03/21/23 21:39:02] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [94206.724 stream] -- Requested 3 device buffers, got 3
[03/21/23 21:39:02] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [94207.129 stream] -- Capturing started
[03/21/23 21:39:02] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [94207.129 stream] -- Switching to HW encoder: the input is (M)JPEG ...
[03/21/23 21:39:02] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [94207.129 stream] -- Using JPEG quality: encoder default
[03/21/23 21:39:02] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [94207.129 stream] -- Creating pool JPEG with 1 workers ...
[03/21/23 21:39:02] crowsnest: DEBUG: ustreamer [cam 1]: -- INFO [94207.130 stream] -- Capturing ...`

But I have no stream in Mainsail / Fluidd when I setup the webcam : 

URL stream : /webcam/?action=stream
URL snapshot : /webcam/?action=snapshot
Service : Adaptative MJPEG-Streamer

But when I stop the service from Mainsail UI and I start ustreamer from commande line : 

`./ustreamer --host 192.168.0.182 --port 8081
-- INFO  [94538.964      main] -- Starting PiKVM uStreamer 5.38 ...
-- INFO  [94538.964      main] -- Using internal blank placeholder
-- INFO  [94538.965      main] -- Listening HTTP on [192.168.0.182]:8081
-- INFO  [94538.966    stream] -- Using V4L2 device: /dev/video0
-- INFO  [94538.966    stream] -- Using desired FPS: 0
================================================================================
-- INFO  [94538.968    stream] -- Device fd=8 opened
-- INFO  [94538.968    stream] -- Using input channel: 0
-- INFO  [94538.968      http] -- Starting HTTP eventloop ...
-- INFO  [94538.969    stream] -- Using resolution: 640x480
-- INFO  [94538.969    stream] -- Using format: YUYV
-- INFO  [94538.969    stream] -- Using HW FPS: 0 -> 90 (coerced)
-- INFO  [94538.969    stream] -- Using IO method: MMAP
-- INFO  [94538.971    stream] -- Requested 5 device buffers, got 5
-- INFO  [94539.358    stream] -- Capturing started
-- INFO  [94539.358    stream] -- Using JPEG quality: 80%
-- INFO  [94539.358    stream] -- Creating pool JPEG with 4 workers ...
-- INFO  [94539.359    stream] -- Capturing ...
-- INFO  [94564.348      http] -- HTTP: NEW client (now=1): [192.168.0.165]:52521, id=f9a68a7396cd1c34
`

The stream and the snapshop are available from my browser using 192.168.0.182:8081/stream and /snapshot

But in this case, I can't have the stream on MainsalOS, using http://192.168.0.182:8081 URLs

What's wrong with my setup ?

### What did you expect to happen

Get stream in Mainsail dashboard

### How to reproduce

I don't know

meteyou, 2023-03-21:
You have to use the nginx redirects /webcam for 8080 and /webcam2 for 8081. Pls read the description in your crowsnest.conf

mikebzh44, 2023-03-22:
Hello and thank you for pointing my mistake :D 

I haven't undestood the link between port for ustreamer and nginx redirections.

My webcam is displayed on my dashboard now.

closed 2023-03-22 by mikebzh44