mainsail-crew/crowsnest #101: unmet dependencies ...Debian, Orange pi

keshka, 2023-03-22:
### What happened

attempted to install via kiauh after Klipper and mainsail installtion. Installation fails due to unmet dependencies. Newer versions of the files it lists are installed. 

sudo apt-get update
and
sudo apt-get upgrade 
were executed.

This is a new system install and new klipper/mainsail/crowsnest setup on an OrangePi 4 LTS

### What did you expect to happen

well....working would be good!

### How to reproduce

orangepi@orangepi4-lts:~/crowsnest$ sudo make install

### Additional information

orangepi@orangepi4-lts:~/crowsnest$ sudo make install
crowsnest - A webcam daemon for multiple Cams and stream services.

	Ahoi!
	Thank you for installing crowsnest ;)
	This will take a while ... 
	Please reboot after installation has finished.

Running apt update first ...
Hit:1 http://ftp.us.debian.org/debian bullseye InRelease
Hit:2 http://ftp.us.debian.org/debian bullseye-updates InRelease
Hit:3 http://ftp.us.debian.org/debian bullseye-backports InRelease
Hit:4 http://mirrors.tuna.tsinghua.edu.cn/debian-security bullseye-security InRelease
Reading package lists...
Installing 'crowsnest' Dependencies ...
Reading package lists...
Building dependency tree...
Reading state information...
build-essential is already the newest version (12.9).
findutils is already the newest version (4.8.0-1).
libjpeg-dev is already the newest version (1:2.0.6-4).
libxcomposite1 is already the newest version (1:0.4.5-1).
libxtst6 is already the newest version (2:1.2.3-1).
libxtst6 set to manually installed.
bsdutils is already the newest version (1:2.36.1-8+deb11u1).
curl is already the newest version (7.74.0-1.3+deb11u7).
git is already the newest version (1:2.30.2-1+deb11u2).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ffmpeg : Depends: libavcodec58 (= 7:4.3.5-0+deb11u1)
          Depends: libavdevice58 (= 7:4.3.5-0+deb11u1) but 7:4.3.2-0+deb11u2 is to be installed
          Depends: libavfilter7 (= 7:4.3.5-0+deb11u1)
          Depends: libavformat58 (= 7:4.3.5-0+deb11u1) but 7:4.3.2-0+deb11u2 is to be installed
          Depends: libavresample4 (= 7:4.3.5-0+deb11u1) but 7:4.3.2-0+deb11u2 is to be installed
          Depends: libavutil56 (= 7:4.3.5-0+deb11u1) but 7:4.3.2-0+deb11u2 is to be installed
          Depends: libpostproc55 (= 7:4.3.5-0+deb11u1) but 7:4.3.2-0+deb11u2 is to be installed
          Depends: libswresample3 (= 7:4.3.5-0+deb11u1) but 7:4.3.2-0+deb11u2 is to be installed
          Depends: libswscale5 (= 7:4.3.5-0+deb11u1) but 7:4.3.2-0+deb11u2 is to be installed
E: Unable to correct problems, you have held broken packages.
make: *** [Makefile:37: install] Error 100

KwadFan, 2023-03-23:
Hey there.
I am sorry but this isnt a bug related to crowsnest itself.

`E: Unable to correct problems, you have held broken packages.`

Some packages, listed in your post, are simply outdated but you set them some how on ahold state, which means apt isnt allowed to install. Try removing and reinstalling these packages.

You could also use MainsailOS for your orangePi, we launched prebuilt Images based on armbian for this model couple days ago.

Regards Kwad

KwadFan, 2023-05-22:
Should be solved with current master branch,
closing for now.

Regards Kwad

closed 2023-05-22 by KwadFan