mainsail-crew/crowsnest #102: [Question] Adding IP webcam using UV4L

mikebzh44, 2023-03-23:
### What happened

Hi.

I have D-Link DCS-930L wifi webcam, I can access the streaming using this URL : 

 http://admin:@192.168.0.27/video.cgi?resolution=VGA

But if I'm right, I can use this URL as is because it's not MJPEG stream.

So I found this article : 

https://www.linux-projects.org/uv4l/tutorials/turn-mjpeg-stream-into-camera/

And after installing all the stuff, I start a new stream : 

> uv4l --driver mjpegstream --auto-video_nr --uri "http://admin:@192.168.0.27/video.cgi?resolution=VGA"

<notice> [core] Trying to loading driver 'mjpegstream' from built-in drivers...
<notice> [core] Loading driver 'mjpegstream' from external plug-in's...
<notice> [core] Device detected!
<warning> [core] Cannot create /dev/video0 because file already exists
<warning> [core] Cannot create /dev/video1 because file already exists
<notice> [core] Registering device node /dev/video2

I think it's working.

Then I add a second cam to crowsnest.conf : 

[cam 2]
mode: mjpg
port: 8082
device: /dev/video2
resolution: 640x480
max_fps: 15

But when crowsnest start : 

[03/23/23 18:58:08] crowsnest: INFO: Configuration of Section [cam 2] looks good. Continue...
[03/23/23 18:58:12] crowsnest: DEBUG: ustreamer [cam 2]: Parameters: --host 127.0.0.1 -p 8082 -d /dev/video2 --device-timeout=2 -r 640x480 -f 15 --allow-origin=* --static /home/pi/crowsnest/ustreamer-www
[03/23/23 18:58:12] crowsnest: DEBUG: ustreamer [cam 2]: -- INFO [2239.832 main] -- Starting PiKVM uStreamer 5.38 ...
[03/23/23 18:58:12] crowsnest: DEBUG: ustreamer [cam 2]: -- INFO [2239.832 main] -- Using internal blank placeholder
[03/23/23 18:58:12] crowsnest: DEBUG: ustreamer [cam 2]: -- INFO [2239.832 main] -- Enabling HTTP file server: /home/pi/crowsnest/ustreamer-www
[03/23/23 18:58:12] crowsnest: DEBUG: ustreamer [cam 2]: -- INFO [2239.833 main] -- Listening HTTP on [127.0.0.1]:8082
[03/23/23 18:58:12] crowsnest: DEBUG: ustreamer [cam 2]: -- INFO [2239.833 stream] -- Using V4L2 device: /dev/video2
[03/23/23 18:58:12] crowsnest: DEBUG: ustreamer [cam 2]: -- INFO [2239.833 stream] -- Using desired FPS: 15
[03/23/23 18:58:12] crowsnest: DEBUG: ustreamer [cam 2]: -- INFO [2239.833 http] -- Starting HTTP eventloop ...
[03/23/23 18:58:12] crowsnest: DEBUG: ustreamer [cam 2]: ================================================================================
[03/23/23 18:58:13] crowsnest: DEBUG: ustreamer [cam 2]: -- INFO [2240.876 stream] -- Device fd=8 opened
[03/23/23 18:58:13] crowsnest: DEBUG: ustreamer [cam 2]: -- INFO [2240.876 stream] -- Using input channel: 0
[03/23/23 18:58:13] crowsnest: DEBUG: ustreamer [cam 2]: -- ERROR [2240.877 stream] -- Can't set device format: Invalid argument
[03/23/23 18:58:13] crowsnest: DEBUG: ustreamer [cam 2]: -- INFO [2240.877 stream] -- Device fd=8 closed
[03/23/23 18:58:13] crowsnest: DEBUG: ustreamer [cam 2]: -- INFO [2240.877 stream] -- Sleeping 1 seconds before new stream init ...
[03/23/23 18:58:14] crowsnest: DEBUG: ustreamer [cam 2]: ================================================================================

Can I use this kind of webcam in Crowsnest ?

Thanks

### What did you expect to happen

Expect to have the video displayed in Mainsail ;)

### How to reproduce

Not possible

KwadFan, 2023-03-24:
I am sorry, but crowsnest is designed to stream locally installed cams ( USB or Raspicams ).
I am not intressted in getting IP Cameras to work using UV4L.

If your IP camera can nativly deliver an http - mjpeg stream just use it in mainsail. If not, you are own your own, sorry mate.

Regards Kwad

closed 2023-03-24 by KwadFan

mikebzh44, 2023-03-25:
OK, thanks.

There is so many layers, it's hard to understand which one did what ;)

I will buy a USB webcam.