mainsail-crew/crowsnest #103: External streams

flow23, 2023-03-25:
### Is your feature request related to a problem? Please describe

No

### Describe the solution you'd like

Is there a way to use crowsnest with external streams (e.g. from a ip camera/ptz camera)?
Either via mjpeg-stream or rtsp.

### Additional information

I couldn't find any information in the README.md nor do I know if there is a way to use an external stream shown in mainsail.

meteyou, 2023-03-25:
Crowsnest is a streamer for usb/CSI webcams to create a stream.

If your webcam already have a stream, you don't need it. Just use it directly in mainsail!

KwadFan, 2023-05-22:
Currently I am not thinking about that ability.
I think crowsnest has its purpose, but reusing external streams was never in my mind.
Therefor Sorry mate, will not do something in that direction. 

Regards Kwad

closed 2023-05-22 by KwadFan