mainsail-crew/crowsnest #106: Stream Problems with 2 Cameras

sahck2003, 2023-04-08:
### What happened

Hello, I have two cameras on my system and use the developer version of Crowsnest. I use this because I have connected an RPI3 camera. The other is a Logitech C270

If I set both cams to multi, both cameras no longer work.
If I only set the RPI3 Camera to Multi and the C270 to mjpg, the Stream from the C270 does not work.
I Can see only the RPI3 Stream.

### What did you expect to happen

Both streams should work

### How to reproduce

Change the mode mpjg to multi for the C270

### Additional information

[crowsnest]
log_path: ~/printer_data/logs/crowsnest.log
log_level: verbose                      # Valid Options are quiet/verbose/debug
delete_log: false                       # Deletes log on every restart, if set to true

[cam picam]
mode: multi                              # mjpg/multi - Multi uses webrtc, mjpg and snapshots at the same time
enable_rtsp: false                      # If multi is used, this enables also usage of an rtsp server
rtsp_port: 8554                         # Set different ports for each device!
port: 8080                              # HTTP/MJPG Stream/Snapshot Port
device: /base/soc/i2c0mux/i2c@1/imx708@1a                     # See Log for available ...
resolution: 4608×2592                     # widthxheight format
max_fps: 30                             # If Hardware Supports this it will be forced, otherwise ignored/coerced.
#custom_flags:                          # You can run the Stream Services with custom flags.
#v4l2ctl:                               # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.

[cam c270]
mode: mjpg                              # mjpg/multi - Multi uses webrtc, mjpg and snapshots at the same time
enable_rtsp: false                      # If multi is used, this enables also usage of an rtsp server
rtsp_port: 8555                         # Set different ports for each device!
port: 8081                              # HTTP/MJPG Stream/Snapshot Port
device: /dev/video2                     # See Log for available ...
resolution: 1268x800                     # widthxheight format
max_fps: 30                             # If Hardware Supports this it will be forced, otherwise ignored/coerced.
#custom_flags:                          # You can run the Stream Services with custom flags.
#v4l2ctl:                               # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.

sahck2003, 2023-04-08:
The Stream works. That was my Fault. But two Cameras on mode Multi are not working for me.

KwadFan, 2023-04-08:
Hey @sahck2003,

Thanks for reporting this, but

1. You are running the develop branch, which the implies, its a developing branch and not for production use.
2. This branch has currently no support and we do not handle bug reports yet.
3. You are using waaaayyy to high resolutions a raspberry pi HW encoder, which camera-streamer uses, is limited to 1080p@60FPS overall. This means it shares it for all cameras... Dont expect wonders please, its a pi not an full GPU like in a PC...

So, I will not close this yet to keep track of, but I think it wont get better due hardware limitations.

Regards Kwad

sahck2003, 2023-04-08:
Hey @KwadFan, I just wanted to report this because a feedback on the developer branch was desired. At least that's how I read it :) I also don't expect this bug to be fixed in this Branch. I just wanted to leave some feedback. 

Regards sahck2003

KwadFan, 2023-04-08:
Great, in that regard, your submit is welcome. Thanks.

ayufan, 2023-04-08:
@sahck2003 I need logs.

sahck2003, 2023-04-09:
Hello , here are the logs.

config.txt

<details><summary>Details</summary>
<p>

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
[... 511 lines cut ...]
[   61.571614] CPU: 3 PID: 1423 Comm: camera-streamer Tainted: G         C         6.1.19-v7+ #1637
[   61.571623] Hardware name: BCM2835
[   61.571636]  unwind_backtrace from show_stack+0x18/0x1c
[   61.571657]  show_stack from dump_stack_lvl+0x68/0x8c
[   61.571672]  dump_stack_lvl from __warn+0x84/0x124
[   61.571686]  __warn from warn_slowpath_fmt+0xac/0x170
[   61.571699]  warn_slowpath_fmt from media_pipeline_stop+0x28/0x34 [mc]
[   61.571752]  media_pipeline_stop [mc] from unicam_stop_streaming+0x8c/0x168 [bcm2835_unicam]
[   61.571822]  unicam_stop_streaming [bcm2835_unicam] from __vb2_queue_cancel+0x30/0x25c [videobuf2_common]
[   61.571890]  __vb2_queue_cancel [videobuf2_common] from vb2_core_queue_release+0x20/0x40 [videobuf2_common]
[   61.571968]  vb2_core_queue_release [videobuf2_common] from _vb2_fop_release+0x6c/0x8c [videobuf2_v4l2]
[   61.572036]  _vb2_fop_release [videobuf2_v4l2] from unicam_v4l2_release+0x44/0xd0 [bcm2835_unicam]
[   61.572084]  unicam_v4l2_release [bcm2835_unicam] from v4l2_release+0xd0/0xdc [videodev]
[   61.572245]  v4l2_release [videodev] from __fput+0x84/0x26c
[   61.572380]  __fput from task_work_run+0xa0/0xcc
[   61.572401]  task_work_run from do_exit+0x394/0xad8
[   61.572419]  do_exit from do_group_exit+0x40/0x8c
[   61.572435]  do_group_exit from copy_siginfo_to_user+0x0/0x5c
[   61.572464] ---[ end trace 0000000000000000 ]---
[   64.328716] usb 1-1.3: reset high-speed USB device number 4 using dwc_otg
[   73.918760] usb 1-1.3: reset high-speed USB device number 4 using dwc_otg
``` 

</p>
</details>

smwoodward, 2023-05-09:
Would you be using a CM4 with a Manta board?

sahck2003, 2023-06-24:
No, Raspberry Pi 3 with an Octopus 1.1.

mryel00, 2024-03-24:
camera-streamer is merged for quite some time now. So I would close this and ask you to either join our Discord or our Github Discussions, if you still encounter this problem. The links can be found in the bot message below.

closed 2024-03-24 by github-actions[bot]